gjr80/weewx-gw1000 #1: CRITICAL error in syslog, data collecting/reporting stopped.

dmtalon, 2020-07-24:
New install of weewx on Raspberry Pi.  Was using ecowitt-client interceptor but switched to this today.  Started up and seemed find however it appears to have crashed.

Started weewx at 15:52, and crashed at 18:36.

**Start**

> Jul 23 17:52:43 raspberrypi systemd[1]: Starting LSB: weewx weather system...
> Jul 23 17:52:44 raspberrypi weewx[17077] INFO __main__: Initializing weewx version 4.1.1
> Jul 23 17:52:44 raspberrypi weewx[17077] INFO __main__: Using Python 3.7.3 (default, Dec 20 2019, 18:57:59) #012[GCC 8.3.0]
> Jul 23 17:52:44 raspberrypi weewx[17077] INFO __main__: Platform Linux-5.4.51-v7+-armv7l-with-debian-10.4
> Jul 23 17:52:44 raspberrypi weewx[17077] INFO __main__: Locale is 'en_US.UTF-8'
> Jul 23 17:52:44 raspberrypi weewx[17077] INFO __main__: PID file is /var/run/weewx.pid
> Jul 23 17:52:44 raspberrypi weewx[17082] INFO __main__: Using configuration file /home/weewx/weewx.conf
> Jul 23 17:52:44 raspberrypi weewx[17082] INFO __main__: Debug is 0
> Jul 23 17:52:44 raspberrypi weewx[17082] INFO weewx.engine: Loading station type GW1000 (user.gw1000)
> Jul 23 17:52:44 raspberrypi weewx[17066]: Starting weewx weather system: weewx.
> Jul 23 17:52:44 raspberrypi systemd[1]: Started LSB: weewx weather system.

**Crash**

> Jul 23 18:35:58 raspberrypi weewx[17082] INFO weeutil.rsyncupload: rsync'd 23 files (171,099 bytes) in 1.64 seconds
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****  Traceback (most recent call last):
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1294, i
> n run
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      self.client.collect_sensor_data()
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1116, i
> n collect_sensor_data
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      filtered_data = self.get_live_sensor_data()
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1147, i
> n get_live_sensor_data
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      filtered_data = self.filter_battery_data(parsed_d
> ata)
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1170, i
> n filter_battery_data
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      sensor_list = self.sensor_id_data
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1213, i
> n sensor_id_data
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      sensor_id_response = self.station.get_sensor_id()
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1475, i
> n get_sensor_id
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      return self.send_cmd_with_retries('CMD_READ_SENSO
> R_ID')
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1520, i
> n send_cmd_with_retries
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      self.check_response(response, self.commands[cmd])
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 1581, i
> n check_response
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cm
> d_code):
> Jul 23 18:36:54 raspberrypi weewx[17082] CRITICAL gw1000: user.gw1000:     ****  IndexError: index out of range

gjr80, 2020-07-24:
Thanks for the report. By the looks of the log extract a request to the API resulted in a malformed response which crashed the code that checks the validity of the response. Whilst it is an unexpected exception it is a non-critical in nature and should result in another attempt to issue the command rather than bringing down the driver/WeeWX. 

I have reworked the code and it should now catch such errors, log them and continue on another attempt (if any are left).

Could you try b5 on the [releases tab](https://github.com/gjr80/weewx-gw1000/releases).

Please let us know if these continually occur.

dmtalon, 2020-07-24:
Will do.  And FYI, I restarted Weewx last night and it's been running since w/o issues.  I'll install the latest and let you know if I encounter any issues.

Thanks!

dmtalon, 2020-07-26:
Looking good, going strong.  Closing.

Thanks!

closed 2020-07-26 by dmtalon