gjr80/weewx-gw1000 #4: Testing the driver with errors

lemuba, 2020-07-24:
root@DietPi:~#  PYTHONPATH=/home/weewx/bin python -m user.gw1000 --test-driver
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/weewx/bin/user/gw1000.py", line 267, in <module>
    import weecfg
  File "/home/weewx/bin/weecfg/__init__.py", line 22, in <module>
    import configobj
ImportError: No module named configobj
root@DietPi:~#  PYTHONPATH=/home/weewx/bin python -m gw1000 --test-driver
/usr/bin/python: No module named gw1000                                                                                      
root@DietPi:~#  PYTHONPATH=/home/weewx/bin python -m user.gw1000 --test-driver                                               
Traceback (most recent call last):                                                                                           
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main                                                       
    "__main__", fname, loader, pkg_name)                                                                                     
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code                                                                  
    exec code in run_globals                                                                                                 
  File "/home/weewx/bin/user/gw1000.py", line 267, in <module>                                                               
    import weecfg                                                                                                            
  File "/home/weewx/bin/weecfg/__init__.py", line 22, in <module>                                                            
    import configobj                                                                                                         
ImportError: No module named configobj                                                                                       
root@DietPi:~#

lemuba, 2020-07-24:
Got it running by running Python 3.5 as default this way:

https://raspberry-valley.azurewebsites.net/Python-Default-Version/

gjr80, 2020-07-26:
I appreciate a solution has been found, in this case changing the default python version that is run when the command `python` is executed. Another solution is to explicitly specify the python version in the command, eg:
```
$ PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --test-driver
```

I expect this problem will come up more regularly since WeeWX 4.x can be installed and run under python2 or python3. I will add some further explanatory comments in the comments and documents where running the driver directly is mentioned, in fact it should also be possible to catch the import error and suggest the user check their python version.

gjr80, 2020-07-27:
Added note about specifying python version to each place where I have found instructions on how to run the driver directly.

Unfortunately I am unable to trap the import/module not found errors so that i can warn the user of the problem.

closed 2020-07-27 by gjr80