gjr80/weewx-gw1000 #5: 0.1.0b5  - Rain and Rainrate missing

lemuba, 2020-07-26:
Froggit/Ecowitt GW1000, V1.6.1, WS57 Lightning Sensor, 7in1 Outdoor Sensor

lemuba, 2020-07-26:
Might it bee that "rainday" needs to be renamed to "dayRain" in the driver output to get recognized correctly in Weewx?

Al least I found this naming/term in my weewx.conf mqtt part...

 [[MQTT]]
        server_url = mqtt://192.168.12.127:9002/
        topic = weather
        unit_system = METRIC
        binding = archive, loop
        aggregation = aggregate
        log_success = False
        log_failure = True
        [[[inputs]]]
            [[[[dayRain]]]]
                name = dayRain_mm
                units = mm
            [[[[rainRate]]]]
                name = rainRate_mm_per_hour
                units = mm_per_hour

gjr80, 2020-07-26:
Haha, it’s far simpler, there is no mapping for WeeWX field `rain` in the field map - the problem you have when you don’t have a rain gauge for testing. You can make the change yourself or you can wait for me to post an updated file. The change that is needed is to edit `gw1000.py`and at line 401 add the following line:
```
        'rain': 'rain',
```
Save `gw1000.py` and run with `—test-driver` and let it run for a few loop packets, you should see a field `rain` in the loop packets. The first loop packet will have the value `None`, subsequent packets will depend on what rain falls at the time.

lemuba, 2020-07-26:
Ok, changed - will see :-)

gjr80, 2020-07-26:
Ok, I may have slightly misunderstood. I thought the issue was just missing field `rain`, but reading again I think you were also after field `dayRain`. If that is the case try adding the following under `[GW1000]` in `weewx.conf`:
```
[GW1000]
    ....
    [[field_map_extensions]]
        dayRain = rainday
```

lemuba, 2020-07-28:
Can be closed, new rain now well recoreded also in NOAA reports.

closed 2020-07-28 by gjr80