gjr80/weewx-gw1000 #6: 0.1.0b5 - Solar Radiation Mising

lemuba, 2020-07-26:
Froggit/Ecowitt GW1000, V1.6.1, WS57 Lightning Sensor, 7in1 Outdoor Sensor

Aaccording to WS-View, it should be currently 63,4 w/m², but nothing displayed in Weewx.

lemuba, 2020-07-26:
root@DietPi:~# PYTHONPATH=/home/weewx/bin python -m user.gw1000 --debug=3 --sensors
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 192.168.12.189:45000                                                                                 
                                                                                                                             
Sensor     Status                                                                                                            
WH65       sensor ID: 79  signal: 0  battery: 4                                                                              
WS68       sensor is registering...                                                                                          
WS80       sensor is registering...                                                                                          
WH40       sensor is registering...                                                                                          
WH32       sensor is registering...                                                                                          
WH31 ch1   sensor is registering...                                                                                          
WH31 ch2   sensor is registering...                                                                                          
WH31 ch3   sensor is registering...                                                                                          
WH31 ch4   sensor is registering...                                                                                          
WH31 ch5   sensor is registering...                                                                                          
WH31 ch6   sensor is registering...                                                                                          
WH31 ch7   sensor is registering...                                                                                          
WH31 ch8   sensor is registering...                                                                                          
WH51 ch1   sensor is registering...                                                                                          
WH51 ch2   sensor is registering...                                                                                          
WH51 ch3   sensor is registering...                                                                                          
WH51 ch4   sensor is registering...                                                                                          
WH51 ch5   sensor is registering...
WH51 ch6   sensor is registering...
WH51 ch7   sensor is registering...
WH51 ch8   sensor is registering...
WH41 ch1   sensor is registering...
WH41 ch2   sensor is registering...
WH41 ch3   sensor is registering...
WH41 ch4   sensor is registering...
WH57       sensor ID: d03e  signal: 3  battery: 4
WH55 ch1   sensor is registering...
WH55 ch2   sensor is registering...
WH55 ch3   sensor is registering...
WH55 ch4   sensor is registering...
root@DietPi:~# 

91,5 w/m² in WS-View, at the same time:

root@DietPi:~# PYTHONPATH=/home/weewx/bin python -m user.gw1000 --debug=3 --live-data
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 192.168.12.189:45000                                                                                 
                                                                                                                             
GW1000 live sensor data: absbarometer: 1003.5, datetime: 1595750970, daymaxwind: 4.6, gustspeed: 0.5, inhumid: 52, intemp: 26.5, light: 11594.0, lightningcount: 0, lightningdettime: None, lightningdist: None, outhumid: 96, outtemp: 17.7, rainday: 0.8, rainevent: 0.5, rainmonth: 84.6, rainrate: 0.0, rainweek: 0.8, rainyear: 179.5, relbarometer: 1003.5, uv: 9.1, uvi: 0, wh57_batt: 3, winddir: 10.6, windspeed: 0.4                                                                                      
root@DietPi:~#

lemuba, 2020-07-26:
88w/m²:

GW1000 live sensor data: absbarometer: 1003.5, datetime: 1595751191, daymaxwind: 4.6, gustspeed: 0.0, inhumid: 52, intemp: 26.5, light: 11144.0, lightningcount: 0, lightningdettime: None, lightningdist: None, outhumid: 96, outtemp: 17.7, rainday: 0.8, rainevent: 0.5, rainmonth: 84.6, rainrate: 0.0, rainweek: 0.8, rainyear: 179.5, relbarometer: 1003.5, uv: 8.8, uvi: 0, wh57_batt: 4, winddir: 6.2, windspeed: 0.0                                                                                       
root@DietPi:~#

gjr80, 2020-07-26:
Thanks but can you provide the corresponding log entries as well please, it should show a lot of hex characters, I need to marry up the raw data from the GW1000 with what is being observed and what comes through when decided as per the GW1000. The logs have the raw data.

lemuba, 2020-07-26:
Puhh, where do I find those logs? Could it be in /var/log/syslog, which is currently empty on my system...?

http://weewx.com/docs/usersguide.htm#Nothing_in_the_log_file

gjr80, 2020-07-26:
That is a good question, it depends on your system. What operating system and version?

lemuba, 2020-07-26:
Dietpi Debian System on a Virtual Machine (Synology Diskstation Virtual Machine Manager). I´m currently installing rsyslog and will check again...

lemuba, 2020-07-26:
271,1 w/m²:

root@DietPi:~# PYTHONPATH=/home/weewx/bin python -m user.gw1000 --debug=3 --sensors
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 192.168.12.189:45000                                                                                 
                                                                                                                             
Sensor     Status                                                                                                            
WH65       sensor ID: 79  signal: 0  battery: 4                                                                              
WS68       sensor is registering...                                                                                          
WS80       sensor is registering...                                                                                          
WH40       sensor is registering...                                                                                          
WH32       sensor is registering...                                                                                          
WH31 ch1   sensor is registering...                                                                                          
WH31 ch2   sensor is registering...                                                                                          
WH31 ch3   sensor is registering...                                                                                          
WH31 ch4   sensor is registering...                                                                                          
WH31 ch5   sensor is registering...                                                                                          
WH31 ch6   sensor is registering...                                                                                          
WH31 ch7   sensor is registering...                                                                                          
WH31 ch8   sensor is registering...                                                                                          
WH51 ch1   sensor is registering...                                                                                          
WH51 ch2   sensor is registering...                                                                                          
WH51 ch3   sensor is registering...                                                                                          
WH51 ch4   sensor is registering...                                                                                          
WH51 ch5   sensor is registering...
WH51 ch6   sensor is registering...
WH51 ch7   sensor is registering...
WH51 ch8   sensor is registering...
WH41 ch1   sensor is registering...
WH41 ch2   sensor is registering...
WH41 ch3   sensor is registering...
WH41 ch4   sensor is registering...
WH57       sensor ID: d03e  signal: 3  battery: 4
WH55 ch1   sensor is registering...
WH55 ch2   sensor is registering...
WH55 ch3   sensor is registering...
WH55 ch4   sensor is registering...
root@DietPi:~# PYTHONPATH=/home/weewx/bin python -m user.gw1000 --debug=3 --live-data
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 192.168.12.189:45000

GW1000 live sensor data: absbarometer: 1004.6, datetime: 1595757565, daymaxwind: 4.6, gustspeed: 1.5, inhumid: 51, intemp: 26.9, light: 34131.0, lightningcount: 0, lightningdettime: None, lightningdist: None, outhumid: 91, outtemp: 19.7, rainday: 1.8, rainevent: 1.0, rainmonth: 85.6, rainrate: 0.0, rainweek: 1.8, rainyear: 180.5, relbarometer: 1004.6, uv: 83.4, uvi: 2, wh57_batt: 3, winddir: 31.8, windspeed: 1.1
root@DietPi:~# 

Log:

Jul 26 11:58:58 DietPi weewx[14707] DEBUG gw1000: __main__: Sending broadcast packet 'FF FF 12 03 15' to '255.255.255.255:46000'
Jul 26 11:58:58 DietPi weewx[14707] DEBUG gw1000: __main__: Received broadcast response 'FF FF 12 00 21 A4 CF 12 B4 FB 23 C0 A8 0C BD AF C8 10 47 57 31 30 30 30 41 2D 57 49 46 49 46 42 32 33 0A'
Jul 26 11:59:00 DietPi weewx[14707] INFO gw1000: __main__: GW1000 was found at 192.168.12.189:45000
Jul 26 11:59:00 DietPi weewx[14707] DEBUG gw1000: __main__: Sending packet 'FF FF 3A 03 3D' to '192.168.12.189:45000'
Jul 26 11:59:00 DietPi weewx[14707] DEBUG gw1000: __main__: Received response 'FF FF 3A D5 00 00 00 00 79 00 04 01 FF FF FF FF FF 00 02 FF FF FF FF FF 00 03 FF FF FF FF 00 00 05 FF FF FF FF 00 00 06 FF FF FF FF 00 00 07 FF FF FF FF 00 00 08 FF FF FF FF 00 00 09 FF FF FF FF 00 00 0A FF FF FF FF 00 00 0B FF FF FF FF 00 00 0C FF FF FF FF 00 00 0D FF FF FF FF 00 00 0E FF FF FF FF 00 00 0F FF FF FF FF 00 00 10 FF FF FF FF 00 00 11 FF FF FF FF 00 00 12 FF FF FF FF 00 00 13 FF FF FF FF 00 00 14 FF FF FF FF 00 00 15 FF FF FF FF 00 00 16 FF FF FF FF 0F 00 17 FF FF FF FF 0F 00 18 FF FF FF FF 0F 00 19 FF FF FF FF 0F 00 1A 00 00 D0 3E 03 04 1B FF FF FF FF FF 00 1C FF FF FF FF FF 00 1D FF FF FF FF FF 00 1E FF FF FF FF FF 00 34'
Jul 26 11:59:23 DietPi weewx[14925] DEBUG gw1000: __main__: Sending broadcast packet 'FF FF 12 03 15' to '255.255.255.255:46000'
Jul 26 11:59:23 DietPi weewx[14925] DEBUG gw1000: __main__: Received broadcast response 'FF FF 12 00 21 A4 CF 12 B4 FB 23 C0 A8 0C BD AF C8 10 47 57 31 30 30 30 41 2D 57 49 46 49 46 42 32 33 0A'
Jul 26 11:59:25 DietPi weewx[14925] INFO gw1000: __main__: GW1000 was found at 192.168.12.189:45000
[... cut ...]
Jul 26 12:00:24 DietPi weewx[7359] INFO weewx.restx: PWSWeather: Published record 2020-07-26 12:00:00 CEST (1595757600)
Jul 26 12:00:26 DietPi weewx[7359] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 2.32 seconds
Jul 26 12:00:34 DietPi weewx[7359] INFO weewx.imagegenerator: Generated 45 images for report SeasonsReport in 7.44 seconds
Jul 26 12:00:34 DietPi weewx[7359] INFO weewx.reportengine: Copied 0 files to /home/weewx/public_html
Jul 26 12:00:34 DietPi weewx[7359] INFO weewx.cheetahgenerator: Generated 6 files for report SmartphoneReport in 0.11 seconds

gjr80, 2020-07-26:
Thanks, exactly what I was after.

lemuba, 2020-07-26:
Ok, just as info - after installing rsyslog, I recognized that all my logs as syslog and user.log are mouted by standard to a 50Mbyte RAM partition and deleted hourly. So not poluting my system;-) If you need anything else, let me know and I will initiate - beeing happy and thankful that you develop this driver!

Matthias

gjr80, 2020-07-26:
Ok, I have enough for now.

gus70, 2020-07-26:
The best way to convert Lux to W/m2 is to divide by 126.7
https://help.ambientweather.net/help/why-is-the-lux-to-w-m-2-conversion-factor-126-7/

gary-hammer, 2020-07-27:
Here's where I get lost. To my eye, it seem that 'light' is being found and parsed by the driver. Yet, weewx fails to use 'light'

Jul 27 10:37:08 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Sending broadcast packet 'FF FF 12 03 15' to '255.255.255.255:46000'
Jul 27 10:37:08 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Received broadcast response 'FF FF 12 00 21 BC DD C2 AF 18 B6 0A 0A 64 7D AF C8 10 47 57 31 30 30 30 42 2D 57 49 46 49 31 38 42 36 44'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] INFO gw1000: __main__: GW1000 was found at 10.10.100.125:45000
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Sending packet 'FF FF 27 03 2A' to '10.10.100.125:45000'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Received response 'FF FF 27 00 51 01 00 FC 06 3F 08 26 DB 09 27 77 02 01 20 07 3F 0A 00 FE 0B 00 08 0C 00 0A 15 00 0D E1 E8 16 08 F4 17 06 4C 0F 00 00 00 FF 5B FF 00 FF FF FF FF FF FF FF FF 19 00 29 0E 00 00 10 00 00 11 00 00 12 00 00 04 79 13 00 00 14 2C 0D 00 00 2A'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: sensor data is '01 00 FC 06 3F 08 26 DB 09 27 77 02 01 20 07 3F 0A 00 FE 0B 00 08 0C 00 0A 15 00 0D E1 E8 16 08 F4 17 06 4C 0F 00 00 00 FF 5B FF 00 FF FF FF FF FF FF FF FF 19 00 29 0E 00 00 10 00 00 11 00 00 12 00 00 04 79 13 00 00 14 2C 0D 00 00'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Parsed data: {'intemp': 25.2, 'inhumid': 63, 'absbarometer': 994.7, 'relbarometer': 1010.3, 'outtemp': 28.8, 'outhumid': 63, 'winddir': 25.4, 'windspeed': 0.8, 'gustspeed': 1.0, 'light': 90980.0, 'uv': 229.2, 'uvi': 6, 'wh40_batt': 0, 'wh26_batt': 0, 'wh25_batt': 0, 'wh24_batt': 0, 'wh31_ch1_batt': 0, 'wh31_ch2_batt': 0, 'wh31_ch3_batt': 0, 'wh31_ch4_batt': 0, 'wh31_ch5_batt': 0, 'wh31_ch6_batt': 0, 'wh31_ch7_batt': 0, 'wh31_ch8_batt': 0, 'wh51_ch1_batt': 0, 'wh51_ch2_batt': 0, 'wh51_ch3_batt': 0, 'wh51_ch4_batt': 0, 'wh51_ch5_batt': 0, 'wh51_ch6_batt': 0, 'wh51_ch7_batt': 0, 'wh51_ch8_batt': 0, 'wh51_ch9_batt': 0, 'wh51_ch10_batt': 0, 'wh51_ch11_batt': 0, 'wh51_ch12_batt': 0, 'wh51_ch13_batt': 0, 'wh51_ch14_batt': 0, 'wh51_ch15_batt': 0, 'wh51_ch16_batt': 0, 'wh57_batt': 255, 'ws68_batt': 1.82, 'ws80_batt': 5.1000000000000005, 'wh41_ch1_batt': None, 'wh41_ch2_batt': None, 'wh41_ch3_batt': None, 'wh41_ch4_batt': None, 'wh55_ch1_batt': None, 'wh55_ch2_batt': None, 'wh55_ch3_batt': None, 'wh55_ch4_batt': None, 'daymaxwind': 4.1, 'rainrate': 0.0, 'rainday': 0.0, 'rainweek': 0.0, 'rainmonth': 114.5, 'rainyear': 516.4, 'rainevent': 0.0, 'datetime': 1595860630}
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Sending packet 'FF FF 3A 03 3D' to '10.10.100.125:45000'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Received response 'FF FF 3A D5 00 FF FF FF FE 00 00 01 00 00 00 8D 5B 04 02 FF FF FF FF FF 00 03 00 00 D1 1D 00 04 05 00 00 00 A5 00 04 06 FF FF FF FF 00 00 07 FF FF FF FF 00 00 08 FF FF FF FF 00 00 09 FF FF FF FF 00 00 0A FF FF FF FF 00 00 0B FF FF FF FF 00 00 0C FF FF FF FF 00 00 0D FF FF FF FF 00 00 0E FF FF FF FF 00 00 0F FF FF FF FF 00 00 10 FF FF FF FF 00 00 11 FF FF FF FF 00 00 12 FF FF FF FF 00 00 13 FF FF FF FF 00 00 14 FF FF FF FF 00 00 15 FF FF FF FF 00 00 16 FF FF FF FF 0F 00 17 FF FF FF FE 0F 00 18 FF FF FF FE 0F 00 19 FF FF FF FE 0F 00 1A FF FF FF FF FF 00 1B FF FF FF FF FF 00 1C FF FF FF FF FF 00 1D FF FF FF FF FF 00 1E FF FF FF FF FF 00 29'
Jul 27 10:37:10 srvr python3[413485]: weewx[413485] DEBUG gw1000: __main__: Filtered parsed data: {'intemp': 25.2, 'inhumid': 63, 'absbarometer': 994.7, 'relbarometer': 1010.3, 'outtemp': 28.8, 'outhumid': 63, 'winddir': 25.4, 'windspeed': 0.8, 'gustspeed': 1.0, 'light': 90980.0, 'uv': 229.2, 'uvi': 6, 'wh40_batt': 0, 'ws68_batt': 1.82, 'daymaxwind': 4.1, 'rainrate': 0.0, 'rainday': 0.0, 'rainweek': 0.0, 'rainmonth': 114.5, 'rainyear': 516.4, 'rainevent': 0.0, 'datetime': 1595860630}

root@srvr:~# PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --debug=3 --live-data
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 10.10.100.125:45000

GW1000 live sensor data: absbarometer: 994.7, datetime: 1595860630, daymaxwind: 4.1, gustspeed: 1.0, inhumid: 63, intemp: 25.2, light: 90980.0, outhumid: 63, outtemp: 28.8, rainday: 0.0, rainevent: 0.0, rainmonth: 114.5, rainrate: 0.0, rainweek: 0.0, rainyear: 516.4, relbarometer: 1010.3, uv: 229.2, uvi: 6, wh40_batt: 0, winddir: 25.4, windspeed: 0.8, ws68_batt: 1.82

weewx.conf entry
[StdCalibrate]
    
    [[Corrections]]
        radiation = light / 126.7 if light is not None else None
# Generic labels, keyed by an observation type.
            [[[[Generic]]]]
                radiation = Radiation

gjr80, 2020-07-27:
I have looked at the raw data coming from the GW1000 and what the API defines as `light` (I will refer to it as refer to it as _illuminance_) is the only 'solar radiation' related observation being emitted by the GW1000. Given this and the linear relationship between the radiation values observed on the console and illuminance values from the API included in some posts on _weewx-user_, it is fairly clear that the GW1000 is dividing illuminance by 126.7 to produce `radiation` for uploads to the likes of WU and other weather sites.

Unfortunately this radiation value is not available via the GW1000 API. The GW1000 driver could calculate and emit it, but that is not the approach used by WeeWX drivers. WeeWX drivers take what data they can obtain from the station, decode and convert units if necessary, and emit that data for WeeWX to process. The driver does not derive observations. WeeWX has the `StdWXCalculate` service for calculating derived observations plus the `StdCalibrate` service can also be used to derive observations that are simple in nature. In this case `radiation` is a derived observation and needs to be derived by WeeWX. Given the debate over whether you can accurately derive `radiation` from an illuminance value, I am not sure we will see `StdWXCalculate` deriving `radiation` from illuminance. That leaves `StdCalibrate`, fortunately `StdCalibrate` can easily be configured by the user to calculate `radiation` from illuminance. Something like the following should produce a field named `radiation` from the GW1000 `light` field:
```
[StdClaibrate]
    ....
    [[Corrections]]
        ....
        radiation = light/126.7 if light is not None else None
```

gjr80, 2020-07-27:
> Here's where I get lost. To my eye, it seem that 'light' is being found and parsed by the driver. 

Correct.

> Yet, weewx fails to use 'light'

What makes you think that? The log extract posted plus the `--live-data` output all show driver emitted data only, they do not show actual loop packets/archive records emitted by a running WeeWX instance. I suggest you run WeeWX directly and observe the contents of the loop packets and archive records. Do they contain `light` and `radiation`? If they contain `light` and not `radiation` then we have a problem and further investigation is required.

gus70, 2020-07-28:
The way that the Lux sensor is intended to be used is to divide by 126.7 and then the user can adjust the gain on the console for their location. Just take in Lux and divide by 126.7 and slap W/m² units on it and call it a day. There is your solar radiation. Yeah it isn't perfect...but the idea of the driver should be to replicate what the console does. You want the numbers to match. It is what it is. A Lux sensor is no substitute for a proper solar radiation sensor but that is what we have to work with.

gary-hammer, 2020-07-28:
> Correct.
>
>
> What makes you think that? The log extract posted plus the `--live-data` output all show driver emitted data only, they do not show actual loop packets/archive records emitted by a [...]

They do not contain light nor radiation. They do contain luminosity

` PYTHONPATH=/home/weewx/bin python3 /home/weewx/bin/weewxd
LOOP:   2020-07-27 21:15:42 EDT (1595898942) altimeter: 29.74131785537736, appTemp: 89.22216742251888, barometer: 29.73374441375, cloudbase: 3055.9655658014062, dateTime: 1595898942, daymaxwind: 6.6, dewpoint: 70.06708940054055, heatindex: 85.44394588835222, humidex: 97.07223283005186, inDewpoint: 60.39412961933657, inHumidity: 54, inTemp: 78.44, luminosity: 0.0, maxSolarRad: 0.0, outHumidity: 68, outTemp: 81.68, pressure: 29.29307660875, rain: None, rainday: 0.2, rainevent: 0.0, rainmonth: 114.7, rain_period: 800, rainRate: 0.0, rainweek: 0.2, rainyear: 516.6, usUnits: 1, UV: 0, uvRadiation: 0.0, windchill: 81.68, windDir: 25.2, windGust: 1.1184709259696521, windSpeed: 0.22369418519393044
`

gjr80, 2020-07-28:
> The way that the Lux sensor is intended to be used is to divide by 126.7 and then the user can adjust the gain on the console for their location. Just take in Lux and divide by 126.7 and slap W/m² [...]

Understand what is happening, but the role of the driver is not to replicate what is on the console but rather to take what data the station provides and provide that data to WeeWX. The WeeWX services can then synthesise whatever derived data may be required. The decision was made some time ago, for a number of reasons, that WeeWX drivers would not derive observations.

gjr80, 2020-07-28:
> They do not contain light nor radiation. They do contain luminosity

Ah, my mistake, the `wview_extended` schema shipped with WeeWX includes field `luminosity` so the default field map maps the 'light' value from the GW1000 to `luminosity`. The `StdCalibrate` stanza needs to be adjusted accordingly:
```
[StdCalibrate]
    ....
    [[Corrections]]
        ....
        radiation = luminosity/126.7 if luminosity is not None else None
```

gary-hammer, 2020-07-28:
> Ah, my mistake, the `wview_extended` schema shipped with WeeWX includes field `luminosity` so the default field map maps the 'light' value from the GW1000 to `luminosity`. The `StdCalibrate` stanza [...]

Bingo! That was it.
Now, if we just get some rain, I can finish testing all sensors.

gus70, 2020-07-28:
Excellent! Good work finding radiation = luminosity/126.7 code. Understood about the driver not deriving values. Glad this functionality was already in the wview_extended.

lemuba, 2020-07-28:
Many thanks, also working for me. Now with the fixed and today counted rain and corrected  wind direction, back there where I was before with the interceptor driver. :-)

closed 2020-07-28 by gjr80