gjr80/weewx-gw1000 #8: build5 Unable to load driver: local variable 'field_map_extensions' referenced before assignment

steepleian, 2020-07-26:
Two Raspberry Pis with identical setups. One is running perfectly, the other gives this error: -

Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****    File "/home/weewx/bin/user/gw1000.py", line 867, in loader
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****      return Gw1000Driver(**config_dict[DRIVER_NAME])
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****    File "/home/weewx/bin/user/gw1000.py", line 952, in __init__
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****      super(Gw1000Driver, self).__init__(**stn_dict)
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****    File "/home/weewx/bin/user/gw1000.py", line 517, in __init__
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****      field_map.update(field_map_extensions)
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL weewx.engine:     ****  UnboundLocalError: local variable 'field_map_extensions' referenced before assignment
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL __main__: Unable to load driver: local variable 'field_map_extensions' referenced before assignment
Jul 27 00:27:06 wxpi4b weewxd[12461]: weewx[12461] CRITICAL __main__:     ****  Exiting...

Field map settings are: -

[GW1000]
    # This section is for the GW1000

    # The driver itself
    driver = user.gw1000
    poll_interval = 16

    
          
        [[field_map]]
            outTemp = outtemp
            windSpeed = windspeed
            UV = uvi
            light = light
            outBatteryStatus = wh65_batt
            outHumidity = outhumid
            windDir = winddir
            windGust = gustspeed
            rain_total = rainday
            inTemp = intemp
            inHumidity = inhumid
            barometer = relbarometer
            inTempBatteryStatus =

gjr80, 2020-07-27:
Ian, can you please post the first 20 lines or so of `class Gw1000Driver __init__()`. My line 517 of b5 if different to that reported in the trace above. I have made some changes to that part of the code for b6 and I would like to understand the exact mechanism that lead to your error.

gjr80, 2020-07-27:
Ian, try wget'ing master now. It is b6 less inclusion of battery states in the default field map. Had wanted to get that in b6 as well but it will have to wait a day.

There have been numerous other changes to the b5 code, unfortunately only documented in the commit comments.

steepleian, 2020-07-27:
Gary, 
You are right. I seem to have two different versions of build 5. The offending one is one that I cut and paste from the raw code on Github about 70min ago. The one that is working OK was from yesterday but both say Version: 0.1.0b5. I am attaching both files.

[gw1000-failing.py.txt](https://github.com/gjr80/weewx-gw1000/files/4979120/gw1000-failing.py.txt)
[gw1000-working.py.txt](https://github.com/gjr80/weewx-gw1000/files/4979121/gw1000-working.py.txt)

steepleian, 2020-07-27:
Confirm that both devices now working correctly since reverting to yesterday's version.

steepleian, 2020-07-27:
Gary, 
Confirm that build 6 is now working OK
Thanks,
Ian

closed 2020-07-27 by steepleian