gjr80/weewx-gw1000 #10: Wind Direction Decimal Place

pmcgeorge, 2020-07-28:
It looks like the decimal place is off on the Wind direction, my current wind direction on the App and on my console is 152 deg, the driver is reporting 15.2, resulting in Weewx seeing Around N constantly. 

 Dump of data is below:

GW1000 live sensor data: 24havpm251: 11.4, absbarometer: 872.5, datetime: 1595939970, daymaxwind: 5.1, gustspeed: 2.0, inhumid: 44, intemp: 22.7, light: 930.0, lightningcount: 0, lightningdettime: 1595896953, lightningdist: 34, outhumid: 63, outtemp: 20.3, pm251: 10.0, rainday: 0.0, rainevent: 0.0, rainmonth: 0.0, rainrate: 0.0, rainweek: 0.0, rainyear: 152.2, relbarometer: 1018.9, soilmoist1: 37, soilmoist2: 52, temp2: -18.3, uv: 1.0, uvi: 0, wh31_ch2_batt: 0, wh41_ch1_batt: 4, wh51_ch1_batt: 0, wh51_ch2_batt: 0, wh57_batt: 5, wh65_batt: 0, winddir: 15.2, windspeed: 1.5

gjr80, 2020-07-28:
Yes, the wind direction decode was using the wrong function. Fixed in b7.

lemuba, 2020-07-28:
Working also for me - thanks! I guess this topic can be closed now.

closed 2020-07-28 by gjr80