gjr80/weewx-gw1000 #11: driver test needs --ip defined not --ip-address

Limb, 2020-07-31:
The README.md states "Note: If necessary you can specify the GW1000 IP address and port using the --ip-address and --port command line options. Refer to the GW1000 driver help using --help for further information."

However the driver actually expects --ip to be defined, and throws an error if --ip-address is used. (The --help command informs you to use --ip)

I would suggest changing the --ip parameter to --ip_address to be consistent with the weewx configuration file, which uses ip_address.

gjr80, 2020-08-01:
Thanks for the feedback, it was a deliberate decision to use `--ip` but I can see it could be confusing. Will fix in b10.

gjr80, 2020-08-01:
Commit 88987259a0ab8cedd2f17cfd496d779337a42796 refers

closed 2020-08-01 by gjr80