gjr80/weewx-gw1000 #12: wh51 batt reading

pmcgeorge, 2020-08-06:
I have two soil moisture sensors (WH51) on my GW1000 and I am getting different data from the New API driver than I was with interceptor.  GW1000 reports a 0 while Interceptor seems to have the voltage of 1.4, I changed the batter and interceptor does see the voltage change.  Is three some data I can provide to help find a way to fix this? Or is this an API issue on GW1000?

On the GW1000 Driver:
GW1000 live sensor data: **** wh51_ch1_batt: 0, wh51_ch2_batt: 0, ****

On Interceptor:
&soilbatt1=1.4&soilbatt2=1.7&

gjr80, 2020-08-06:
That's interesting, I had not noticed. My first thought it is an API limitation as the API provides a single bit for battery state for the each WH51, it is either 1 or 0, 1 being low and 0 being OK (some other sensors have 4 bits and use an arbitrary level of 0-5 and a couple of others report actual battery voltage in volts in a whole byte).

I have fired up a machine running the interceptor driver and for my three WH51s I am seeing 1.7 as well. I have no idea what that means, 1.7V for a 1.5V alkaline cell does not sound right, especially for one that has been in use for a while. I have also looked at my Ecowitt dashboard (or whatever it is called) online, and whilst I can see all my sensors being reported I cannot find any battery state data so no luck trying to track down the 1.7 meaning that way.

I have a person I can ask who has contacts at Ecowitt, might be a long shot but I might see if I can find out what the 1.7 means. As far as the API based driver is concerned though there is not much that can be done if the API only supports 0 or 1.

pmcgeorge, 2020-08-06:
I thought that might be the case.  The 1.7 I have is a newer battery and the 1.4 is about 6 months old.  Funny thing is the database actually shows it varies from 1.3 to 1.4.   More curiosity than anything else, I'll see up for 0 good and 1 low.

InToSSH, 2020-12-20:
Hello, I have the WH68 (Froggit DP300) anemometer and looking at live data from your driver I can see value of 1.8, so that's strange aswell.
Also in the original WS View app this is the only sensor that shows the battery level as an icon with bars.
All other sensors don't show the battery value, just 0 (which should mean OK as discussed here).

gjr80, 2021-04-15:
I have been unable to find anything further on this. The GW1000 firmware v1.6.6 substantially changes the way in which battery status is reported. These changes were implemented in the GW1000 driver v0.3.0 and are largely internal within the API with reporting to the user remaining the same. It certainly does not add battery voltage levels for the sensors mentioned in this issue.

Closing this issue for now, it can be re-opened if needed.

closed 2021-04-15 by gjr80