gjr80/weewx-gw1000 #13: WeeWX "Today Rain" not matching GW1000

Limb, 2020-08-12:
My WeeWX "Rain Today" field is not matching the values shown by the GW1000, WeeWX-GW1000 driver, or the data WeeWX is pushing up to Weather Underground. 

Rain Today shows "0.02 in" on the standard report, whereas the other sources show the correct value of 0.06 inches (even the data that WeeWX itself is uploading to Wunderground which is weird). I'm not 100% sure if this is an error with how the driver is calculating the rain value for WeeWX?

[image]

`2020-08-12 09:57:12 EDT (1597240632): UV: 1, dateTime: 1597240632, dayRain: 1.6, daymaxwind: 2.6, inHumidity: 59, inTemp: 26.9, luminosity: 25260.0, monthRain: 9.2, outHumidity: 90, outTemp: 26.6, pressure: 1018.3, rain: None, rainRate: 0.0, relbarometer: 1018.3, stormRain: 1.6, usUnits: 17, uvradiation: 9.9, weekRain: 1.6, wh26_batt: 0, wh40_batt: 0, wh68_batt: 1.64, windDir: 162, windGust: 0.0, windSpeed: 0.0, yearRain: 39.7`

[image]
[image]

gjr80, 2020-08-13:
OK, let's see if we can break this down a bit to work out what is right/wrong or consistent/inconsistent.

The GW1000 driver obtains a number of rainfall fields from the GW1000; these fields are (based on a the default field map): 
```
stormRain
hourRain
dayRain
weekRain
monthRain
yearRain
totalRain
```
The driver passes these fields to WeeWX via loop packets and WeeWX synthesises archive records containing these fields. Assuming the `[Accumulator]` stanza in `weewx.conf` has been set correctly the extractor function for each of these fields should be `last` meaning the last loop packet value seen will be used as the value in the archive record. Question, is `[Accumulator]` set correctly? The default extractor function is `average` which if not set correctly could lead to small under reporting of rainfall.

When WeeWX publishes to WU WeeWX looks for the field `dayRain` in the archive record (or loop packet if using RF) and if it exists that is what is published to WU. If `dayRain` does not exist then WeeWX synthesises `dayRain` by querying the archive and summing all of the `rain` values for today's archive records and publishes the resulting value as today's rainfall.

The value for today's rainfall in the Standard skin (and for that matter the Seasons skin too) does not use the `dayRain` field rather it uses the sum of the rain in today's archive records.

So going back to your observations:

> My WeeWX "Rain Today" field is not matching the values shown by the GW1000, WeeWX-GW1000 driver, or the data WeeWX is pushing up to Weather Underground.
>
> Rain Today shows "0.02 in" on the standard [...]

The rain today value shown by the GW1000 (I presume through the WS View app) should be absolutely correct. The `dayRain` value obtained and emitted by the GW1000 driver appears to agree with the GW1000. As explained the data published by WeeWX to WU will be the `dayRain` field if it exists, it does exist and is correct so WU has the correct day rainfall data. The 'rain today' data in skin based reports uses WeeWX field `rain` summed over the day and that is showing incorrect data.

So it would appear that everything is working as it should but it looks like there is a problem with field `rain` either as emitted by the driver or as processed by WeeWX. Given WeeWX' maturity in handling data fields such as `rain` the obvious place to look is the driver. WeeWX works on per-period rainfall values, ie in the archive the field `rain` in any given record holds the amount of rainfall seen during the period covered by that archive record. Unfortunately the GW1000 only emits cumulative rainfall data so the driver needs to calculate the per-period rainfall by calculating the difference between successive values of one of the cumulative fields. The driver attempts to use the 'longest period' field `totalRain` to do this calculation though if `totalRain` does not exist it will look for `yearRain` and then `monthRain`. There are a few issues with this approach:

1. When the driver starts up and the first lot of data is received from the GW1000 the driver has no idea what the previous `totalRain` value was so field `rain` cannot be calculated and is set to None. So any rainfall in that period is lost to WeeWX. this is only an issue if it is raining at the time of start up.

2. The chosen cumulative field will reset to zero or wraparound at some time, whilst the previous and current values are known it is impossible to determine how much rain fell in the period. In these cases the current cumulative value is used as field `rain` but again it impossible to know if that value is correct or is under (it won't be over though). Any 'under' calculated rainfall is lost to WeeWX. The more frequently a cumulative field wraps around the more often there could be lost rainfall, that is why we use the longest period cumulative rainfall field rather than `hourRain`.

3. If there was rainfall on the day that WeeWX was first used with the GW1000 driver and some of that rainfall fell before WeeWX was started then that rainfall will not be seen by WeeWX. This only applies to the first day the driver was used.

Long way to get around to it but it is quite possible that rainfall can be slightly under according to WeeWX though such cases should be few and far between. The only real way to track down the issue is through recording and analysing the source cumulative rainfall data and the corresponding calculated `rain` data and seeing when the discrepancy occurs and then hopefully identifying why and fixing it if possible.

If you are up for this (I don't have a rain gauge connected to my GW1000) you can capture the necessary data in your log next time it rains and we can see what we can determine. The current driver does have the ability to log the necessary data but it will log a whole pile of other data as well and it will be a nightmare to decipher and analyse. What I can do is implement a `debug_rainfall` config option that when set will just log the current and last cumulative value and the calculated `rain` value, that should make it pretty easy to identify and analyse the data.

gjr80, 2020-08-13:
Just to follow on, I noticed you don't have a field `totalRain`, not really a problem as the driver is coded such that it will drop down to use `yearRain`. The field being used should be logged at startu p (well when the first GW1000 data arrives). You might want to have a look in your log at the last start up and confirm what field is being used.

Limb, 2020-08-13:
So the startup log looks like it's using `rainyear` for the total (does the fact that this is called `yearRain` in the driver matter or are the two mapped together?):
```
Aug 13 08:47:47 weewx weewx[10756] INFO gw1000: user.gw1000: using 'rainyear' for rain total
Aug 13 08:47:47 weewx weewx[10756] INFO gw1000: user.gw1000: skipping rain measurement of 39.7: no last rain
```

My extractors are all set to `last` which is the default inserted by the extension install script, are you saying one or more of these should actually be set to `average`?
```
[[stormRain]]
    extractor = last
[[hourRain]]
    extractor = last
[[dayRain]]
    extractor = last
[[weekRain]]
    extractor = last
[[monthRain]]
    extractor = last
[[yearRain]]
    extractor = last
[[totalRain]]
    extractor = last
```

We are supposed to be expecting rain in a few hours, but not much. I also have no issue with just hitting my WH40 rain sensor with a hose to force simulate some rainfall. Would logging the data from `--test-driver` provide you the information you need without adding a `debug_rainfall` command?

gjr80, 2020-08-14:
OK. `rainyear` is fine, I imagine it resets once a year which shoudl be more than adequate. 'rainyear' is the internal field name that the driver uses and `rainYear` is the WeeWX mapped field name that follows the convention started by WeeWX with `dayRain`. The driver is using the `rainyear` field as `rain` is calculated before the fields are mapped.

You extractor functions are fine. WeeWX uses a default extractor function of average, so if we want to use something other than average we need to explicitly tell WeeWX. Hence all the add-in extractor functions for the driver are something other than average. WeeWX also sets the extractor function for a number of commonly used fields (`rain` being a pertinent case) and we don't need to specify an extractor for these fields.

You could run the driver directly with `--test-driver` but that will only show you the final field values and then only for loop packets (no WeeWX generated archive records) plus it won't show you any of the intermediate values. You could run WeeWX directly which will include archive records but not the intermediate values. Implementing `debug_rain` is only a handful of lines, I should have some time this afternoon to get it coded.

Limb, 2020-08-14:
I monitored the rain today and the weewx value mapped correctly to the driver (granted it was only 0.03 inches of rain fall). There is a good chance I was messing with weewx when this issue occured originally and I was throwing off it's calculation.

Now that I have a better understanding it looks like the driver starts off with a value of 0 and the "rain" value only updates for one loop packet any time there is change in the totalRain value, setting "rain" to the the difference. I was under the impression that a restart of weewx wouldn't impact this but since the driver has no previous value it makes sense that restarting weewx during a rainfall would cause this issue (weewx essentially misses a loop packet where "rain" would have been set).

I would say you hold off on coding this test, since I don't think this is an actual issue, it was operator error on my part. If I see this issue crop up again I can reopen the issue.

I apologise that this seems to have been a case of my misunderstanding of how weewx works and not an issue with the driver!

gjr80, 2020-08-14:
No problems. Yep, in a loop packet `rain` holds the total rainfall for the period covered by the loop packet (which is usually the period since the last loop packet was received). It will usually be 0 if no rain fell in that loop period but can be None if there was a WeeWX startup/restart (which is in effect as good as 0). Not many stations natively emit such a value, the Davis stations do but almost all others have to calculate it.

I'll probably add `debug_rain` (and `debug_wind`) anyway in slow time, issues with incorrect rain and wind often come up.

I'll close and as you said you can re-open if it recurs.

Limb, 2020-08-16:
I guess I spoke too soon :) The issue is still happening. Had a rainfall of 0.26 inches and weewx is reporting 0.16 inches. Today's rainfall has also been more steady then the other days small amount of rain.

I updated to the latest commit to enable `debug_rain = 2` and I'm getting the following outputs:

```
Aug 16 10:27:40 weewx weewx[24352] INFO gw1000: user.gw1000: parsed collector rain data: rainevent=6.4 rainrate=4.8 rainday=6.4 rainweek=6.4 rainmonth=16.3 rainyear=46.8
Aug 16 10:27:40 weewx weewx[24352] INFO gw1000: user.gw1000: calculate_rain: last_rain=46.7 new_total=46.8 calculated rain=46.8
Aug 16 10:27:40 weewx weewx[24352] INFO gw1000: user.gw1000: genLoopPackets: packet['rain']=0.09999999999999432 loop packet timestamped 1597588060
```

```
Aug 16 10:31:01 weewx weewx[24352] INFO gw1000: user.gw1000: parsed collector rain data: rainevent=6.5 rainrate=3.0 rainday=6.5 rainweek=6.5 rainmonth=16.4 rainyear=46.9
Aug 16 10:31:01 weewx weewx[24352] INFO gw1000: user.gw1000: calculate_rain: last_rain=46.8 new_total=46.9 calculated rain=46.9
Aug 16 10:31:01 weewx weewx[24352] INFO gw1000: user.gw1000: genLoopPackets: packet['rain']=0.10000000000000142 loop packet timestamped 1597588261
```

```
Aug 16 10:36:43 weewx weewx[24352] INFO gw1000: user.gw1000: parsed collector rain data: rainevent=6.6 rainrate=1.8 rainday=6.6 rainweek=6.6 rainmonth=16.5 rainyear=47.0
Aug 16 10:36:43 weewx weewx[24352] INFO gw1000: user.gw1000: calculate_rain: last_rain=46.9 new_total=47.0 calculated rain=47.0
Aug 16 10:36:43 weewx weewx[24352] INFO gw1000: user.gw1000: genLoopPackets: packet['rain']=0.10000000000000142 loop packet timestamped 1597588603
Aug 16 10:37:03 weewx weewx[24352] INFO gw1000: user.gw1000: parsed collector rain data: rainevent=6.6 rainrate=1.2 rainday=6.6 rainweek=6.6 rainmonth=16.5 rainyear=47.0
Aug 16 10:37:03 weewx weewx[24352] INFO gw1000: user.gw1000: calculate_rain: last_rain=47.0 new_total=47.0 calculated rain=47.0
Aug 16 10:37:03 weewx weewx[24352] INFO gw1000: user.gw1000: genLoopPackets: packet['rain']=0.0 loop packet timestamped 1597588623
```

Again, the data on weather underground matches the WS View app, so I do believe the driver is working correctly otherwise weather underground wouldn't match. I pulled the weewx database and summed the `rain` column for today and got the value that weewx is reporting. So I'm wondering if somehow weewx is not generating the records correctly, and somehow throwing out data from the driver? Looking at the database I do see some locations where record generation seems to have skipped (i have it set to every 20 seconds, and sometimes the next database entry is 40 seconds later instead of 20).

The following are my GW1000 settings and record generation settings, is it possible they are somehow conflicting? 

```
[GW1000]
    driver = user.gw1000
    ip_address = 10.100.1.24
    port = 45000
    poll_interval = 20
    debug_rain = 2
```

```
[StdArchive]
                                                                                                                                # If the station hardware supports data logging then the archive interval
    # will be downloaded from the station. Otherwise, specify it (in seconds).
    # archive_interval = 300
    archive_interval = 20
                                                                                                                                # If possible, new archive records are downloaded from the station                                                          # hardware. If the hardware does not support this, then new archive                                                         # records will be generated in software.
    # Set the following to "software" to force software record generation.
    record_generation = software                                                                                            
    # Whether to include LOOP data in hi/low statistics
    loop_hilo = True                                                                                                        
    # The data binding used to save archive records                                                                             data_binding = wx_binding
```

Does the driver support hardware record generation?

gjr80, 2020-08-17:
No problems, we should be able to track this down. As outlined previously the issue lies with the `rain` field. This field is produced by the GW1000 driver based on data from the GW1000 API. The accumulated `rain` data is processed and saved to database by WeeWX main. The issue could be WeeWX doing something odd with the accumulated `rain` data but I think is almost certainly not the case. The algorithm used by the GW1000 driver to calculate `rain` is standard across many drivers that works just fine. The likely source could be a logic error in the processing of the rain related data somewhere in the GW1000 driver. Some dodgy data from the GW1000 API could be a possibility but unlikely. Would be handy if I had an Ecowitt  rain gauge but unfortunately I don't.

One thing I did see in your `weewx.conf` is that you have `archive_interval = 20`. Is that a typo? That is very short, [`archive_interval` needs to be divisible by 60](http://weewx.com/docs/usersguide.htm#StdArchive). It also needs to be greater than `poll_interval` or you will have archive periods with no data. I don't think this is the cause of the rain problem but odd archive intervals have in the past caused some really odd behaviour.

I don't see anything else standing out in what you posted other than a mistake by me in some of the `debug_rain` code in b11. I have created a new branch on GitHub while I get `debug_rain` working properly. Could I get you to replace `gw1000.py` with this version:
```
$ wget https://raw.githubusercontent.com/gjr80/weewx-gw1000/debug_rain/bin/user/gw1000.py
```
the only change is one line that will alter the 'calculate_rain:' log entry.

Set `debug = 0` and leave `debug_rain = 2`. You will need to restart WeeWX of course. I'll leave it up to you as to how to test; either wait for some rain or make some artificially. Either way we need to see the error happen.

Would like to see the WeeWX startup portion of the log and of course the entries when the rain goes awry. I don't believe GitHub issue allow you to attach files, I am quite happy looking through long logs rather than pieces of cutting and pasting so feel free to email me the log if you wish gjroderick at gmail dot com.

Limb, 2020-08-18:
> One thing I did see in your `weewx.conf` is that you have `archive_interval = 20`. Is that a typo? That is very short, [`archive_interval` needs to be divisible by [...]

This would explain the missing records in my weewx db. I had changed the record generation to 20 seconds so that Weather Underground would get pushed data more often (the comments in the config file do not mention the divisible by 60 part!). I changed the interval back to the default of 300 and we had some heavy rainfall yesterday, and all three places including weewx showed the same values for total rainfall! I think this may have all been due to my misconfiguration. 

I did setup a log capture anyway so you could verify:
[weewx.syslog.txt](https://github.com/gjr80/weewx-gw1000/files/5090816/weewx.syslog.txt)

The first rainfall event starts at Aug 17 21:59:15.

gjr80, 2020-08-18:
That’s interesting, sounds like we can chalk up another one to odd or unusual archive intervals that cause some really odd/unexpected behaviour. Will have a word with Tom about changing the comments in `weewx.conf`. For a while it also needed to be whole minutes that were factors of 60, eg 60, 120, 180, 240, 300, 360, 600 but not 420, 480 etc. I will have a look at those logs later but I suspect if you are seeing everything agree the logs will just back that up and show it’s working fine.

Regards more frequent updates of WU, you can use WUs rapid fire to update WU on every loop packet (which in the case of the GW1000 driver is every `poll_interval` seconds). Rapid fire does have some quirks as well, but they are generally limited to what we call ‘partial packet stations’, good news is that stations using the GW1000 are not partial packet. Rapid fire is covered in the User’s Guide (http://www.weewx.com/docs/usersguide.htm#Wunderground)

gjr80, 2020-12-23:
Closing as it appears the issue has been solved. Can be re-opened if necessary.

closed 2020-12-23 by gjr80