gjr80/weewx-gw1000 #14: Solar Radiation and UV Values Incorrect

gary-hammer, 2020-08-26:
I have CumulusMX installed on a separate system.
WeeWX 4.1.1 and b11 of this driver installed on another system.

Both the console and CumulusMX report the same values for Radiation and UV Index.
WeeWX is showing a far lesser value for both.

Console values- Radiation 750 W/m2 UV Index 7
WeeWX values- Radiation 159 W/m² UV Index 1.0

What else is needed to investigate?

gjr80, 2020-08-26:
Could you run the gw1000 driver directly with `debug=3`, for a setup.py install:
```
$ PYTHONPATH=/home/weewx/bin python -m user.gw1000 --test-driver --debug=3
```
or for a package install:
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver --debug=3
```

No need to stop WeeWX if it is running, it should happily keep running. The above command will emit loop packets to console and should place a number of extra debug entries in the log. Could you let the driver run directly for a couple of loop packets whilst noting the radiation and UV readings from the display/Cumulus. Stop the driver running directly with ctrl C.

If you could post the console output, a log extract while the driver was run directly and the corresponding display/Cumulus values that should be all I need.

gary-hammer, 2020-08-26:
I will get the information you requested on Saturday.
Sun is down and I'm out of town starting tomorrow.

gjr80, 2020-08-27:
OK. Now that I am home I have sat down and had a look through the gw1000 driver code. I see nothing obviously wrong, the GW1000 emits two UV related obs, one that is called 'UV' and returned in microwatts/sq metre (which is not the same as WeeWX field `radiation`) and another called 'UVI' that is an index from 0-15. The former takes 2 bytes and the latter 1 byte. The default GW1000 driver field map maps GW1000 UVI to WeeWX field UV. There is really only one way to decode a single byte but I guess it is possible I am selecting the incorrect byte. Unfortunately I don't have anything connected to my GW1000 that provides UV/UVI. I do have a capture from another user that had a WH65 array that provided UV/UVI data but that is of limited use as I don't have the corresponding display reading to verify. That is where your data will help.

WeeWX field `radiation` is another issue; the GW1000 driver does not populate `radiation` nor does the `StdWXCalculate` service calculate it out of the box. Do you know where you are getting `radiation` from? I am aware of at least one WeeWX driver that applies a scaling factor to a light based observation in Lux to come up with `radiation`. Whilst the GW1000 driver does provide a field `luminosity` in Lux it is not mapped in any way to `radiation`.

A couple of other questions that might shed some light on things. Are you running any other non-standard or custom services under WeeWX? Have you changed the GW1000 field map with a `[[field_map]]` or `[[field_map_extensions]]` stanza under `[GW1000]` in `weewx.conf`? Finally what equipment/sensors do you have connected to the GW1000?

gary-hammer, 2020-08-29:
In answer to where does radiation come from, this was the solution given a while back on a different build. As I recall 'light' was replaced by 'luminosity'.

I'm attaching the data you requested, but today is rainy so if we need higher values for UV Index, I'll try again on a better day.
[DirOut.txt](https://github.com/gjr80/weewx-gw1000/files/5145941/DirOut.txt)

gjr80, 2020-08-29:
>In answer to where does radiation come from, this was the solution given a while back on a different build. As I recall 'light' was replaced by 'luminosity'.

OK, so we need to look at the luminosity value from the driver. I’m guessing Cumulus/station console display radiation **or** luminosity but not both? (`light` is an internal field used by the GW1000 driver, `luminosity` is the corresponding WeeWX field name)

> I'm attaching the data you requested

The data provided is exactly what I need but I also need a log extract from the same period, the log has the raw data received from the GW1000 API, it will be a lot of hex characters, eg further down in [this comment](https://github.com/gjr80/weewx-gw1000/issues/6#issuecomment-663968143). Could you also let me know what sensors are connected to/registered with the GW1000

gary-hammer, 2020-08-30:
Sorry, log attached.

I have the Ecowitt HP2551-C, GW1000, WS68, WH40, WH32-EP, WH32B
CumulusMX does not expose what is coming in from the GW1000 AFAIK. It does report/display Radiation.
[GW1000Log.txt](https://github.com/gjr80/weewx-gw1000/files/5146138/GW1000Log.txt)

gary-hammer, 2020-09-01:
WeeWX, CumulusMX, and my 2551-C all report the same solar values.
No idea why, I did install b12, but saw nothing in the change log.

At any rate, closing this issue.

closed 2020-09-01 by gary-hammer