gjr80/weewx-gw1000 #18: Battery levels for WH40 WH51 are zero

wardienet, 2020-10-16:
Firstly, this is brilliant driver so thanks! I have a WS80 multi, WH40 rain, WH32B indoor, WH41 AQ, WH51 soil, WH57 lightning. Below is sample output. I can see the battery levels coming through except the WH40 and WH51 seem to be zero's. The WS-View app is reporting good battery levels on everything as is the ecowitt.net alert monitoring (i.e. lack of alerts). Maybe a bug? The below is from first running test driver.

Output as per install instructions Step 5 test driver:
"2020-10-15 23:21:30 BST (1602800490): UV: 0, dateTime: 1602800490, dayRain: 0.3, daymaxwind: 7.0, inHumidity: 53, inTemp: 21.1, lightning_distance: None, lightning_last_det_time: None, lightning_strike_count: None, luminosity: 0.0, monthRain: 0.5, outHumidity: 97, outTemp: 7.4, pm2_5: 11.0, pm2_51_24hav: 8.1, pressure: 1012.7, rain: None, rainRate: 0.0, relbarometer: 1024.6, soilMoist1: 59, stormRain: 0.0, usUnits: 17, uvradiation: 0.0, weekRain: 0.5, wh40_batt: 0, wh41_ch1_batt: 4, wh51_ch1_batt: 0, wh57_batt: 5, windDir: 241, windGust: 0.8, windSpeed: 0.0, ws80_batt: 3.28, yearRain: 0.5" 

Also MQTT LOOP out I can see the same 0's e.g.:
beaufort_count: "0.571428571429"
cloudbase_meter: "161.465568243"
outHumidity: "97.2142857143"
pressure_mbar: "1012.66428571"
rain_cm: "0.0"
interval_minute: "5.0"
barometer_mbar: "1026.3257574"
yearRain_cm: "0.05"
windrun_km: "0.167143272575"
dewpoint_C: "6.95153953628"
ws80_batt: "3.28"
pm2_51_24hav: "8.3"
windGust_kph: "9.00002236942"
heatindex_C: "7.36428571429"
dayRain_cm: "0.03"
stormRain_cm: "0.0"
inDewpoint_C: "11.0476782125"
daymaxwind: "7.0"
wh51_ch1_batt: "0.0"
weekRain: "0.5"
altimeter_mbar: "1026.03501945"
windchill_C: "7.26328516716"
appTemp_C: "6.26522101744"
outTemp_C: "7.36428571429"
windGustDir: "245.0"
pm2_5_microgram_per_meter_cubed: "11.0"
lightning_strike_count: "0.0"
humidex_C: "7.37520274268"
wh57_batt: "5.0"
soilMoist1_centibar: "59.0"
rain24_cm: "0.0381"
wh41_ch1_batt: "4.0"
hourRain_cm: "0.0"
inTemp_C: "21.0"
windSpeed_kph: "2.0057192709"
usUnits: "16.0"
uvradiation: "0.0"
wh40_batt: "0.0"
monthRain_cm: "0.05"
luminosity: "0.0"
UV: "0.0"
rainRate_cm_per_hour: "0.0"
dateTime: "1602801600.0"
windDir: "345.229304126"
relbarometer: "1024.56428571"
ET_cm: "0.0"
inHumidity: "53.0"

gjr80, 2020-10-16:
What you are seeing is expected behaviour. The sensors used with the GW1000 report battery state in one of four ways; some report 0 or 1 (0 being OK and 1 being low), some report a value from 0 to 5 (2-5 being OK and 0 or 1 being low), some report battery voltage and I have one that reports 0 to 6 (same as 0 to 5 but 6 indicates external DC power). Refer to the [Sensor battery states](https://github.com/gjr80/weewx-gw1000/wiki/Sensor-battery-states) wiki entry.

The WH40 and WH50 both use the 0 or 1 battery state reporting.

The GW1000 driver reports these values as is rather than interpreting them, that is left the reporting layer (for example the Seasons skin sensor status panel).

wardienet, 2020-10-16:
> What you are seeing is expected behaviour. The sensors used with the GW1000 report battery state in one of four ways; some report 0 or 1 (0 being OK and 1 being low), some report a value from 0 to 5 [...]

Many thanks - my bad for not finding the wiki doc. Very clear and really appreciate you documenting all that. If you want me to test anything with my setup, give me a shout e.g. the WH32B. I'm also about to set up a couple of WH31's too.

closed 2020-10-16 by wardienet