gjr80/weewx-gw1000 #19: Not reading solar radiation properly

Distortionist, 2020-10-23:
I am not getting the solar radiation reading, just UV index.

Distortionist, 2020-10-23:
you have a couple of errors in some radiation. I corrected them and now I get radiation display, except it is wrong. I can't figure it out. Super low. Please correct the errors with solar radiation. Maybe it is because I am getting readings from a WS-2000 but the readings are correct on the WS-2000 console and the GW1000, something gets lost on the way.

gjr80, 2020-10-23:
Do you think you might share some details? Impossible to do anything based upon what information you have provided so far.

Distortionist, 2020-10-23:
Sorry, I am so new to this and everything is a learning curve right now.

I was getting empty solar radiation readings which I usually get in lux on my console, but I can set it to m/w2 and fc. While trying to figure that out, it was error in mappings. UV was supposed to be uv in small case, and uvRadiation was supposed to be uvradiation. I couldn't figure out where to edit the values but when I uncompressed your installation tar.gz i found the install script with the wrong values in there so I corrected them.  Now I get my radiation readings and UV index. in weewx it shows up in w/m2, so when I switch the console readings to w/m2 it is about 4 times higher than what weewx is reporting, when it was like 200w/m2 it was reading something like 50 in weewx. I have yet to figure out how to fix it. I have been searching google and I have tried setting a few corrections but I can't see them making any difference even when I used extreme numbers. I must be doing something wrong trying to correct it.  I was hoping you would know what is going on and patch it for me, or tell me how to fix it.

I am sorry I am not very clear, I have been swamped in my learning curve. I will do my best to explain more if you need it. I already deleted the tar.gz so I can't look up specifics. I just know I corrected 2 uv and radiation related things in the mappings and it started working with wrong radiation values, uv index seems correct.

gjr80, 2020-10-24:
OK. A few things.

If you have field mapping issues then the preferred way to deal with that is to edit/add field map settings in `weewx.conf` under `[GW1000]`, if you start editing the code you may have success in the short term but chances are your changes will be wiped next time you update the driver.

That being said I suggest we forget about mappings for the present and focus on the data from the GW1000. Can you try running the GW1000 driver directly using one of the following commands:
```
$ PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --live-data --debug=3
```
or
```
$ PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --live-data --debug=3
```
Use the first command if you installed WeeWX using `setup.py` or use the second command if you installed WeeWX using one of the packages. Also, if you are running WeeWX under python2 you will need to change `python3` to `python2` in the above commands. there is no need to stop WeeWX to run these commands but it may make it a little easier to extract information from the WeeWX log if WeeWX is not running.

Once the appropriate variant of the above commands is run you should see some output on the console similar to this:
```
Using configuration file /home/weewx/weewx.conf
debug level is '3'

IP address to be obtained by discovery
Port number obtained from station config

Interrogating GW1000 at 192.168.2.32:45000

GW1000 live sensor data: absbarometer: 1009.1, datetime: 1603498246, humid1: 66, humid2: 58, inhumid: 55, intemp: 28.6, lightningcount: 0, lightningdettime: None, lightningdist: None, outhumid: 63, outtemp: 26.1, pm251: 10.0, pm251_24hav: 9.3, relbarometer: 1009.1, soilmoist1: 25, soilmoist2: 18, temp1: 28.9, temp2: 23.9, wh26_batt: 0, wh26_sig: 4, wh31_ch1_batt: 0, wh31_ch1_sig: 4, wh31_ch2_batt: 0, wh31_ch2_sig: 4, wh41_ch1_batt: 6, wh41_ch1_sig: 4, wh51_ch1_batt: 0, wh51_ch1_sig: 4, wh51_ch2_batt: 0, wh51_ch2_sig: 4, wh51_ch3_batt: 0, wh51_ch3_sig: 0, wh57_batt: 5, wh57_sig: 4
```

Take a copy of this output and post it back here.

You should also see some entries in the WeeWX log similar to these:
```
Oct 24 10:10:43 buster20 weewx[759] DEBUG __main__: Sending broadcast packet 'FF FF 12 03 15' to '255.255.255.255:46000'
Oct 24 10:10:43 buster20 weewx[759] DEBUG __main__: Received broadcast response 'FF FF 12 00 20 50 02 91 E3 FD 32 C0 A8 02 20 AF C8 0F 47 57 31 30 30 30 2D 57 49 46 49 46 44 33 32 C1'
Oct 24 10:10:43 buster20 weewx[759] DEBUG __main__: Received broadcast response 'FF FF 12 00 20 DC 4F 22 58 B7 FF C0 A8 02 1F AF C8 0F 47 57 31 30 30 30 2D 57 49 46 49 42 37 46 46 3C'
Oct 24 10:10:45 buster20 weewx[759] INFO __main__: Multiple GW1000 were found at 192.168.2.32:45000, 192.168.2.31:45000
Oct 24 10:10:45 buster20 weewx[759] DEBUG __main__: Sending packet 'FF FF 26 03 29' to '192.168.2.32:45000'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Received response 'FF FF 26 09 50 02 91 E3 FD 32 24'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Sending packet 'FF FF 30 03 33' to '192.168.2.32:45000'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Received response 'FF FF 30 0B 00 00 5F 93 FD A5 5E 01 2E'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Sending packet 'FF FF 27 03 2A' to '192.168.2.32:45000'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Received response 'FF FF 27 00 45 01 01 1E 06 37 08 27 6B 09 27 6B 02 01 05 07 3F 2A 00 64 4D 00 5D 2C 19 2E 12 1A 01 21 22 42 1B 00 EF 23 3A 4C 06 00 00 00 05 FF FF FF F6 FF FF FF FF FF FF FF 62 00 00 00 00 61 FF FF FF FF 60 FF 71'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: sensor data is '01 01 1E 06 37 08 27 6B 09 27 6B 02 01 05 07 3F 2A 00 64 4D 00 5D 2C 19 2E 12 1A 01 21 22 42 1B 00 EF 23 3A 4C 06 00 00 00 05 FF FF FF F6 FF FF FF FF FF FF FF 62 00 00 00 00 61 FF FF FF FF 60 FF'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Parsed data: {'intemp': 28.6, 'inhumid': 55, 'absbarometer': 1009.1, 'relbarometer': 1009.1, 'outtemp': 26.1, 'outhumid': 63, 'pm251': 10.0, 'pm251_24hav': 9.3, 'soilmoist1': 25, 'soilmoist2': 18, 'temp1': 28.9, 'humid1': 66, 'temp2': 23.9, 'humid2': 58, 'wh40_batt': 0, 'wh26_batt': 0, 'wh25_batt': 0, 'wh24_batt': 0, 'wh31_ch1_batt': 0, 'wh31_ch2_batt': 0, 'wh31_ch3_batt': 0, 'wh31_ch4_batt': 0, 'wh31_ch5_batt': 0, 'wh31_ch6_batt': 0, 'wh31_ch7_batt': 0, 'wh31_ch8_batt': 0, 'wh51_ch1_batt': 0, 'wh51_ch2_batt': 0, 'wh51_ch3_batt': 0, 'wh51_ch4_batt': 0, 'wh51_ch5_batt': 0, 'wh51_ch6_batt': 0, 'wh51_ch7_batt': 0, 'wh51_ch8_batt': 0, 'wh51_ch9_batt': 0, 'wh51_ch10_batt': 0, 'wh51_ch11_batt': 0, 'wh51_ch12_batt': 0, 'wh51_ch13_batt': 0, 'wh51_ch14_batt': 0, 'wh51_ch15_batt': 0, 'wh51_ch16_batt': 0, 'wh57_batt': 5, 'wh68_batt': 5.1000000000000005, 'ws80_batt': 5.1000000000000005, 'wh41_ch1_batt': 6, 'wh41_ch2_batt': None, 'wh41_ch3_batt': None, 'wh41_ch4_batt': None, 'wh55_ch1_batt': None, 'wh55_ch2_batt': None, 'wh55_ch3_batt': None, 'wh55_ch4_batt': None, 'lightningcount': 0, 'lightningdettime': None, 'lightningdist': None, 'datetime': 1603498246}
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Sending packet 'FF FF 3A 03 3D' to '192.168.2.32:45000'
Oct 24 10:10:46 buster20 weewx[759] DEBUG __main__: Received response 'FF FF 3A D5 00 FF FF FF FE 00 00 01 FF FF FF FE FF 00 02 FF FF FF FE FF 00 03 FF FF FF FE 00 00 05 00 00 00 E4 00 04 06 00 00 00 5B 00 04 07 00 00 00 BE 00 04 08 FF FF FF FE 00 00 09 FF FF FF FE 00 00 0A FF FF FF FE 00 00 0B FF FF FF FE 00 00 0C FF FF FF FE 00 00 0D FF FF FF FE 00 00 0E 00 00 CD 19 00 04 [... cut ...] tt': 0, 'wh51_ch2_batt': 0, 'wh51_ch3_batt': 0, 'wh57_batt': 5, 'wh41_ch1_batt': 6, 'lightningcount': 0, 'lightningdettime': None, 'lightningdist': None, 'datetime': 1603498246, 'wh26_sig': 4, 'wh31_ch1_sig': 4, 'wh31_ch2_sig': 4, 'wh51_ch1_sig': 4, 'wh51_ch2_sig': 4, 'wh51_ch3_sig': 0, 'wh41_ch1_sig': 4, 'wh57_sig': 4}
```

Please take copy of these log entries and post them back here as well.

Please post as text and not as images of a screenshot, screenshots are difficult to read and analyse.

Distortionist, 2020-10-25:
Thank you for your response, here is the result of the command above, I
think you would rather me run it during sunshine, I will do it again
tomorrow, here is the night time result though if it helps at all.

GW1000 live sensor data: absbarometer: 997.2, datetime: 1603586667,
daymaxwind: 14.8, gustspeed: 2.6, humid1: 44, humid2: 33, humid3: 46,
humid4: 82, humid5: 60, humid6: 60, humid7: 56, inhumid: 44, intemp: 28.4,
leak1: 0, leak2: 0, light: 0.0, outhumid: 99, outtemp: 12.2, rainday: 0.0,
rainevent: 0.0, rainmonth: 44.6, rainrate: 0.0, rainweek: 2.3, rainyear:
44.6, relbarometer: 1008.1, temp1: 22.1, temp2: 23.4, temp3: 18.8, temp4:
15.7, temp5: 18.8, temp6: 23.0, temp7: 24.0, temp8: 19.5, uv: 0.1, uvi: 0,
wh31_ch1_batt: 0, wh31_ch2_batt: 0, wh31_ch3_batt: 0, wh31_ch4_batt: 0,
wh31_ch5_batt: 0, wh31_ch6_batt: 0, wh31_ch7_batt: 0, wh31_ch8_batt: 0,
wh55_ch1_batt: 5, wh55_ch2_batt: 5, wh65_batt: 0, winddir: 305, windspeed:
2.4

Here is the output of weewx log.

Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Sending packet
'FF FF 26 03 29' to '192.168.1.250:45000'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Received
response 'FF FF 26 09 8C AA B5 C6 75 A2 F7'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Sending packet
'FF FF 30 03 33' to '192.168.1.250:45000'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Received
response 'FF FF 30 0B 02 01 5F 94 A2 11 19 01 FE'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Sending packet
'FF FF 27 03 2A' to '192.168.1.250:45000'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Received
response 'FF FF 27 00 7B 01 01 10 06 2F 08 26 F4 09 27 61 02 00 7A 07 63 0A
01 33 0B 00 1E 0C 00 24 15 00 00 00 00 16 00 01 17 00 1A 00 DD 22 2C 1B 00
EA 23 21 1C 00 BC 24 2E 1D 00 9D 25 52 1E 00 BA 26 3D 1F 00 E6 27 3C 20 00
F0 28 38 21 00 C3 4C 0F 00 00 00 FF FF FF FF FF FF 05 05 FF FF FF FF 58 00
59 00 19 00 94 0E 00 00 10 00 00 11 00 17 12 00 00 01 BE 13 00 00 01 BE 0D
00 00 EB'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: sensor data is
'01 01 10 06 2F 08 26 F4 09 27 61 02 00 7A 07 63 0A 01 33 0B 00 1E 0C 00 24
15 00 00 00 00 16 00 01 17 00 1A 00 DD 22 2C 1B 00 EA 23 21 1C 00 BC 24 2E
1D 00 9D 25 52 1E 00 BA 26 3D 1F 00 E6 27 3C 20 00 F0 28 38 21 00 C3 4C 0F
00 00 00 FF FF FF FF FF FF 05 05 FF FF FF FF 58 00 59 00 19 00 94 0E 00 00
10 00 00 11 00 17 12 00 00 01 BE 13 00 00 01 BE 0D 00 00'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Parsed data:
{'intemp': 27.2, 'inhumid': 47, 'absbarometer': 997.2, 'relbarometer':
1008.1, 'outtemp': 12.2, 'outhumid': 99, 'winddir': 307, 'windspeed': 3.0,
'gustspeed': 3.6, 'light': 0.0, 'uv': 0.1, 'uvi': 0, 'temp1': 22.1,
'humid1': 44, 'temp2': 23.4, 'humid2': 33, 'temp3': 18.8, 'humid3': 46,
'temp4': 15.7, 'humid4': 82, 'temp5': 18.6, 'humid5': 61, 'temp6': 23.0,
'humid6': 60, 'temp7': 24.0, 'humid7': 56, 'temp8': 19.5, 'wh40_batt': 0,
'wh26_batt': 0, 'wh25_batt': 0, 'wh65_batt': 0, 'wh31_ch1_batt': 0,
'wh31_ch2_batt': 0, 'wh31_ch3_batt': 0, 'wh31_ch4_batt': 0,
'wh31_ch5_batt': 0, 'wh31_ch6_batt': 0, 'wh31_ch7_batt': 0,
'wh31_ch8_batt': 0, 'wh51_ch1_batt': 0, 'wh51_ch2_batt': 0,
'wh51_ch3_batt': 0, 'wh51_ch4_batt': 0, 'wh51_ch5_batt': 0,
'wh51_ch6_batt': 0, 'wh51_ch7_batt': 0, 'wh51_ch8_batt': 0,
'wh51_ch9_batt': 0, 'wh51_ch10_batt': 0, 'wh51_ch11_batt': 0,
'wh51_ch12_batt': 0, 'wh51_ch13_batt': 0, 'wh51_ch14_batt': 0,
'wh51_ch15_batt': 0, 'wh51_ch16_batt': 0, 'wh57_batt': 255, 'wh68_batt':
5.1000000000000005, 'ws80_batt': 5.1000000000000005, 'wh41_ch1_batt': None,
'wh41_ch2_batt': None, 'wh41_ch3_batt': None, 'wh41_ch4_batt': None,
'wh55_ch1_batt': 5, 'wh55_ch2_batt': 5, 'wh55_ch3_batt': None,
'wh55_ch4_batt': None, 'leak1': 0, 'leak2': 0, 'daymaxwind': 14.8,
'rainrate': 0.0, 'rainday': 0.0, 'rainweek': 2.3, 'rainmonth': 44.6,
'rainyear': 44.6, 'rainevent': 0.0, 'datetime': 1603587138}
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Sending packet
'FF FF 3A 03 3D' to '192.168.1.250:45000'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Received
response 'FF FF 3A D5 00 00 00 00 96 00 03 01 FF FF FF FF FF 00 02 FF FF FF
FF FF 00 03 FF FF FF FF 00 00 05 FF FF FF FF 00 00 06 00 00 00 C6 00 04 07
00 00 00 40 00 04 08 00 00 00 96 00 04 09 00 00 00 D9 00 04 0A 00 00 00 07
00 04 0B 00 00 00 10 00 04 0C 00 00 00 EF 00 04 0D 00 00 00 E4 00 04 0E FF
FF FF FF 00 00 0F FF FF FF FF 00 00 10 FF FF FF FF 00 00 11 FF FF FF FF 00
00 12 FF FF FF FF 00 00 13 FF FF FF FF 00 00 14 FF FF FF FF 00 00 15 FF FF
FF FF 00 00 16 FF FF FF FF 0F 00 17 FF FF FF FE 0F 00 18 FF FF FF FE 0F 00
19 FF FF FF FE 0F 00 1A FF FF FF FF FF 00 1B 00 00 D8 7A 05 04 1C 00 10 D8
58 05 04 1D FF FF FF FF FF 00 1E FF FF FF FF FF 00 80'
Oct 24 21:52:18 LivingstoneCove weewx[28349] DEBUG __main__: Filtered
parsed data: {'intemp': 27.2, 'inhumid': 47, 'absbarometer': 997.2,
'relbarometer': 1008.1, 'outtemp': 12.2, 'outhumid': 99, 'winddir': 307,
'windspeed': 3.0, 'gustspeed': 3.6, 'light': 0.0, 'uv': 0.1, 'uvi': 0,
'temp1': 22.1, 'humid1': 44, 'temp2': 23.4, 'humid2': 33, 'temp3': 18.8,
'humid3': 46, 'temp4': 15.7, 'humid4': 82, 'temp5': 18.6, 'humid5': 61,
'temp6': 23.0, 'humid6': 60, 'temp7': 24.0, 'humid7': 56, 'temp8': 19.5,
'wh65_batt': 0, 'wh31_ch1_batt': 0, 'wh31_ch2_batt': 0, 'wh31_ch3_batt': 0,
'wh31_ch4_batt': 0, 'wh31_ch5_batt': 0, 'wh31_ch6_batt': 0,
'wh31_ch7_batt': 0, 'wh31_ch8_batt': 0, 'wh55_ch1_batt': 5,
'wh55_ch2_batt': 5, 'leak1': 0, 'leak2': 0, 'daymaxwind': 14.8, 'rainrate':
0.0, 'rainday': 0.0, 'rainweek': 2.3, 'rainmonth': 44.6, 'rainyear': 44.6,
'rainevent': 0.0, 'datetime': 1603587138}

gjr80, 2020-10-25:
Yes, that is exactly what I am after but when there is sun shining and you have some non-zero solar radiation and uv readings. Would also help if you could provide the corresponding readings from your station display/console when you run the command.

Distortionist, 2020-10-25:
Alright, its pretty crappy outside but I got some readings.
Weewx is reporting 4W/m2 while the consoles are reporting 37.2 (the GW1000
feeding the information to weewx)
WS-2000 console is reporting 37.19.

Here is the output from a minute before taking the pictures

GW1000 live sensor data: absbarometer: 1002.4, datetime: 1603627327,
daymaxwind: 21.9, gustspeed: 15.8, humid1: 45, humid2: 35, humid3: 48,
humid4: 82, humid5: 58, humid6: 59, humid7: 53, inhumid: 40, intemp: 27.5,
leak1: 0, leak2: 0, light: 4774.0, outhumid: 75, outtemp: 6.6, rainday:
0.0, rainevent: 0.0, rainmonth: 44.6, rainrate: 0.0, rainweek: 0.0,
rainyear: 44.6, relbarometer: 1013.3, temp1: 22.1, temp2: 23.4, temp3:
18.7, temp4: 15.2, temp5: 18.0, temp6: 21.1, temp7: 22.5, temp8: 19.3, uv:
4.0, uvi: 0, wh31_ch1_batt: 0, wh31_ch2_batt: 0, wh31_ch3_batt: 0,
wh31_ch4_batt: 0, wh31_ch5_batt: 0, wh31_ch6_batt: 0, wh31_ch7_batt: 0,
wh31_ch8_batt: 0, wh55_ch1_batt: 5, wh55_ch2_batt: 5, wh65_batt: 0,
winddir: 333, windspeed: 14.3

Here is weewx

Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Sending packet
'FF FF 26 03 29' to '192.168.1.250:45000'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Received
response 'FF FF 26 09 8C AA B5 C6 75 A2 F7'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Sending packet
'FF FF 30 03 33' to '192.168.1.250:45000'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Received
response 'FF FF 30 0B 02 01 5F 95 3F 0E 19 01 99
'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Sending packet
'FF FF 27 03 2A' to '192.168.1.250:45000'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Received
response 'FF FF 27 00 7B 01 01 13 06 28 08 27 28
 09 27 95 02 00 42 07 4B 0A 01 4D 0B 00 8F 0C 00 9E 15 00 00 BA 7C 16 00 28
17 00 1A 00 DD 22 2D 1B 00 EA 23 23 1C 00
 BB 24 30 1D 00 98 25 52 1E 00 B4 26 3A 1F 00 D3 27 3B 20 00 E1 28 35 21 00
C1 4C 0F 00 00 00 FF FF FF FF FF FF 05 05
 FF FF FF FF 58 00 59 00 19 00 DB 0E 00 00 10 00 00 11 00 00 12 00 00 01 BE
13 00 00 01 BE 0D 00 00 60'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: sensor data is
'01 01 13 06 28 08 27 28 09 27 95 02 00 42
 07 4B 0A 01 4D 0B 00 8F 0C 00 9E 15 00 00 BA 7C 16 00 28 17 00 1A 00 DD 22
2D 1B 00 EA 23 23 1C 00 BB 24 30 1D 00 98
 25 52 1E 00 B4 26 3A 1F 00 D3 27 3B 20 00 E1 28 35 21 00 C1 4C 0F 00 00 00
FF FF FF FF FF FF 05 05 FF FF FF FF 58 00
 59 00 19 00 DB 0E 00 00 10 00 00 11 00 00 12 00 00 01 BE 13 00 00 01 BE 0D
00 00'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Parsed data:
{'intemp': 27.5, 'inhumid': 40, 'absbaromete
r': 1002.4, 'relbarometer': 1013.3, 'outtemp': 6.6, 'outhumid': 75,
'winddir': 333, 'windspeed': 14.3, 'gustspeed': 1
5.8, 'light': 4774.0, 'uv': 4.0, 'uvi': 0, 'temp1': 22.1, 'humid1': 45,
'temp2': 23.4, 'humid2': 35, 'temp3': 18.7, '
humid3': 48, 'temp4': 15.2, 'humid4': 82, 'temp5': 18.0, 'humid5': 58,
'temp6': 21.1, 'humid6': 59, 'temp7': 22.5, 'h
umid7': 53, 'temp8': 19.3, 'wh40_batt': 0, 'wh26_batt': 0, 'wh25_batt': 0,
'wh65_batt': 0, 'wh31_ch1_batt': 0, 'wh31_
ch2_batt': 0, 'wh31_ch3_batt': 0, 'wh31_ch4_batt': 0, 'wh31_ch5_batt': 0,
'wh31_ch6_batt': 0, 'wh31_ch7_batt': 0, 'wh
31_ch8_batt': 0, 'wh51_ch1_batt': 0, 'wh51_ch2_batt': 0, 'wh51_ch3_batt':
0, 'wh51_ch4_batt': 0, 'wh51_ch5_batt': 0,
'wh51_ch6_batt': 0, 'wh51_ch7_batt': 0, 'wh51_ch8_batt': 0,
'wh51_ch9_batt': 0, 'wh51_ch10_batt': 0, 'wh51_ch11_batt'
: 0, 'wh51_ch12_batt': 0, 'wh51_ch13_batt': 0, 'wh51_ch14_batt': 0,
'wh51_ch15_batt': 0, 'wh51_ch16_batt': 0, 'wh57_b
att': 255, 'wh68_batt': 5.1000000000000005, 'ws80_batt':
5.1000000000000005, 'wh41_ch1_batt': None, 'wh41_ch2_batt':
None, 'wh41_ch3_batt': None, 'wh41_ch4_batt': None, 'wh55_ch1_batt': 5,
'wh55_ch2_batt': 5, 'wh55_ch3_batt': None, 'w
h55_ch4_batt': None, 'leak1': 0, 'leak2': 0, 'daymaxwind': 21.9,
'rainrate': 0.0, 'rainday': 0.0, 'rainweek': 0.0, 'r
ainmonth': 44.6, 'rainyear': 44.6, 'rainevent': 0.0, 'datetime': 1603627327}
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Sending packet
'FF FF 3A 03 3D' to '192.168.1.250:45000'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Received
response 'FF FF 3A D5 00 00 00 00 96 00 04 01 FF
 FF FF FF FF 00 02 FF FF FF FF FF 00 03 FF FF FF FF 00 00 05 FF FF FF FF 00
00 06 00 00 00 C6 00 04 07 00 00 00 40 00
 04 08 00 00 00 96 00 04 09 00 00 00 D9 00 04 0A 00 00 00 07 00 04 0B 00 00
00 10 00 04 0C 00 00 00 EF 00 04 0D 00 00
 00 E4 00 04 0E FF FF FF FF 00 00 0F FF FF FF FF 00 00 10 FF FF FF FF 00 00
11 FF FF FF FF 00 00 12 FF FF FF FF 00 00
 13 FF FF FF FF 00 00 14 FF FF FF FF 00 00 15 FF FF FF FF 00 00 16 FF FF FF
FF 0F 00 17 FF FF FF FE 0F 00 18 FF FF FF
 FE 0F 00 19 FF FF FF FE 0F 00 1A FF FF FF FF FF 00 1B 00 00 D8 7A 05 04 1C
00 10 D8 58 05 04 1D FF FF FF FF FF 00 1E
 FF FF FF FF FF 00 81'
Oct 25 09:02:07 LivingstoneCove weewx[3927] DEBUG __main__: Filtered parsed
data: {'intemp': 27.5, 'inhumid': 40, 'ab
sbarometer': 1002.4, 'relbarometer': 1013.3, 'outtemp': 6.6, 'outhumid':
75, 'winddir': 333, 'windspeed': 14.3, 'gust
speed': 15.8, 'light': 4774.0, 'uv': 4.0, 'uvi': 0, 'temp1': 22.1,
'humid1': 45, 'temp2': 23.4, 'humid2': 35, 'temp3'
: 18.7, 'humid3': 48, 'temp4': 15.2, 'humid4': 82, 'temp5': 18.0, 'humid5':
58, 'temp6': 21.1, 'humid6': 59, 'temp7':
 22.5, 'humid7': 53, 'temp8': 19.3, 'wh65_batt': 0, 'wh31_ch1_batt': 0,
'wh31_ch2_batt': 0, 'wh31_ch3_batt': 0, 'wh31
_ch4_batt': 0, 'wh31_ch5_batt': 0, 'wh31_ch6_batt': 0, 'wh31_ch7_batt': 0,
'wh31_ch8_batt': 0, 'wh55_ch1_batt': 5, 'w
h55_ch2_batt': 5, 'leak1': 0, 'leak2': 0, 'daymaxwind': 21.9, 'rainrate':
0.0, 'rainday': 0.0, 'rainweek': 0.0, 'rain
month': 44.6, 'rainyear': 44.6, 'rainevent': 0.0, 'datetime': 1603627327}

Distortionist, 2020-10-25:
Pics of readings.

Distortionist, 2020-10-25:
'luminosity': 'light',
        'radiation': 'uv',
        'UV': 'uvi',

That is the problematic area in gw1000,py those are after me editing to get
some readings. I got the UV index right but something is wrong somewhere.
I tried to look up the previous values but it looks like you updated the
driver in github? either that or i am losing my mind.

gjr80, 2020-10-26:
Thank you. As far as I can tell the driver is correctly identifying and decoding the light an UV data. The problem appears to be the mapping that you have introduced. 

A little background. The GW1000API provides three light/UV related observations. These are what Ecowitt refers to as 'Light', 'UV' and 'UVI'. 'LIGHT' is a value in Lux, 'UV' is a value in microwatts per square metre and 'UVI' is an index in the range 0-15. Of note there is no solar irradiance (what WeeWX refers to as field 'radiation') data. Ecowitt uses a common approximation for solar radiation in Watts per square metre by dividing the 'LIGHT' value by 126.7, that is what you are seeing on your console. Note that GW1000 'UV' value is in microwatts per square metre but solar irradiance is measured in Watts per square metre, the 'UV' value from the GW1000 API is not solar irradiance (aka 'radiation').

The WeeWX approach to drivers is that the driver obtains and decodes data from the station, it does not derive data. So since the GW1000 API does not provide solar irradiance data (even though Ecowitt consoles display it) the GW1000 driver does not calculate WeeWX field 'radiation'. Rather those users that wish to use the Ecowitt approximation should implement the calculation in the WeeWX `StdCalibrate` service. Something like the following in `weewx.conf`:

```
[StdCalibrate]
    [[Corrections]]
        radiation = luminosity/126.7 if luminosity is not None else None
```
You might want to read through issue #6 

The mapping you are using now incorrectly maps the GW1000 'UV' value to the WeeWX field 'radiation', this needs to be removed and the mapping returned to the default. If you modified python code I suggest you undo the change for replace the file with the distributed version. If you altered the mapping by making a change to `weewx.conf` then that change will need to be removed.

Distortionist, 2020-10-26:
Thank you for your response.
I spend a good chunk of yesterday trying to figure out what is going on. I
have come to the conclusion the driver I used was an older one I used
following outdated documentation somewhere.
All the changes I did to the installation script are applied in the newest
version. I had setup weewx with a simulator driver while i figured and got
everything ready, I installed the GW1000 driver before I got my station
just to test it but kept using the simulation.

Yesterday, I downloaded the current version and reinstalled it, but I see
no difference. The changes I made were following instructions in from the
weewx google group.  The correction should fix it and in the forums they
seem to work but when I tried it did not work.
I think I used light and luminosity but I could be mistaken, I tried
several entries so I will start over today and try.

I am trying before sending this email and I think I found out why it did
not work, I think my error is found!

So before contacting you I have read all that stuff, and I tried doing the
correction and it failed, I know why now. I first tried radiation =
light/126.7 if light is not None esle None, then I tried replacing light
with luminosity, except I forgot to change it in the second part "if
luminosity is not None else None".
My readings are much closer now! Thank you so much for helping me hunt this
down. I really appreciate it, and in the end, it was my error. Is there any
way you can include the correction values in the driver install script for
others? I know I have been trying to figure this out for a while, other
newbs like me may have a hard time too. Maybe at least include it in the
documentation, but for now I am very happy and grateful for your driver and
help. Thank you.

I have minor questions, so if we get the radiation from light, and uv index
from uvi, why is the uv value there and so far off from the calculated
result from light in w/m2? Also why is it .1 to .3 for me during night time
while light is 0? It can't be the moonlight if light is 0 right?

We for now this is fixed but I would really love to understand those
values, especially the problematic uv one that gives me numbers at night,
and if its supposed to be a w/m2 value why is it so far off from the one
calculated from light?

Thank you for your time man, I honestly appreciate it. I've spent hours
hunting this down, it may be easy for some, but everything here is new to
me and the learning curve is a bit challenging.

gjr80, 2020-10-26:
Good that you have it sorted. I’m not prepared to put anything to do with deriving `radiation` in the driver or the installer. I will though create an entry in the driver wiki.

Regards the Ecowitt `UV` and the `radiation` value, they are significantly different because they measure two different types of radiation. `UV` measures the intensity of the ultraviolet radiation whereas `radiation` is a measure of the intensity of the sunlight falling on the surface of the Earth. `radiation` includes ultraviolet, visible and infrared portions of the spectrum. That is why the both use similar units (power over surface area) but the UV value is much lower than `radiation`.

Why provide a UV radiation value at all? I don’t know, I suspect most folks will use the UV index rather than a value in microwatts per square metre. It may be a case of it is available so why not provide it.

Why do you have a value at night? Short answer is I don’t know. I am not familiar with the Ecowitt sensors. It may be that Ecowitt has made some compromises in the type of sensor they are using and it may have some sensitivity to some residual light, remember this a consumer grade station so there will be compromises. If it particularly bothers you you could probably add another correction to `StdCalibrate` that zeros the value when say luminosity is zero.

Distortionist, 2020-10-26:
That makes sense. What doesn't is why provide the UV radiation value at
all, because it is not displayed anywhere, not on the WS-2000 console, not
on the GW1000 app, or the website, it is just there. The sensor is actually
from an Ambient Weather WS-2000. All 3 readings come from the same senor
and if lhe light includes the uv radiation and it is 0, it is weird that it
is spitting out a value still. All readings can be calibrated from the
WS-2000 console and the GW1000 app, except that one!

I don't think I need to 0 it out anymore because now I get the proper
reading. I believe the uv reading is now being ignored, and only light and
uvi are in use!

Thank you so much for your help. I appreciate it, and your explanation for
these values. You are awesome.

closed 2020-10-27 by gjr80