gjr80/weewx-gw1000 #20: Unable to add pm2_51_24hav to graph.

Distortionist, 2020-12-22:
What is wrong with this? It doesn't work and gives an error. I am trying to graph the PM 2.5 24H average along with the current reading. This is what I have in my graphs.conf

`        [[chart7]]
             title = PM 2.5
             [[[pm2_5]]]
                name = PM 2.5
                zIndex = 1
                color = "#ffc83f"
              [[[pm2_51_24hav]]]
                 name = 24H Average
                 color = "#f7f2b4"
                 y_label = "ug/m3"

`

This is the reading from the driver :

`2020-12-21 20:15:48 AST (1608596148): UV: 0, dateTime: 1608596148, dayRain: 13.2, daymaxwind: 15.8, extraHumid1: 53, extraHumid2: 48, extraHumid3: 54, extraHumid4: 75, extraHumid5: 54, extraHumid6: 45, extraHumid7: 43, extraTemp1: 23.3, extraTemp2: 24.3, extraTemp3: 21.1, extraTemp4: 9.4, extraTemp5: 16.9, extraTemp6: 25.4, extraTemp7: 25.9, extraTemp8: 18.9, inHumidity: 39, inTemp: 27.8, leak1: 0, leak2: 0, lightning_distance: None, lightning_last_det_time: None, lightning_strike_count: None, luminosity: 0.0, monthRain: 74.7, outHumidity: 99, outTemp: 9.4, pm2_5: 9.9, pm2_51_24hav: 15.5, pressure: 992.2, radiation: 0.2, rain: None, rainRate: 6.0, relbarometer: 1003.4, stormRain: 13.2, usUnits: 17, weekRain: 13.2, wh31_ch1_batt: 0, wh31_ch2_batt: 0, wh31_ch3_batt: 0, wh31_ch4_batt: 0, wh31_ch5_batt: 0, wh31_ch6_batt: 0, wh31_ch7_batt: 0, wh31_ch8_batt: 0, wh41_ch1_batt: 4, wh55_ch1_batt: 5, wh55_ch2_batt: 4, wh57_batt: 5, wh65_batt: 0, windDir: 123, windGust: 4.1, windSpeed: 2.8, yearRain: 194.2
`

And this is the error I am getting.

`Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine: Caught unrecoverable exception in generator 'user.belchertown.HighchartsJsonGenerator'
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  Error trying to use database binding wx_binding to graph observation pm2_51_24hav. Error was: pm2_51_24hav.
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/user/belchertown.py", line 2335, in get_observation_data
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      (time_start_vt, time_stop_vt, obs_vt) = archive.getSqlVectors(TimeSpan(start_ts, end_ts), obs_lookup, aggregate_type, aggregate_interval)
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/weewx/manager.py", line 452, in getSqlVectors
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      aggregate_type, aggregate_interval)
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/weewx/xtypes.py", line 91, in get_series
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      raise weewx.UnknownType(obs_type)
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  weewx.UnknownType: pm2_51_24hav
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  During handling of the above exception, another exception occurred:
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/weewx/reportengine.py", line 197, in run
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      obj.start()
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/weewx/reportengine.py", line 280, in start
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      self.run()
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/user/belchertown.py", line 1833, in run
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      series_data = self.get_observation_data(binding, archive, observation_type, minstamp, maxstamp, aggregate_type, aggregate_interval, time_length, xAxis_groupby, xAxis_categories, mirrored_value, weatherRange_obs_lookup, wind_rose_color)
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****    File "/usr/share/weewx/user/belchertown.py", line 2337, in get_observation_data
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****      raise Warning( "Error trying to use database binding %s to graph observation %s. Error was: %s." % (binding, obs_lookup, e) )
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  Warning: Error trying to use database binding wx_binding to graph observation pm2_51_24hav. Error was: pm2_51_24hav.
Dec 21 20:42:30 LivingstoneCove weewx[28947] ERROR weewx.reportengine:         ****  Generator terminated
`

What am I doing wrong? The graph works without the bottom section for the average reading, so if i remove the following, it works with no errors :

`             [[[pm2_51_24hav]]]
                 name = 24H Average
                 color = "#f7f2b4"
                 y_label = "ug/m3"
`

gjr80, 2020-12-22:
Have you added field `pm2_51_24hav` to your database? I can't speak with certainty re the Belchertown plot generator but I understand it is similar to the WeeWX image generator in that it requires the fields that are being plotted to exist in the archive.

The section [Adding a new type to the database](http://www.weewx.com/docs/customizing.htm#add_archive_type) in the Customization Guide may be useful.

Distortionist, 2020-12-22:
I don't really know. I thought all the readings from the driver get put in
the database. It worked with other sensors without me messing about too
much. The driver definitely reports it to weewx.

gjr80, 2020-12-22:
Fields emitted by the driver are only saved to database if the field concerned exists in the in-use database schema. The GW1000 driver includes many fields that are not in the WeeWX schema, such fields are available in reports through use of the '$current.xxxx' tag but you cannot plot them or calculate any aggregates (ie `$day.xxxxx.max`) because WeeWX has no record of previous values.

The best way to view your schema is to query your database, but I am away from my pc and cannot tell you the exact sql commands to do that. The next best way is to look at the schema as written in one of your python files. If you look in `weewx.conf` under `[Databases]` you should have a stanza `[[wx_binding]]`, under that should be a line `schema =`, it will most likely be `schema = schemas.wview_extended.schema`. If you look at the WeeWX python code on your system (will be in `/home/weewx/bin` or `/usr/share/weewx` depending on your WeeWX install type) go to the `schemas` directory, look in the file `wview_extended` and find the line `table =`, it will be something like:

```
table = [('dateTime',             'INTEGER NOT NULL UNIQUE PRIMARY KEY'),
         ('usUnits',              'INTEGER NOT NULL'),
         ('interval',             'INTEGER NOT NULL'),
         ('altimeter',            'REAL'),
         ('appTemp',              'REAL'),
         ('appTemp1',             'REAL'),
         ('barometer',            'REAL'),
```

Here `altimeter`, `appTemp`, `appTemp1` are fields in your schema. Look down the list, there is no `pm2_51_24hav`. If you have modified you schema the line `schema = ...` in `weewx.conf` will likely be different and the schema it refers to will likely be different and you will need to check if it includes `pm2_51_24hav`.

If you don't remember modifying your schema almost certainly the issue is that `pm2_51_24hav` is not in your schema. If that is the case you will need to work through the link in my first post.

Distortionist, 2020-12-22:
Thank you so much for the explanation and instructions. I think you are
bang on. I'll have a look at those when I am back home.

Distortionist, 2020-12-23:
Thank you again, with your help and direction it is working and I learned something new. You rock!

closed 2020-12-23 by Distortionist