gjr80/weewx-gw1000 #22: Disppearing battery levels

harmlessdrudge, 2020-12-26:
I set up a new test system last night with Ubuntu 20.04.1 running on a Pi with WeeWX 4.2, installed weewx-g1000 (0.1.1b12) and, using default _Seasons_ skin got to see battery level indicators for my additional temperature & humidity and pressure sensors. The readings all said battery level was **LOW** although a battery tests shows all at 100%.

Today, to my surprise, the battery levels have all disappeared from the skin.

Don't think I had too much to drink yesterday. Any ideas?

I think the only deviation from the installation instructions (I installed automatically, used wee_extension) was manually changing the data_services entry here from "," but can't recall if that was before or after seeing battery indicators.

```
    [[Services]]
        prep_services = weewx.engine.StdTimeSynch
        data_services = user.gw1000.Gw1000Service
```

gjr80, 2020-12-27:
OK, so you are running the GW1000 driver as a service rather than a driver; really makes no difference but in your case the GW1000 driver is adding GW1000 observations to an existing loop packet rather than creating the loop packets (as is the case if the GW1000 driver was run as a driver). Should not make any real difference as far as the Seasons skin is concerned.

A few things to check. First of all, did you modify the Seasons skin to display the GW1000 battery state data as per [this wiki page](https://github.com/gjr80/weewx-gw1000/wiki/Adapting-the-Seasons-skin-to-display-battery-states)? You need to make some changes to the Seasons skin as the Seasons skin checks a number of specific battery states fields to see if any battery state data exists, if it cannot find any data nothing is displayed. You need to make sure it is looking for the GW1000 provided fields. Did you happen to modify the GW1000 field map at all? This would have invoked changing `[[field_map]]` or `[[field_map_extensions]]` stanzas under `[GW1000]` in `weewx.conf`. If you did change any of the battery state fields then you would need to tailor your changes to the Seasons skin accordingly.

If you have made the necessary change to the Seasons skin then try [running WeeWX directly](http://weewx.com/docs/usersguide.htm#Running_directly) so you can see the loop packets (lines starting with LOOP:) and archive records (lines starting with REC:) on the console. Have a look at the loop packet and archive record contents, do they contain the battery state fields for your GW1000 sensors? What appears now on your Seasons page?

See what that shows up and we can take it from there.

harmlessdrudge, 2020-12-27:
Actually, I just followed the installation instructions with, ahem, a glass of wine in hand, and hadn't paid too much attention to the difference. I discovered the wiki page after I posted above. Maybe it would be helpful to have an introductory discussion of both with pros, cons & requirements if any of each? No, didn't change skin. It worked immediately, then didn't.

I just blindly (er, cheerfully) assumed the "data_services = ," was an installation glitch. So, I have both driver and service configs in place. Subsequently went back and re-read wiki. Thought about modifying this skin and decided against for now. In the first instance am just setting up a new install to replace an old FineOffset WH1080 and will be using a different skin (customised version of sofaskin). When all that's done I'll look at adding battery level indicators to my current skin. Perhaps it was my "correction" that caused a problem? I don't have any [[field_map]] or [[field_map_extensions]] in weewx.com. I just added the Accumulator section. 

TBH I didn't expect this to work as well as it did. First choice was MWall's driver which seems quite old. The better looking docs, far more recent updates, and the logic of being able to interrogate the device, persuaded me to try yours with immediate satisfactory results (thanks!). Once I've finished replicating my existing set up I'll revert. Hoping meanwhile to see an updated version as I can see you have one in the works. Best wishes for 2021.

harmlessdrudge, 2020-12-27:
FWIW

`PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --test-driver
Using configuration file /etc/weewx/weewx.conf

Interrogating GW1000 at 192.168.1.54:45000

2020-12-27 12:31:30 UTC (1609072290): UV: 0, dateTime: 1609072290, dayRain: 1.5, daymaxwind: 7.7, extraHumid1: 72, extraHumid2: 92, extraHumid3: 87, extraTemp1: 10.6, extraTemp2: 3.8, extraTemp3: 5.7, inHumidity: 53, inTemp: 19.0, luminosity: 6262.0, monthRain: 231.4, outHumidity: 90, outTemp: 3.8, pressure: 957.5, rain: None, rainRate: 0.0, relbarometer: 972.5, stormRain: 19.3, usUnits: 17, uvradiation: 5.1, weekRain: 1.5, wh26_batt: 0, wh31_ch1_batt: 0, wh31_ch2_batt: 0, wh31_ch3_batt: 0, wh65_batt: 0, windDir: 214, windGust: 3.6, windSpeed: 2.5, yearRain: 1238.6
`

harmlessdrudge, 2020-12-27:
All working perfectly now. I reinstated the service, added the devices I have (wh31_ch1_batt: 0, wh31_ch2_batt: 0, wh31_ch3_batt: 0, wh65_batt) to **sensors.inc** and added labels to **skin.conf**. _Still_ don't understand why I first saw some sensors with low battery, then none at all. Haven't reproduced the former.

Subsequently, I mistakenly assumed I should do some uncommenting in the new **sensors.inc**. That didn't go well. Might be worth clarifying how this works, or at least indicating that the only thing to be edited is the device list, if indeed that's correct.

gjr80, 2020-12-28:
> All working perfectly now. 

Good.

> _Still_ don't understand why I first saw some sensors with low battery, then none at all. Haven't reproduced the former.

I can't explain that, if you did not modify `sensors.inc` as per the wiki the Seasons skin will not have known anything about GW1000 sensor battery states so I can't see how it could decode and display them. 

> Subsequently, I mistakenly assumed I should do some uncommenting in the new **sensors.inc**. That didn't go well. 

I've looked at the instructions for `sensors.inc` and the only changes that need to be made to `sensors.inc` is to modify `$sensor_batt_data` to ensure your sensors are included. Nothing about uncommenting anything. If there is something in the instructions that is incorrect or unclear please let me know what.

> Might be worth clarifying how this works, or at least indicating that the only thing to be edited is the device list, if indeed that's correct.

Point taken. I have added a one liner at the end of the install instructions referring the user to the GW100 driver wiki homepage. That will appear once the next version is released.

closed 2021-04-15 by gjr80