gjr80/weewx-gw1000 #29: Read or write SSID?

bitroost, 2021-03-26:
Since there is a 'CMD_WRITE_SSID' in Station.commands, would there be a way to change WLAN settings via the api? Or at least retrieve the SSID the device is connected to?

I couldn't find a way to retrieve the connection details in the WS View app so i am afraid it's not possible, but still wanted check if it maybe is, before having to reset and setup via the App. I recently get lots of 'No route to host' errors and wondered if it is because the device is connected to a weak WLAN. Don't remember what i set up originally.

```
Mar 25 09:15:10 rsp1 weewx[23048] DEBUG user.gw1000: Failed attempt 18 to send command 'CMD_READ_SENSOR_ID_NEW': [Errno 113] No route to host
Mar 25 09:15:43 rsp1 weewx[23048] DEBUG user.gw1000: Failed attempt 19 to send command 'CMD_READ_SENSOR_ID_NEW': [Errno 113] No route to host
Mar 25 09:16:16 rsp1 weewx[23048] DEBUG user.gw1000: Failed attempt 20 to send command 'CMD_READ_SENSOR_ID_NEW': [Errno 113] No route to host
Mar 25 09:16:16 rsp1 weewx[23048] ERROR user.gw1000: Failed to obtain response to command 'CMD_READ_SENSOR_ID_NEW' after 20 attempts
Mar 25 09:16:16 rsp1 weewx[23048] DEBUG user.gw1000: IP address specified in 'weewx.conf', re-discovery was not attempted
```

gjr80, 2021-03-26:
Indeed there is a command `CMD_WRITE_SSID` to set the SSID used by the GW1000; however, it is not something I have implemented in the driver mainly for reasons of simplicity and the lack of any need for the driver to use  the command. If someone can demonstrate a need for the command and provide details of a use case I am happy to look at it. But I would need a strong case as changing the device SSID can be dangerous as it could easily result in lost connectivity.
I am not aware of any means to obtain the device SSID via the WS View app, it is possible for the driver to obtain the SSID for all connected GW1000 via the API. Can certainly add an option to the driver to display the SSID for connected devices. I will look at this for v0.3.1.
I am not sure how much utility displaying the SSID will be in your case, remember that if the driver cannot connect to a GW1000 then the driver cannot send API commands to the GW1000 (well they will be sent but never received and thus no response sent). The GW1000 may or may not respond to broadcast commands (which is how the driver may obtain the GW1000 SSID) but if the GW1000 is in a wifi network segment with a ‘weak’ connection it is quite likely the GW1000 will not respond.

gjr80, 2021-06-20:
Have looked further into obtaining the SSID from the GW1000. The GW1000 API does provide access to a field named 'SSID' when GW1000 are discovered on the local network by UDP broadcast; however, experimentation reveals that the 'SSID' field returned is in fact the SSID used by the GW1000 when used in setup mode with the WS View app. It is not the SSID used by the GW1000 once the GW1000 is configured and connected to a wifi network.

In any case I have made this SSID field available when the driver is run directly and it will appear in v0.4.0.

closed 2021-06-20 by gjr80