gjr80/weewx-gw1000 #30: Calculated radiation field

alexdelprete, 2021-03-27:
Hi, I was not seeing solar radiation data from sent to WU or PWSWeather, then I finally found your article in the wiki (https://github.com/gjr80/weewx-gw1000/wiki/Understanding-GW1000-radiation-fields), thanks for writing it.

Two things regarding the radiation field:

1. In the article, you refer for the calculation to the **StdCalculate** service/stanza, but it should be the **StdCalibrate**.
2. What is the difference between the weewx **uvradiation** field (to which the UV gw1000 API field is mapped) and the calculated radiation field? Only the unit of measurement? I can't understand the difference between the radiation and uvradiation fields.

Thanks for your support,

Alessandro

gjr80, 2021-03-28:
Thank you, have updated the wiki article to remove `StdCalculate`.

To be truthful I don't know what the `uvradiation` field emitted the driver (and obtained via the GW1000 API) is. As outlined in the wiki article, the GW1000 API provides two UV related fields. These fields are labelled 'UV (&mu;W/m<sup>2</sup>)' and 'UVI (0-15 index)'. As WeeWX uses the field `UV` for UV index the GW1000 'UVI' field is default mapped to the WeeWX `UV` field and the GW1000 'UV' field is default mapped to the `uvradiation` field (Note that the `uvradiation` field does not appear in any of the schemas shipped with WeeWX so a default WeeWX install will not save `uvradiation` to database).

I am guessing that the GW1000 'UV' field is a measure of absolute UV flux level. The GW1000 API does not directly provide a solar insolation value, although the GW1000 will approximate a solar insolation value based on the `luminosity` value when uploading to Weather Underground. The GW1000 'UV' field value (WeeWX field `uvradiation`) is measured in &mu;W/m<sup>2</sup> with the maximum value being 65 535&mu;W/m<sup>2</sup> (which is 0.065W/m<sup>2</sup>). Solar insolation values are measured in W/m<sup>2</sup> with typical values up to 1500W/m<sup>2</sup>. So it is clear the GW1000 'UV' field (WeeWX field `uvradiation`) is not the same as the solar insolation.

I suggest if you want to know anything further about the UV radiation value emitted by the GW1000 you contact Ecowitt.

alexdelprete, 2021-03-28:
I also found out that the ecowitt custom upload protocol (the http call from the GW1000) actually provides different fields respect to the API, including solar radiation in W/m2, but probably it calculates it before including it in the data array it posts through http.

I don't understand why they differentiate things between the API vs custom upload. Weird stuff.

Thanks for the feedback. I'll contact Lucy in the feature to check if she has some documentation on this.

closed 2021-03-28 by gjr80