gjr80/weewx-gw1000 #31: FW Version (--firmware-version)

alexdelprete, 2021-03-27:
Just a small glitch, found while I was testing the command line options of the driver: when using the fw version option, the output is missing the 3rd digit. It should be v1.6.6.

> ➜  ~ PYTHONPATH=/home/weewx/bin python -m user.gw1000 --config=/etc/weewx/ecowitt.conf --firmware-version
> Using configuration file /etc/weewx/ecowitt.conf
> 
> Interrogating GW1000 at 10.1.10.118:45000
> 
> GW1000 firmware version string: **GW1000A_V1.6**.
>

gjr80, 2021-03-28:
Working fine for me:
```
gary@cockatoo:~ $ PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --firmware-version --ip-address=192.168.2.32
Using configuration file /home/weewx/weewx.conf

Interrogating GW1000 at 192.168.2.32:45000

GW1000 firmware version string: GW1000_V1.6.6
```
Not that it should matter but what version of python does 'python' run on your system? Also what version of the driver are you running?
Could you run the same command with `--debug=3`, ie:
```
$ PYTHONPATH=/home/weewx/bin python -m user.gw1000 --config=/etc/weewx/ecowitt.conf --firmware-version --debug=3
```
This will put some extended debug info in your weewx log that will look something like this:
```
Mar 28 10:32:15 cockatoo weewx[16306] DEBUG __main__: Sending packet 'FF FF 26 03 29' to '192.168.2.32:45000'
Mar 28 10:32:15 cockatoo weewx[16306] DEBUG __main__: Received response 'FF FF 26 09 50 02 91 E3 FD 32 24'
Mar 28 10:32:15 cockatoo weewx[16306] DEBUG __main__: Sending packet 'FF FF 30 03 33' to '192.168.2.32:45000'
Mar 28 10:32:16 cockatoo weewx[16306] DEBUG __main__: Received response 'FF FF 30 0B 00 01 60 60 5B 2E 5E 01 E4'
Mar 28 10:32:16 cockatoo weewx[16306] DEBUG __main__: Sending packet 'FF FF 50 03 53' to '192.168.2.32:45000'
Mar 28 10:32:16 cockatoo weewx[16306] DEBUG __main__: Received response 'FF FF 50 11 0D 47 57 31 30 30 30 5F 56 31 2E 36 2E 36 7B'
```
Could you please copy and post the entire log output resulting from that command.

alexdelprete, 2021-03-28:
Here you go:

> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Sending packet 'FF FF 26 03 29' to '10.1.10.118:45000'
> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Received response 'FF FF 26 09 F4 CF A2 69 DF E7 C3'
> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Sending packet 'FF FF 30 03 33' to '10.1.10.118:45000'
> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Received response 'FF FF 30 0B 01 01 60 60 03 E1 2B 01 0D'
> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Sending packet 'FF FF 50 03 53' to '10.1.10.118:45000'
> Mar 28 04:19:45 rpi4argon weewx[356674] DEBUG __main__: Received response 'FF FF 50 12 0E 47 57 31 30 30 30 41 5F 56 31 2E 36 2E 36 BE'
> ➜  ~ python -V
> Python 3.8.6
> ➜  ~ PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --config=/etc/weewx/ecowitt.conf --version
> GW1000 driver version: 0.3.0
> ➜  ~
>

gjr80, 2021-03-28:
Thanks. The good news is that the driver is obtaining the full (and correct) raw firmware version from the GW1000. The firmware string is in these bytes:
```
47 57 31 30 30 30 41 5F 56 31 2E 36 2E 36
``` 
which in ASCII is `GW1000A_V1.6.6`.

I just had a quick look at the firmware decode function in the driver and it is a little less robust than I would like. I am about ot head out but will re-work it later and see how that goes. I am a little concerned though that my GW1000 is showing the correct firmware string.

alexdelprete, 2021-03-28:
could it simply be something related to my terminal settings or something specific to my server? if there's a bug it shouldn't work also in your setup. strange. let me know if you need me to test something.

gjr80, 2021-03-28:
Aha, yours:
```
GW1000 firmware version string: GW1000A_V1.6.
```
and mine:
```
GW1000 firmware version string: GW1000_V1.6.6
```
You have 'GW1000A' and I have 'GW1000'. I don't know the significance of the 'A' (perhaps a hardware revision) but when coupled with my incorrect decoding of the firmware string it had the effect of dropping the final character.

It should be fixed in v0.3.1, could you try v0.3.1 by downloading it with:
```
$ wget -P /var/tmp https://raw.githubusercontent.com/gjr80/weewx-gw1000/master/bin/user/gw1000.py
```
and then copy` /var/tmp/gw1000.py` over the top of your existing version. Try the command again and please let me know how that goes.

alexdelprete, 2021-03-28:
IIRC, the letter A means 868Mhz.

I tried the version you linked, it works:

```
➜  user  PYTHONPATH=/home/weewx/bin python -m user.gw1000 --config=/etc/weewx/ecowitt.conf --firmware-version
Using configuration file /etc/weewx/ecowitt.conf

Interrogating GW1000 at 10.1.10.118:45000

GW1000 firmware version string: GW1000A_V1.6.6
```

gjr80, 2021-03-28:
Thanks, it helps when you decide according the the API rather than take shortcuts. I will release this fix plus a couple of others later today.

closed 2021-03-28 by gjr80