gjr80/weewx-gw1000 #32: Feature request: write calibration data

alexdelprete, 2021-03-27:
Hi,

the function to read calibration data of the sensors is excellent. Very useful to document/backup the configuration of the system.

I was wondering if the API allows to not only read that data, but to write it, so to have a sort of backup/restore utility of the calibration configuration.

The use case is that in the past, some fw updates deleted/reset all calibration values, and if you have lot of sensors it's a tedious and error-prone process inserting all values again.

It would be a great function, IMHO.

gjr80, 2021-03-28:
It's my intent to eventually add the ability to to the driver to be able to download some/the majority of the user configurable settings and save them to a config file. I also intend to add the ability to read such a config file and set the the relevant user configurable settings on the GW1000. I am not sure what user configurable settings I will support, but I would expect at the very least the calibration offsets and server upload configs would be included.

Unfortunately it is a case of available time and priorities so I can't give an indicative timeframe, at present most of my GW1000 time is taken up with supportin new sensors and other changes to the API.

alexdelprete, 2021-03-28:
That's great, and I didn't mean to rush anything, take your time, it was just an idea that I had for some time, and it's great that you had it planned in your to-do list.

Thanks a lot.

gjr80, 2021-03-28:
No offence taken :)

closed 2021-03-28 by gjr80