gjr80/weewx-gw1000 #33: Solar Radiation W/m2 and brigtness Lux

Sjouke62, 2021-03-30:
Is there a solution for reading Solar Radiation W/m2 and brigtness Lux not showing up in weewx.

UV index is reading ok.
Is this a driver problem or a Weewx problem.

The diver works fine!

Is this becouse of switching from interceptor driver to API driver? Do I need a fresh weewx install

What do you reccomend?

gjr80, 2021-03-30:
Solar insolation (or what WeeWX refers to as 'radiation') is a bit of an odd case. The GW1000 approximates/derives solar insolation from its light sensor and this approximated value is uploaded to sites such as Weather Underground and Ecowitt.net if you configure the GW1000 to do so. It is this data that the interceptor driver was intercepting so you would have seen the interceptor driver emit a field named `radiation`.

However, the GW1000 API (which the GW1000 driver is based upon) does not provide this radiation value, instead it provides the light value in Lux, so consequently the GW1000 driver does not emit a `radiation` value. You can have WeeWX do the same calculation the GW1000 does to approximate `radiation` by placing an entry in the `[StdCalibrate] [[Corrections]]` stanza. This is covered in the GW1000 wiki entry [Understanding GW1000 radiation fields](https://github.com/gjr80/weewx-gw1000/wiki/Understanding-GW1000-radiation-fields).

I am not sure what you mean by 'brigtness Lux not showing up in weewx', perhaps if you have a read through the wiki entry linked above that might answer your question.

Sjouke62, 2021-03-30:
That works!

 

I read trough the wiki entry and edited weewx.conf as discribed. Te Solar Radiation and Lux are now showing up in the weather34 skin (and also in the Weewx skin webpage)

 

Thanks for your quick response.