gjr80/weewx-gw1000 #35: WeeWX 4.5.1 update

harmlessdrudge, 2021-04-03:
I ran the update to WeeWX from 4.4.0 to 4.5.1 this evening and found my skin a bit munged afterwards. I think my mistake -- lesson learned -- was in not reverting to the Seasons skin for the update. I had borrowed and slightly changed the original sensors.inc file for a different skin. My original sensors.inc and the new one from Seasons are attached. After the update two changes were apparent

1. The graphs now included a signal quality graph, but dropping to zero with daylight (although data is still received by both GW1000 and HP2551); is this expected?
2. The display for the sensors now included some unwanted N/A displays, and of course reverting to the default doesn't work, and more than just the list of sensors is different. 

[sensors-old.txt](https://github.com/gjr80/weewx-gw1000/files/6253464/sensors-old.txt)
[sensors-inc.txt](https://github.com/gjr80/weewx-gw1000/files/6253465/sensors-inc.txt)

Even if I had included rather than copied the Seasons/templates/sensors.inc file I presume that the results would have been the same. Should the list of sensors be a separate include file? 

I've commented out the sensors display card for now and will wait for yr advice about fixing and best way of avoiding this getting hit by future updates.

gjr80, 2021-04-15:
If done as per the Users Guide/Upgrade Guide an upgrade should not modify any skin files unless explicitly stated in the Upgrade Guide (and there have been no occurrences since v3.9.0 was released). I run a modified Seasons skin and have had not problems at all during an upgrade. 

I am not sure what you mean by:

gjr80, 2021-04-15:
If done as per the Users Guide/Upgrade Guide an upgrade should not modify any skin files unless explicitly stated in the Upgrade Guide (and there have been no such occurrences since v3.9.0 was released). I run a modified Seasons skin and have not had skin problems during any upgrades.

I am not sure what you mean by "_I had borrowed and slightly changed the original sensors.inc file for a different skin_". So you mean you modified `sensors.inc` in the Seasons skin or you took the Seasons skin `sensors.inc` and modified it/used it in another skin? I'll assume it's the former.

In any case looking at the `sensors.inc` files you attached it looks like `sensors-old.txt` is a modified version of `sensors.inc` as described in the GW1000 wiki. The `sensors-inc.txt` looks like a plain vanilla Season skin `sensors.inc`. The signal quality graph and N/A displays you describe sound entirely consistent with your Seasons skin reverting to the plain vanilla version which includes a plot of `rxCheckPercent` (which the GW1000 driver does not emit) and the N/A displays will likely be due to sensor states being displayed for sensors that the GW1000 does use/emit. (You should be able to work out what is being plotted by looking at the `[ImageGenerator]` stanza of the Seasons skin config file)

Can you not simply replace current Seasons skin `sensors.inc` with a copy of your modified (or -old) `sensors.inc` ? Have you tried this?

As for avoiding the situation in future. There really should be no need to add further `.inc` files; for one there is already enough obfuscation and re-direction in the Seasons skin and secondly what is to say that such an `.inc` file wouldn't have been reverted to original as your `sensors.inc` was? Keeping a backup of any modified skins would be prudent. I would also take note of what happens next time you upgrade, if the issue occurs again it is a (WeeWX) bug and should be posted to the [weewx-user Google group](https://groups.google.com/g/weewx-user). Do make sure you keep and provide a copy of the terminal screen dialogue from the upgrade though.

harmlessdrudge, 2021-05-14:
Apologies for the delayed reply.
> I'll assume it's the former
No, in fact, it was the latter.

The signal graph and N/A displays were new and completely different respectively after the update using my skin, not Seasons. I changed nothing in the [ImageGenerator]. My thinking in asking if separating the sensors list would be an idea was to make it skin independent, that's all. I'll get back to this later.

gjr80, 2021-10-01:
closing for now, no activity in 6 months. Can be re-opened later if required.

closed 2021-10-01 by gjr80