gjr80/weewx-gw1000 #36: GW-1000 Service stops updating after a period of time

lowercasebean, 2021-04-12:
I am experiencing an issue it seems with data from the GW-1000 not updating. After a period of time, maybe 30 minutes or so my readings for relative pressure don’t update, from what I can tell the GW-1000 is still providing the data when this happens. I have tried debugging etc but I can’t get to the bottom of the issue.
The only real output I can see directly from the syslog is the below every 20 seconds

`weewx[31904] DEBUG user.gw1000: Next update in 20 seconds`

For some background, I have the RTL SDR driver installed and this is receiving data from my weather station. The GW-1000 extension (latest version) is installed as a service to supplement my weather station data and add the barometric pressure and internal temperature and humidity.

Is there anything I can do to troubleshoot my issue further or can you offer any advice as to possible cause(s)?

Live data reading (from after data stopped updating in weewx):

```
$ PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --live-data

Using configuration file /etc/weewx/weewx.conf
Interrogating GW1000 at 192.168.5.241:45000
GW1000 live sensor data (2021-04-12 12:29:13 IST (1618226953)): absbarometer: 1013.4hPa, humid1: 46%, humid2: 59%, inhumid: 54%, intemp: 20.9°C, relbarometer: 1029.4hPa, temp1: 26.1°C, temp2: 10.8°C, wh31_ch1_batt: 0, wh31_ch1_sig: 2, wh31_ch2_batt: 0, wh31_ch2_sig: 4
```

Weewx conf file (I cut down on most of the default setting to make it easier to read. Anything I’ve removed is unrelated to GW-1000 or SDR.
```
Weewx.conf
# WEEWX CONFIGURATION FILE
#
# Copyright (c) 2009-2019 Tom Keffer <tkeffer@gmail.com>
# See the file LICENSE.txt for your rights.

##############################################################################

debug = 1

WEEWX_ROOT = /

log_success = True

log_failure = True

socket_timeout = 20

version = 4.5.1

##############################################################################

#   This section is for information about the station.

[Station]
    
    altitude = 132, meter

    station_type = SDR

    rain_year_start = 1
    
    week_start = 6

##############################################################################

[SDR]

    driver = user.sdr
    cmd = rtl_433 -M utc -F json -f 868312k -s 1024k -Y classic
    path = /usr/local/bin:${PATH}
    ld_library_path = /usr/bin
    [[sensor_map]]
        windGust = wind_gust.220.FOWHx080Packet
        outBatteryStatus = battery.220.FOWHx080Packet
        rain_total = rain_total.220.FOWHx080Packet
        windSpeed = wind_speed.220.FOWHx080Packet
        windDir = wind_dir.220.FOWHx080Packet
        outHumidity = humidity.220.FOWHx080Packet
[... 333 lines cut ...]
    [[wh51_ch12_sig]]
        extractor = last
    [[wh51_ch13_sig]]
        extractor = last
    [[wh51_ch14_sig]]
        extractor = last
    [[wh51_ch15_sig]]
        extractor = last
    [[wh51_ch16_sig]]
        extractor = last
    [[wh55_ch1_sig]]
        extractor = last
    [[wh55_ch2_sig]]
        extractor = last
    [[wh55_ch3_sig]]
        extractor = last
    [[wh55_ch4_sig]]
        extractor = last
    [[wh57_sig]]
        extractor = last
    [[wh68_sig]]
        extractor = last
    [[ws80_sig]]
        extractor = last
```

gjr80, 2021-04-15:
It's impossible to say what is going on without seeing the log. The issue could be with the GW1000 driver or it could be a SDR driver issue (when operated as a service the GW1000 driver is somewhat beholden to the driver in use). I suggest you edit `weewx.conf`, set `debug = 1`, save `weewx.conf` and restart WeeWX. Let WeeWX run for at least 15 minutes and then take a log extract being sure to capture the entire WeeWX start up as well as the next 15 minutes of activity. Post the log here.

lowercasebean, 2021-04-22:
Thanks for your response, I have been attempting to capture the log from the start of the Weewx service up to and including the GW-1000 updates ceasing but it seems to run great for a couple of hours before the updates stop. I do understand that running the gw1000 driver as a service complicates things a little like you said, hopefully the logs will help figure out what the cause of the updates ceasing could be. Let me know if you need any further info. I've added a screenshot of the barometer graph for the week just to give you an idea of how long it runs for before it stops updating the readings. I have restarted the service a couple of times this week.

````
Apr 22 12:50:40 weather systemd[1]: weewx.service: Succeeded.
Apr 22 12:50:40 weather systemd[1]: Stopped LSB: weewx weather system.
Apr 22 13:02:07 weather systemd[1]: Starting LSB: weewx weather system...
Apr 22 13:02:08 weather weewx[14321] INFO __main__: Initializing weewx version 4.5.1
Apr 22 13:02:08 weather weewx[14321] INFO __main__: Using Python 3.7.3 (default, Jan 22 2021, 20:04:44) #012[GCC 8.3.0]
Apr 22 13:02:08 weather weewx[14321] INFO __main__: Platform Linux-5.10.17-v7+-armv7l-with-debian-10.9
Apr 22 13:02:08 weather weewx[14321] INFO __main__: Locale is 'C.UTF-8'
Apr 22 13:02:08 weather weewx[14321] INFO __main__: PID file is /var/run/weewx.pid
Apr 22 13:02:08 weather weewx[14325] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Apr 22 13:02:08 weather weewx[14325] INFO __main__: Debug is 1
Apr 22 13:02:08 weather weewx[14325] DEBUG __main__: Initializing engine
Apr 22 13:02:08 weather weewx[14325] INFO weewx.engine: Loading station type SDR (user.sdr)
Apr 22 13:02:08 weather weewx[14310]: Starting weewx weather system: weewx.
Apr 22 13:02:08 weather systemd[1]: Started LSB: weewx weather system.
Apr 22 13:02:08 weather weewx[14325] INFO user.sdr: driver version is 0.80
Apr 22 13:02:08 weather weewx[14325] INFO user.sdr: sensor map is {'windGust': 'wind_gust.220.FOWHx080Packet', 'outBatteryStatus': 'battery.220.FOWHx080Packet', 'rain_total': 'rain_total.220.FOWHx080Packet', 'windSpeed': 'wind_speed.220.FOWHx080Packet', 'windDir': 'wind_dir.220.FOWHx080Packet', 'outHumidity': 'humidity.220.FOWHx080Packet', 'outTemp': 'temperature.220.FOWHx080Packet', 'polyTemp': 'temperature.221.AmbientWH31EPacket', 'polyHumidity': 'humidity.221.AmbientWH31EPacket', 'polyBattery': 'battery.221.AmbientWH31EPacket', 'patioTemp': 'temperature.155.AmbientWH31EPacket', 'patioHumidity': 'humidity.155.AmbientWH31EPacket', 'patioBattery': 'battery.155.AmbientWH31EPacket'}
Apr 22 13:02:08 weather weewx[14325] INFO user.sdr: deltas is {'rain': 'rain_total'}
Apr 22 13:02:08 weather weewx[14325] INFO user.sdr: startup process 'rtl_433 -M utc -F json -f 868312k -s 1024k -Y classic'
Apr 22 13:02:08 weather weewx[14325] DEBUG user.sdr: start async reader for stdout-thread
Apr 22 13:02:08 weather weewx[14325] DEBUG user.sdr: start async reader for stderr-thread
Apr 22 13:02:08 weather weewx[14325] DEBUG weewx.engine: Loading service weewx.engine.StdTimeSynch
Apr 22 13:02:08 weather weewx[14325] DEBUG weewx.engine: Finished loading service weewx.engine.StdTimeSynch
Apr 22 13:02:08 weather weewx[14325] DEBUG weewx.engine: Loading service user.gw1000.Gw1000Service
Apr 22 13:02:08 weather weewx[14325] INFO user.gw1000: field map is {'barometer': 'relbarometer', 'co2': 'co2', 'co2_24h_avg': 'co2_24h_avg', 'dateTime': 'datetime', 'daymaxwind': 'daymaxwind', 'dayRain': 'rainday', 'dewpoint': 'dewpoint', 'extraHumid1': 'humid1', 'extraHumid2': 'humid2', 'extraHumid3': 'humid3', 'extraHumid4': 'humid4', 'extraHumid5': 'humid5', 'extraHumid6': 'humid6', 'extraHumid7': 'humid7', 'extraHumid8': 'humid8', 'extraHumid17': 'humid17', 'extraTemp1': 'temp1', 'extraTemp2': 'temp2', 'extraTemp3': 'temp3', 'extraTemp4': 'temp4', 'extraTemp5': 'temp5', 'extraTemp6': 'temp6', 'extraTemp7': 'temp7', 'extraTemp8': 'temp8', 'extraTemp9': 'temp9', 'extraTemp10': 'temp10', 'extraTemp11': 'temp11', 'extraTemp12': 'temp12', 'extraTemp13': 'temp13', 'extraTemp14': 'temp14', 'extraTemp15': 'temp15', 'extraTemp16': 'temp16', 'extraTemp17': 'temp17', 'heatindex': 'heatindex', 'hourRain': 'rainhour', 'inHumidity': 'inhumid', 'inTemp': 'intemp', 'leafWet1': 'leafwet1', 'leafWet2': 'leafwet2', 'leafWet3': 'leafwet3', 'leafWet4': 'leafwet4', 'leafWet5': 'leafwet5', 'leafWet6': 'leafwet6', 'leafWet7': 'leafwet7', 'leafWet8': 'leafwet8', 'leak1': 'leak1', 'leak2': 'leak2', 'leak3': 'leak3', 'leak4': 'leak4', 'lightning_distance': 'lightningdist', 'lightning_last_det_time': 'lightningdettime', 'lightning_strike_count': 'lightning_strike_count', 'lightningcount': 'lightningcount', 'luminosity': 'light', 'monthRain': 'rainmonth', 'outHumidity': 'outhumid', 'outTemp': 'outtemp', 'pm2_5': 'pm251', 'pm2_51_24h_avg': 'pm251_24h_avg', 'pm2_52': 'pm252', 'pm2_52_24h_avg': 'pm252_24h_avg', 'pm2_53': 'pm253', 'pm2_53_24h_avg': 'pm253_24h_avg', 'pm2_54': 'pm254', 'pm2_54_24h_avg': 'pm254_24h_avg', 'pm2_55': 'pm255', 'pm2_55_24h_avg': 'pm255_24h_avg', 'pm10': 'pm10', 'pm10_24h_avg': 'pm10_24h_avg', 'pressure': 'absbarometer', 'rain': 'rain', 'rainRate': 'rainrate', 'soilMoist1': 'soilmoist1', 'soilMoist2': 'soilmoist2', 'soilMoist3': 'soilmoist3', 'soilMoist4': 'soilmoist4', 'soilMoist5': 'soilmoist5', 'soilMoist6': 'soilmoist6', 'soilMoist7': 'soilmoist7', 'soilMoist8': 'soilmoist8', 'soilMoist9': 'soilmoist9', 'soilMoist10': 'soilmoist10', 'soilMoist11': 'soilmoist11', 'soilMoist12': 'soilmois [... cut ...]
Apr 22 13:19:26 weather weewx[14325] INFO weewx.restx: MQTT: Published record 2021-04-22 13:19:22 IST (1619093962)
Apr 22 13:19:28 weather weewx[14325] DEBUG user.sdr: lines=[]
Apr 22 13:19:31 weather weewx[14325] INFO weewx.restx: Wunderground-RF: Published record 2021-04-22 13:19:22 IST (1619093962)
Apr 22 13:19:31 weather weewx[14325] DEBUG user.sdr: lines=[]
Apr 22 13:19:34 weather weewx[14325] DEBUG user.sdr: lines=[]
Apr 22 13:19:36 weather weewx[14325] DEBUG user.gw1000: Next update in 20 seconds
````
[image]

Laephis, 2021-06-08:
Hello,

I've been experiencing a similar issue/behavior (GW-1000 stops updating periodically), but it doesn't involve the RTL SDR driver.  Essentially, the external data (outdoor temp, humidity, barometer) stop updating for a while and display "N/A", then randomly start working again.  The internal temps continue to read correctly and remain uninterrupted whenever this happens.  

I have logs and screenshots I can share, but wasn't sure if you wanted them attached to this issue or a separate one created.  Thanks!

gjr80, 2021-06-09:
> Hello,
>
> I've been experiencing a similar issue/behavior (GW-1000 stops updating periodically), but it doesn't involve the RTL SDR driver. Essentially, the external data (outdoor temp, humidity, [...]

Are you running the GW1000 driver as a driver or as service? I am almost certain that this issue is due to some peculiarities when the GW1000 driver is run as a service. If you are running as a driver then probably best you raise a new issue. Logs speak a thousand words so do provide them.

gjr80, 2021-06-09:
@lowercasebean, my apologies for dropping the ball on this. The log show that the GW1000 driver is continuing to operate and does not die but unfortunately it does not tell me anything about why there is no data being presented. To do that I will need to give an instrumented version of the GW1000 driver that will log more detail. If this issue is still ongoing let me know and I will provide you with the instrumented driver.

Laephis, 2021-06-10:
> Are you running the GW1000 driver as a driver or as service? I am almost certain that this issue is due to some peculiarities when the GW1000 driver is run as a service. If you are running as a [...]

It's been a while since I set it all up, but I'm pretty sure I'm running it as a driver (used the wee_extension install method).  I open another issue and post some log examples.  Thanks!

lowercasebean, 2021-06-12:
> @lowercasebean, my apologies for dropping the ball on this. The log show that the GW1000 driver is continuing to operate and does not die but unfortunately it does not tell me anything about why [...]

@gjr80 It's no problem. I can report that at the moment it seems to have stabilized a bit, in the interim I have only needed to restart the service a handful of times due to dropouts. I do think that it was a bit more unstable when I was restarting the service daily for backup purposes (I switched this off long before creating an issue in case this was somehow the culprit); which leads me to think the issue could be something to do with when the service is started. I wonder if the SDR driver is competing with the GW1000 Service and that may be causing the issue? Either way I have no problem running the instrumented driver and posting logs if / when a dropout occurs

gjr80, 2021-06-16:
OK. Let me put together an instrumented version that won't fill the logs as much as debug = 3.

gjr80, 2021-06-22:
In sitting down and instrumenting the code I think I have discovered what may be going on to cause your problem.

In simplified terms the portion of the GW1000 driver that polls the GW1000 is run in its own thread so that it is not constrained by the WeeWX event driven architecture. The driver thread just polls away every `poll_interval` seconds and passes data up to the main portion of the GW1000 driver run by the main WeeWX code via a python queue. When a loop packet arrives from the driver being used by WeeWX (in your case the SDR driver) then the portion of the GW1000 code running as a service grabs the next item from the queue and if it is observation data it adds it to the SDR emitted loop packet. All is good. However, if the driver emits loop packets less frequently than the GW1000 driver places them in the queue then the queue can build up in size and GW1000 data missed. Worst case the queue continues to grow in size until it hits its maximum size. Eventually the GW1000 service is getting packets from the queue that are sufficiently old that they are considered stale and discarded, therefore the GW1000 data is not added to the loop packet and ultimately you get your gaps in GW1000 data.

Looking at your logs I see your SDR loop packets being emitted at different intervals, some are 60 seconds, some 20, some less. But the GW1000 driver keeps queuing data every 20 seconds so I am pretty confident you will eventually run into the problem outlined above.

We can easily check, can you move your `gw1000.py` aside (just rename `/usr/share/weewx/user/gw1000.py` to `gw1000_orig.py`) and replace it with the attached version (you will need to rename it from `gw1000.txt` to `gw1000.py`). Restart WeeWX and let it run for an hour or so. This version will log some details of the SDR emitted packet, as well as what the GW1000 service picks up from the queue and the size of the queue. It will log about four extra lines per loop packet (so your logs shouldn't grow too exponentially). Post the log extract from restart through until the hour elapsed. You can turn off the extra logging by placing `log_inst = False` under `[GW1000]` in `weewx.conf` and restarting WeeWX, eg:
```
[GW1000]
    ....
    log_inst = False
    ....
```

Once we confirm that is in fact what is happening I can refactor the queue processing code to fix it.

[gw1000.txt](https://github.com/gjr80/weewx-gw1000/files/6691966/gw1000.txt)

lowercasebean, 2021-06-24:
I left the new driver running for an hour (starting after a reboot). Your explanation sounds very likely given that it can run fine for hours, days or weeks after restarting the service and then suddenly stop working. Let me know if you need any more info to help get to the bottom of the issue. It was working fine again up until a few days ago when it apparently quit updating again, I actually didn't notice until I went changing the driver over to the new one to get the logs.

I'm attaching the (sanitized) debug log in text format, its over the 65536 characters allowed in a GitHub comment apparently.

[syslog_debug.txt](https://github.com/gjr80/weewx-gw1000/files/6711437/syslog_debug.txt)

gjr80, 2021-06-26:
Thanks. Interestingly there is some evidence in the log extract you attached of items backing up in the queue but they are being progressively cleared. So not quite the failure mode I was describing, but then again there could be a variation of what I described if there happens to be an extended period with no SDR packets coming in. There weren't any such periods in the log so not sure.

In any case I have re-written the method that manages the queue when run as a service. I just need to go through and test it to make sure it is well behaved. Once I am happy with it I will give you a link to download and install.

gjr80, 2021-06-28:
I have reworked the Gw1000Service code that handles the augmentation of loop packets with GW1000 data and am happy it is well behaved. There are a few other changes in there mainly to do with processing of battery state data (battery state is shown as None if there is no sensor signal level data) and some command line and debug options but they should not bother you.

Could I get you to try the `gw1000.py` from the `development` branch. To install:

1. make a copy of your existing `/usr/share/weewx/user/gw1000.py` so you can revert to it if things go wrong
2. download the `development` branch `gw1000.py`:
```
$ wget -O /usr/share/weewx/user/gw1000.py https://raw.githubusercontent.com/gjr80/weewx-gw1000/development/bin/user/gw1000.py
```
3. restart WeeWX

lowercasebean, 2021-06-28:
Thank you! I have just installed it on my weewx system, I will let you know if I notice any issues.

lowercasebean, 2021-07-06:
After a week or so of running the new code I haven't seen any issues with updating. I think we can call it a success and you can close this issue, if you are happy to. Thanks again for all the help and the awesome software!

gjr80, 2021-07-06:
That is good news. The updated code will be included as part of the v0.4.0 release.

closed 2021-07-06 by gjr80