gjr80/weewx-gw1000 #37: Field Map: light -> illuminance

alexdelprete, 2021-04-14:
I was having problems with the default mapping (with Influxdb and Grafana) and found the Field Map page in the wiki.

I found out, reading a couple of posts by Tom, that the field that better represents the light sensor should be in reality be "illuminance", introduced in weewx 4.2, and not luminosity.

https://github.com/weewx/weewx/issues/613#issuecomment-716136794
https://groups.google.com/g/weewx-user/c/gcjK6o_1QXE/m/qMMW3Mg_BwAJ

I used a [[field_map_extensions]] stanza, under [GW1000], to remap it:

```
[[field_map_extensions]]
     illuminance = light

```

With this I solved the issue I had exporting data to influxdb, but I thought it would be useful to have your opinion about the default mappings.

gjr80, 2021-04-15:
You have stumbled onto an issue with a bit of history. Yes you are correct that the GW1000 field `light` is in fact illuminance and should be mapped to a field named `illuminance`. The problem is that before WeeWX v4.2.0 the `wview_extended` schema included a field `luminosity` but did not include a field `luminance`. WeeWX v4.2.0 included limited support for `illuminance`  (this  consisted largely of assigning `luminance` to the `group_illuminance` unit group) but never actually included `illuminance` in the `wview_extended` schema (or any other schema for that matter). Interestingly, while `illuminance` was assigned to a unit group, `luminosity` was not assigned to a unit group. Earlier this year Tom and I had some off-line discussions over these discrepancies when the weewx-user thread you linked was started. Essentially we had two choices; we could modify the `wview_extended` schema to include `illuminance` and maybe/maybe not remove `luminosity` with commensurate changes to the unit groups as well as GW1000 and fousb drivers. Alternatively, we could leave the `wview_schema` as is (ie `luminosity` but no `luminance`), fix the unit groups to be consistent and leave the GW1000 and fousb drivers as is. 

Ultimately the decision was made (some months after [WeeWX issue 613](https://github.com/weewx/weewx/issues/613) was closed) to adopt the latter, leaving the `wview_extended` schema with `luminosity` but no `illuminance` accepting this was technically incorrect but avoiding a complicated upgrade procedure that would be required if there were changes to the `wview_extended` schema. If you look at the [current (v4.5.1) `wview_extended` schema](https://github.com/weewx/weewx/blob/master/bin/schemas/wview_extended.py#L77) you will see it includes `luminosity` and has no `illuminance`.

One of the tasks that fell out of the decision to retain `luminosity` was to include some some suitable explanatory words in the WeeWX documentation. This was to be the cue for me to include some appropriate words in the GW1000 driver documentation, but it appears the WeeWX documentation was never updated. I will take this up with Tom.

In the end you have solved your issue in exactly the way it should, through altering your GW1000 driver field map.

alexdelprete, 2021-04-15:
> while `illuminance` was assigned to a unit group, `luminosity` was not assigned to a unit group.

that's what triggered my attention to the problem: when the influxdb driver creates the records, you can see `fieldname_unit` as the name of the record. But the `luminosity ` record didn't have the `_unit` suffix. So I started investigating...

Question #1: why adding `illuminance` to the schema is such a breaking change? In the past the schema has actually been changed, IIRC. 

Question #2: has `luminosity ` been added to the unit group? Because in the last weewx version I still noticed that the influxdb driver exports the field without unit.

Thanks a lot as always for the in-depth explanation, it really helps me understanding how things work with weewx, I appreciate it.

closed 2021-04-15 by alexdelprete