gjr80/weewx-gw1000 #39: Battery information

wifi75, 2021-04-22:
Hello how I can get battery information?
on my weewx template  is not present:
https://meteo.iu3cyv.eu/weewx/

I Have also mqtt service enable how I can send battery with mqtt broker?

gjr80, 2021-04-24:
What you need to do really depends on what you want to display. Do you want to just display the current battery state or do you want to graph it on a plot? Do you want to display aggregates? I would suggest there is no real value in displaying battery state aggregates. I would suggest you just start displaying the current value and work from there.

You are using the WeeWX Standard skin which is fairly basic in structure and easy to modify. You will need to edit the `/etc/weewx/skins/Standard/index.html.tmpl` file but make sure you keep a backup in case you need to go back to a working configuration if you make a mistake. I suggest to you edit the file and have a look through the template code. In particular look at the 'Current Conditions' panel. That should give you some idea of the structure and code required. You will see WeeWX tags such as `$current.outTemp` and `$current.outHumidity`, these tags display current WeeWX observations, in this case the current outside temperature and the current outside humidity. You might choose to create a new panel for your battery state info, perhaps 'Sensors' or 'Sensor State'.

You will need to find out the names of the battery state fields you wish to display. You can do this by running the GW1000 driver directly with the `--test-driver` command line option (step 4 in the [GW1000 driver Quick Start Guide](https://github.com/gjr80/weewx-gw1000/wiki/Quick-Start-Guide) shows how to run the driver directly). This will display to screen the fields that the driver will emit. Suppose you see field `wh57_batt` then you would use `$current.wh57_batt` in a template to display the current WH57 battery state data.

If you want to plot the battery state info you will need to make sure the applicable fields are included in your database schema. You should refer to the [Adding a new type to an existing database](http://weewx.com/docs/customizing.htm#add_archive_type) section in the [Customisation Guide](http://weewx.com/docs/customizing.htm). Once you have a field in the schema, and provided the same field is emitted by the driver, WeeWX will automatically save that data to database. Once you have the battery data being stored in the database you can then have the Standard skin create plots by modifying the `[ImageGenerator]` stanza in the Standard skin config file (`/etc/weewx/skins/Standard/skin.conf`) and you can modify `/etc/weewx/skins/Standard/index.html.tmpl` to display the plots (again have a look at `/etc/weewx/skins/Standard/index.html.tmpl` and see how the other plots are displayed and follow by example.

MQTT support is beyond the scope of the GW1000 driver. I suggest you post to the [weewx-user Google Group](https://groups.google.com/g/weewx-user) with your questions.

gjr80, 2021-06-09:
Closing due to lack of activity. Can be re-opened if required.

closed 2021-06-09 by gjr80