gjr80/weewx-gw1000 #41: Driver periodically fails to update weather data

Laephis, 2021-06-10:
I've been experiencing occasional periods of time where WeeWX isn't receiving data while using the gw1000 driver.  I've tried to look for patterns, but nothing has jumped out at me yet.  Sometimes restarting the weewx service or rebooting my pi resolves it, sometimes it doesn't.  Then it all shows up again.  The problem appears isolated to the outside temp, heat index, wind chill, humidity, and barometer readings.  The other sensors continue to work and are never affected by this.  I've attached some screenshots showing examples of problem.  Also, here's ~15 minutes of data from the debug log file after first starting up (and experiencing the issue.)  The only change I've made to the system (other than to update the driver to 0.3.1 and weewx to 4.5.1) is a couple firmware updates on the gw1000.  

```
Jun  8 11:10:21 weather weewx[1808] DEBUG __main__: Initializing engine
Jun  8 11:10:23 weather weewx[1808] DEBUG user.gw1000: max tries is 3, retry wait time is 10 seconds
Jun  8 11:10:23 weather weewx[1808] DEBUG user.gw1000: broadcast address is 255.255.255.255:46000, socket timeout is 2 seconds
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdTimeSynch
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdTimeSynch
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdConvert
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdConvert
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdCalibrate
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdCalibrate
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdQC
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdQC
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.wxservices.StdWXCalculate
Jun  8 11:10:23 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.wxservices.StdWXCalculate
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdWXXTypes
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdWXXTypes
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdPressureCooker
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdPressureCooker
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdRainRater
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdRainRater
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdDelta
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdDelta
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdArchive
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Use LOOP data in hi/low calculations: 1
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdArchive
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdStationRegistry
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdStationRegistry
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdWunderground
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdWunderground
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdPWSweather
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdPWSweather
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdCWOP
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdCWOP
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdWOW
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdWOW
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.restx.StdAWEKAS
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.restx.StdAWEKAS
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdPrint
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdPrint
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Loading service weewx.engine.StdReport
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Finished loading service weewx.engine.StdReport
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.engine: Station does not support reading the time
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:10:23 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:10:43 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:11:03 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:11:23 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:11:43 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:12:03 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:12:23 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:12:44 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:13:04 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:13:24 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:13:44 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:14:04 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:14:24 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:14:45 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:15:05 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:15:25 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:15:30 weather weewx[1808] DEBUG weewx.reportengine: Running reports for latest time in the database.
Jun  8 11:15:30 weather weewx[1808] DEBUG weewx.reportengine: Running report 'SeasonsReport'
Jun  8 11:15:30 weather weewx[1808] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Seasons/skin.conf for report 'SeasonsReport'
Jun  8 11:15:30 weather weewx[1808] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras', 'weewx.cheetahgenerator.JSONHelpers']
Jun  8 11:15:30 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:15:31 weather weewx[1808] DEBUG weewx.restx: CWOP: Connected to server cwop.aprs.net:14580
Jun  8 11:15:35 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
[... 92 lines cut ...]
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Mobile/skin.conf for report 'MobileReport'
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.cheetahgenerator: Using search list ['weewx.cheetahgenerator.Almanac', 'weewx.cheetahgenerator.Station', 'weewx.cheetahgenerator.Current', 'weewx.cheetahgenerator.Stats', 'weewx.cheetahgenerator.UnitInfo', 'weewx.cheetahgenerator.Extras', 'weewx.cheetahgenerator.JSONHelpers']
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.manager: Daily summary version is 4.0
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.reportengine: Report 'StandardReport' not enabled. Skipping.
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.reportengine: Running report 'Belchertown'
Jun  8 11:25:36 weather weewx[1808] DEBUG weewx.reportengine: Found configuration file /etc/weewx/skins/Belchertown/skin.conf for report 'Belchertown'
[... cut ...]
Jun  8 11:26:10 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:26:30 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:26:50 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:27:10 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:27:30 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds
Jun  8 11:27:50 weather weewx[1808] DEBUG user.gw1000: Next update in 20 seconds

```

[image]
[image]
[image]

Zaschii, 2021-06-10:
some issues with driver  0.3.1 and weewx to 4.5.1 too. 

```
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000: Unexpected exception occurred while checking response to attempt 1 to send command 'CMD_READ…x out of range
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  Traceback (most recent call last):
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3345, in send_cmd_with_retries
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      self.check_response(response, self.commands[cmd])
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range

```

gjr80, 2021-06-11:
> some issues with driver 0.3.1 and weewx 4.5.1 too.
> 
> ```
> Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
> Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
> Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000: Unexpected exception occurred while checking response to attempt 1 to send command 'CMD_READ…x out of range
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  Traceback (most recent call last):
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3345, in send_cmd_with_retries
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      self.check_response(response, self.commands[cmd])
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
> Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range
> ```

The problem you are experiencing is quite different and has nothing to do with the original issue posted by @Laephis. I have moved your problem to issue #42. Please continue any discussion there.

gjr80, 2021-06-11:
@Laephis. A few questions. You mentioned in #36 that you are not running the SDR driver but you did not mention whether you are running the GW1000 driver as a driver or as a service with some other driver? (the log extract above would show it but unfortunately you did not include the entire WeeWX startup).

My first thoughts are that perhaps one or more sensors are not being picked up by the GW1000. What hardware does your Ecowitt setup comprise? `dewpoint`, `windchill` and `heatindex` are all derived obs that are calculated by WeeWX, they will almost certainly be missing due to the missing `outTemp` (`outTemp` is a pre-requisite for their calculation). 

Have you modified the Seasons skin to display the GW1000 sensor status data? If so what does it show? 

It would be useful to see the output of the following commands:
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --sensors
```
and
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --live-data
```
(depending on your system you might need to replace `python` with `python3` in the above commands)

These commands will show details on the sensors linked to the GW1000 as well as the current data from the linked sensors.

One other thing, does your GW1000 send data to Ecowitt.net? If so does Ecowitt.net show any missing data?

Laephis, 2021-06-11:
@gjr80 Thanks for taking the time to look into this, appreciate the response and I'll try my best to answer your questions.

I have an HP3501 kit (with associated sensors) in addition to the GW1000.  (I got the latter in order to be able to do more with the data with things like weewx, cumulusmx, etc.)  

I don't recall doing any modification to the Seasons skin, mostly because I just use belchertown to display the data on a tablet. The screenshot I attached previously is what it looks like when it's not working (just displays "N/A")

Here's the output of the commands you listed:
```
pi@weather:~ $ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --sensors
Using configuration file /etc/weewx/weewx.conf

Interrogating GW1000 at 192.168.1.203:45000

Sensor     Status
WH68       sensor ID: a8  signal: 4  battery: 1.84 (OK)
WH65       sensor is registering...
WH40       sensor ID: c526  signal: 4  battery: 0 (OK)
WS80       sensor is registering...
WH26       sensor ID: 98  signal: 4  battery: 0 (OK)
WH31 ch2   sensor ID: 1e  signal: 4  battery: 0 (OK)
WH31 ch1   sensor ID: b  signal: 4  battery: 0 (OK)
WH31 ch4   sensor is registering...
WH31 ch3   sensor is registering...
WH31 ch6   sensor is registering...
WH31 ch5   sensor is registering...
WH31 ch8   sensor is registering...
WH31 ch7   sensor is registering...
WH51 ch2   sensor is registering...
WH51 ch1   sensor is registering...
WH51 ch4   sensor is registering...
WH51 ch3   sensor is registering...
WH51 ch6   sensor is registering...
WH51 ch5   sensor is registering...
WH51 ch8   sensor is registering...
WH51 ch7   sensor is registering...
WH41 ch2   sensor is registering...
WH41 ch1   sensor is registering...
WH41 ch4   sensor is registering...
WH41 ch3   sensor is registering...
WH55 ch1   sensor is registering...
WH57       sensor ID: dc21  signal: 4  battery: 5 (OK)
WH55 ch3   sensor is registering...
WH55 ch2   sensor is registering...
WH34 ch1   sensor is registering...
WH55 ch4   sensor is registering...
WH34 ch3   sensor is registering...
WH34 ch2   sensor is registering...
WH34 ch5   sensor is registering...
WH34 ch4   sensor is registering...
WH34 ch7   sensor is registering...
WH34 ch6   sensor is registering...
WH45       sensor is registering...
WH34 ch8   sensor is registering...
WH35 ch2   sensor is registering...
WH35 ch1   sensor is registering...
WH35 ch4   sensor is registering...
WH35 ch3   sensor is registering...
WH35 ch6   sensor is registering...
WH35 ch5   sensor is registering...
WH35 ch8   sensor is registering...
WH35 ch7   sensor is registering...
pi@weather:~ $
```
```
pi@weather:~ $ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --live-data
Using configuration file /etc/weewx/weewx.conf

Interrogating GW1000 at 192.168.1.203:45000

GW1000 live sensor data (2021-06-10 20:39:47 CDT (1623375587)): absbarometer: 977.6hPa, daymaxwind: 28kph, gustspeed: 2kph, humid1: 37%, humid2: 56%, inhumid: 36%, intemp: 21.6°C, light: 0lx, lightningcount: 3, lightningdettime: 10-Jun-2021 19:26, lightningdist: 20.0km, outhumid: 62%, outtemp: 26.9°C, rainday: 0.0mm, rainevent: 0.0mm, rainmonth: 0.1mm, rainrate: 0.0mm/h, rainweek: 0.1mm, rainyear: 145.3mm, relbarometer: 1005.1hPa, temp1: 25.3°C, temp2: 15.6°C, uv: 0W/m², uvi: 0.0, wh26_batt: 0, wh26_sig: 4, wh31_ch1_batt: 0, wh31_ch1_sig: 4, wh31_ch2_batt: 0, wh31_ch2_sig: 4, wh40_batt: 0, wh40_sig: 4, wh57_batt: 5, wh57_sig: 4, wh68_batt: 1.8V, wh68_sig: 4, winddir: 88°, windspeed: 1kph
pi@weather:~ $
```

Finally, yes, I do send data to Ecowitt.net.  (But normally have the website set to display the HP3501 device.) I checked and noticed missing data on June 7 and 8 for the GW1000, but the HP3501 device did not have any missing data.  Does this mean the problem is likely with the GW1000?  If so, sorry to have sent you down this rabbit hole.  The strange thing is that everything has been 100% correct all day yesterday and today, so I can't make sense of it.

gjr80, 2021-06-11:
Thanks. I only have a GW1000 and a handful of discrete sensors (WH31, WH32, WH43, WH51 and WH57) so once consoles, 5/7-in-1 and rain gauges get involved I'm a little in the dark.

So what sensor provides your outside temperature/humidity? The `--sensors` output suggests WH26 but I don't think that will be correct, that output is more a limitation of the GW1000 API in that it cannot distinguish (well it provides no means for an API user to distinguish) between a WH26 and any other sensor providing outdoor temp/humidity (I have a WH32 providing outdoor temp/humidity and the GW1000 API says it is a WH26).

The reason I asked about the modified Sensors page was that if modified it would show us a plot of your sensor battery state and signal level. If you haven't made that change (or implemented anything similar in Belchertown) don't worry, that doesn't prove the cause but adds to the picture. 

Checking against Ecowitt.net is useful as it receives it data independent of WeeWX so if we see a corresponding outage there it is suggestive of a hardware problem rather than a WeeWX/GW1000 driver problem. When you say there was missing GW1000 data on 7 and 8 June was that on Ecowitt.net? If so that lends support to the theory the GW1000 lost contact with one or more sensors (I have a WH51 that is some 25metres from my two GW1000 with an internal wall, window and external brick wall in the path and I have periodic lost contact with the sensor. Since it worked faultlessly for weeks I have a theory that once the battery gets down a bit the signal is more unreliable but that is only supposition.).

Another thing you could check is the WS View app if you have it installed. When the outdoor temp/humidity disappear do they also disappear from the WS View app. Also have a look at the More -> Sensors ID page. Is there any signal level showing against your outdoor temp/humidity sensor? (this might be a bit imperfect as the GW1000 signal level is a quasi signal level - it just shows how many of the last four expected packets were received from the sensor). If there are issues on the WS View app then that is also suggestive of a hardware issue.

gjr80, 2021-10-01:
Nothing for over three months, closing now. Can be re-opened if required.

closed 2021-10-01 by gjr80