gjr80/weewx-gw1000 #42: some issues with driver  0.3.1 and weewx 4.5.1

gjr80, 2021-06-11:
some issues with driver  0.3.1 and weewx 4.5.1 too. 

```
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
Jun 10 21:31:43 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000: Unexpected exception occurred while checking response to attempt 1 to send command 'CMD_READ…x out of range
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  Traceback (most recent call last):
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3345, in send_cmd_with_retries
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      self.check_response(response, self.commands[cmd])
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
Jun 10 21:32:07 sandbox python3[5721]: weewx[5721] ERROR user.gw1000:     ****  IndexError: index out of range

```

_Originally posted by @Zaschii in https://github.com/gjr80/weewx-gw1000/issues/41#issuecomment-858986187_

gjr80, 2021-06-11:
Am going to need some more information on this.

Is this repeatable (ie does it happen regularly), sporadic (does it happen irregularly) or a one off occurrence? If repeatable can you edit `weewx.conf`, set `debug = 1`, save `weewx.conf` and restart WeeWX and wait for the error to occur. Then post the WeeWX log from when you restarted WeeWX through until the error occurs. Please don't edit the log extract and make sure you include the full WeeWX startup.

Can you post the output from the following commands:
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --sensors
```
and
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --live-data
```
(depending on your system you might need to replace `python` with `python3` in the above commands)

Zaschii, 2021-06-11:
Thanks for your quick reply.

Yesterday i restarted the gw1000 device, since this reboot the issue didnt occur again.

I will provide you the details if the issue will popup again.

BR

Zaschii, 2021-06-15:
OK errors are back again, so there is a issue with driver 0.3.1 and weewx 4.5.1 and the newest GW1000 firmware.

Error occurs directly after weewx restart, i attached the debug log...

`Jun 15 07:59:12 sandbox systemd[1]: Stopping LSB: weewx weather system...
Jun 15 07:59:12 sandbox weewx[5721] INFO __main__: Received signal TERM (15).
Jun 15 07:59:12 sandbox weewx[5721] INFO weewx.engine: Main loop exiting. Shutting engine down.
Jun 15 07:59:12 sandbox weewx[5721] INFO weewx.engine: Shutting down StdReport thread
Jun 15 07:59:13 sandbox weewx[5721] INFO user.gw1000: Gw1000Collector thread has been terminated
Jun 15 07:59:23 sandbox weewx[5721] ERROR user.gw1000: Unable to shut down Gw1000Collector thread
Jun 15 07:59:23 sandbox weewx[5721] INFO __main__: Terminating weewx version 4.5.1
Jun 15 07:59:27 sandbox weewx[23412]: Stopping weewx weather system: weewx....
Jun 15 07:59:27 sandbox systemd[1]: weewx.service: Succeeded.
Jun 15 07:59:27 sandbox systemd[1]: Stopped LSB: weewx weather system.
Jun 15 07:59:32 sandbox systemd[1]: Starting LSB: weewx weather system...
Jun 15 07:59:32 sandbox weewx[23494] INFO __main__: Initializing weewx version 4.5.1
Jun 15 07:59:32 sandbox weewx[23494] INFO __main__: Using Python 3.8.7 (default, Dec 22 2020, 15:32:53) #012[GCC 8.3.0]
Jun 15 07:59:32 sandbox weewx[23494] INFO __main__: Platform Linux-4.19.0-16-amd64-x86_64-with-glibc2.28
Jun 15 07:59:32 sandbox weewx[23494] INFO __main__: Locale is 'de_DE.UTF-8'
Jun 15 07:59:32 sandbox weewx[23494] INFO __main__: PID file is /var/run/weewx.pid
Jun 15 07:59:32 sandbox weewx[23483]: Starting weewx weather system: weewx.
Jun 15 07:59:32 sandbox systemd[1]: Started LSB: weewx weather system.
Jun 15 07:59:32 sandbox weewx[23497] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Jun 15 07:59:32 sandbox weewx[23497] INFO __main__: Debug is 1
Jun 15 07:59:32 sandbox weewx[23497] DEBUG __main__: Initializing engine
Jun 15 07:59:32 sandbox weewx[23497] INFO weewx.engine: Loading station type GW1000 (user.gw1000)
Jun 15 07:59:33 sandbox weewx[23497] INFO user.gw1000: field map is {'co2': 'co2', 'co2_24h_avg': 'co2_24h_avg', 'dateTime': 'datetime', 'daymaxwind': 'daymaxwind', 'dayRain': 'rainday', 'dewpoint': 'dewpoint', 'extraHumid1': 'humid1', 'extraHumid2': 'humid2', 'extraHumid3': 'humid3', 'extraHumid4': 'humid4', 'extraHumid5': 'humid5', 'extraHumid6': 'humid6', 'extraHumid7': 'humid7', 'extraHumid8': 'humid8', 'extraHumid17': 'humid17', 'extraTemp1': 'temp1', 'extraTemp2': 'temp2', 'extraTemp3': 'temp3', 'extraTemp4': 'temp4', 'extraTemp5': 'temp5', 'extraTemp6': 'temp6', 'extraTemp7': 'temp7', 'extraTemp8': 'temp8', 'extraTemp9': 'temp9', 'extraTemp10': 'temp10', 'extraTemp11': 'temp11', 'extraTemp12': 'temp12', 'extraTemp13': 'temp13', 'extraTemp14': 'temp14', 'extraTemp15': 'temp15', 'extraTemp16': 'temp16', 'extraTemp17': 'temp17', 'heatindex': 'heatindex', 'hourRain': 'rainhour', 'inHumidity': 'inhumid', 'inTemp': 'intemp', 'leafWet1': 'leafwet1', 'leafWet2': 'leafwet2', 'leafWet3': 'leafwet3', 'leafWet4': 'leafwet4', 'leafWet5': 'leafwet5', 'leafWet6': 'leafwet6', 'leafWet7': 'leafwet7', 'leafWet8': 'leafwet8', 'leak1': 'leak1', 'leak2': 'leak2', 'leak3': 'leak3', 'leak4': 'leak4', 'lightning_distance': 'lightningdist', 'lightning_last_det_time': 'lightningdettime', 'lightning_strike_count': 'lightning_strike_count', 'lightningcount': 'lightningcount', 'luminosity': 'light', 'monthRain': 'rainmonth', 'outHumidity': 'outhumid', 'outTemp': 'outtemp', 'pm2_5': 'pm251', 'pm2_51_24h_avg': 'pm251_24h_avg', 'pm2_52': 'pm252', 'pm2_52_24h_avg': 'pm252_24h_avg', 'pm2_53': 'pm253', 'pm2_53_24h_avg': 'pm253_24h_avg', 'pm2_54': 'pm254', 'pm2_54_24h_avg': 'pm254_24h_avg', 'pm2_55': 'pm255', 'pm2_55_24h_avg': 'pm255_24h_avg', 'pm10': 'pm10', 'pm10_24h_avg': 'pm10_24h_avg', 'pressure': 'absbarometer', 'rain': 'rain', 'rainRate': 'rainrate', 'relbarometer': 'relbarometer', 'soilMoist1': 'soilmoist1', 'soilMoist2': 'soilmoist2', 'soilMoist3': 'soilmoist3', 'soilMoist4': 'soilmoist4', 'soilMoist5': 'soilmoist5', 'soilMoist6': 'soilmoist6', 'soilMoist7': 'soilmoist7', 'soilMoist8': 'soilmoist8', 'soilMoist9': 'soilmoist9', 'soilMoist10': 'soilmoist10', 'soilMoist11': 'soilmoist11', 'soilMoist12': 'soilmoist12', 'soilMoist13': 'soilmoist13', 'soilMoist14': 'soilmoist14', 'soilMoist15': 'soilmoist15', 'soilMoist16': 'soilmoist16', 'soilTemp1': 'soiltemp1', 'soilTemp2': 'soiltemp2', 'soilTemp3': 'soiltemp3', 'soilTemp4': 'soiltemp4', 'soilTemp5': 'soiltemp5', 'soilTemp6': 'soiltemp6', 'soilTemp7': 'soiltemp7', 'soilTemp8': 'soiltemp8', 'soilTemp9': 'soiltemp9', 'soilTemp10': 'soiltemp10', 'soilTemp11': 'soiltemp11', 'soilTemp12': 'soiltemp12', 'soilTemp13': 'soiltemp13', 'soilTemp14': 'soiltemp14', 'soilTemp15': 'soiltemp15', 'soilTemp16': 'soiltemp16', 'stormRain': 'rainevent', 'totalRain': 'raintotals', 'UV': 'uvi', 'uvradiation': 'uv', 'weekRain': 'rainweek', 'wh25_batt': 'wh25_batt', 'wh25_sig': 'wh25_sig', 'wh26_batt': 'wh26_batt', 'wh26_sig': 'wh26_sig', 'wh31_ch1_batt': 'wh31_ch1_batt', 'wh31_ch1_sig': 'wh31_ch1_sig', 'wh31_ch2_batt': 'wh31_ch2_batt', 'wh31_ch2_sig': 'wh31_ch2_sig', 'wh31_ch3_batt': 'wh31_ch3_batt', 'wh31_ch3_sig': 'wh31_ch3_sig', 'wh31_ch4_batt': 'wh31_ch4_batt', 'wh31_ch4_sig': 'wh31_ch4_sig', 'wh31_ch5_batt': 'wh31_ch5_batt', 'wh31_ch5_sig': 'wh31_ch5_sig', 'wh31_ch6_batt': 'wh31_ch6_batt', 'wh31_ch6_sig': 'wh31_ch6_sig', 'wh31_ch7_batt': 'wh31_ch7_batt', 'wh31_ch7_sig': 'wh31_ch7_sig', 'wh31_ch8 [... cut ...]
Jun 15 08:03:50 sandbox weewx[23497] ERROR user.gw1000:     ****      self.check_response(response, self.commands[cmd])
Jun 15 08:03:50 sandbox weewx[23497] ERROR user.gw1000:     ****    File "/usr/share/weewx/user/gw1000.py", line 3421, in check_response
Jun 15 08:03:50 sandbox weewx[23497] ERROR user.gw1000:     ****      if six.indexbytes(response, 2) == six.byte2int(cmd_code):
Jun 15 08:03:50 sandbox weewx[23497] ERROR user.gw1000:     ****  IndexError: index out of range
Jun 15 08:03:50 sandbox weewx[23497] DEBUG user.gw1000: Next update in 60 seconds
`

gjr80, 2021-06-15:
Thanks, am going to seen in detail the response from the GW1000. Can you edit `weewx.conf`, set `debug = 3`, save `weewx.conf` and restart WeeWX. This will log the actual hex bytes sent to and received from the GW1000. Hopefully it will misbehave again. Once you get the error again can you post a log extract from startup until the error occurs.

This will put a lot of data in your log so once the error occurs stop WeeWX and revert to `debug = 1` (or 0) then restart WeeWX. If the error does not show up let me know and I will give you an instrumented version of the driver that can be left running as it will just log the GW1000 response when the error occurs.

Zaschii, 2021-06-15:
ok i will prepare the new debug output.

meanwhile here is the output of...

`root@sandbox:/var/log#PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --sensors
Using configuration file /etc/weewx/weewx.conf
debug level is '1'

IP address obtained from station config
Port number obtained from station config

Interrogating GW1000 at 192.168.XXX.XXX:45000

Sensor     Status
WH65       sensor ID: c3  signal: 4  battery: 0 (OK)
WH68       sensor is registering...
WS80       sensor is registering...
WH40       sensor is registering...
WH26       sensor is registering...
WH31 ch1   sensor ID: 53  signal: 4  battery: 0 (OK)
WH31 ch2   sensor is registering...
WH31 ch3   sensor is registering...
WH31 ch4   sensor is registering...
WH31 ch5   sensor is registering...
WH31 ch6   sensor is registering...
WH31 ch7   sensor is registering...
WH31 ch8   sensor is registering...
WH51 ch1   sensor ID: c7c3  signal: 4  battery: 0 (OK)
WH51 ch2   sensor is registering...
WH51 ch3   sensor is registering...
WH51 ch4   sensor is registering...
WH51 ch5   sensor is registering...
WH51 ch6   sensor is registering...
WH51 ch7   sensor is registering...
WH51 ch8   sensor is registering...
WH41 ch1   sensor is registering...
WH41 ch2   sensor is registering...
WH41 ch3   sensor is registering...
WH41 ch4   sensor is registering...
WH57       sensor ID: d946  signal: 4  battery: 4 (OK)
WH55 ch1   sensor is registering...
WH55 ch2   sensor is registering...
WH55 ch3   sensor is registering...
WH55 ch4   sensor is registering...
WH34 ch1   sensor is registering...
WH34 ch2   sensor is registering...
WH34 ch3   sensor is registering...
WH34 ch4   sensor is registering...
WH34 ch5   sensor is registering...
WH34 ch6   sensor is registering...
WH34 ch7   sensor is registering...
WH34 ch8   sensor is registering...
WH45       sensor is registering...
WH35 ch1   sensor is registering...
WH35 ch2   sensor is registering...
WH35 ch3   sensor is registering...
WH35 ch4   sensor is registering...
WH35 ch5   sensor is registering...
WH35 ch6   sensor is registering...
WH35 ch7   sensor is registering...
WH35 ch8   sensor is registering...
`

Zaschii, 2021-06-15:
```
#PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --live-data
Using configuration file /etc/weewx/weewx.conf
debug level is '1'

IP address obtained from station config
Port number obtained from station config

Interrogating GW1000 at 192.168.XXX.XXX:45000

GW1000 live sensor data (2021-06-15 08:47:15 CEST (1623739635)): absbarometer: 967.2hPa, daymaxwind: 5kph, gustspeed: 4kph, humid1: 49%, inhumid: 46%, intemp: 24.6°C, light: 46934lx, lightningcount: 1, lightningdettime: 15-Jun-2021 06:22, lightningdist: 27.0km, outhumid: 51%, outtemp: 21.2°C, rainday: 0.0mm, rainevent: 0.0mm, rainmonth: 26.1mm, rainrate: 0.0mm/h, rainweek: 0.0mm, rainyear: 319.4mm, relbarometer: 967.2hPa, soilmoist1: 67%, temp1: 22.7°C, uv: 125W/m², uvi: 3.0, wh31_ch1_batt: 0, wh31_ch1_sig: 4, wh51_ch1_batt: 0, wh51_ch1_sig: 4, wh57_batt: 4, wh57_sig: 4, wh65_batt: 0, wh65_sig: 4, winddir: 136°, windspeed: 3kph
```

Zaschii, 2021-06-15:
debug output with `debug = 3`

[weewx_issue_debug3.txt](https://github.com/gjr80/weewx-gw1000/files/6653567/weewx_issue_debug3.txt)

gjr80, 2021-06-15:
Thank you. Are you running the GW1000 driver as both a driver and a service?

Zaschii, 2021-06-15:
oh yes, thats possible.... should i remove this line from my conf file:
`data_services = user.gw1000.Gw1000Service`

or what would be the recommend use case?

Zaschii, 2021-06-15:
without this this service line i still have the following errors in the log...

```
Jun 15 10:01:39 sandbox weewx[24448] DEBUG user.gw1000: Failed to obtain response to attempt 1 to send command 'CMD_READ_SENSOR_ID_NEW': timed out
Jun 15 10:01:50 sandbox weewx[24448] DEBUG user.gw1000: Next update in 60 seconds
Jun 15 10:02:40 sandbox weewx[24448] DEBUG user.gw1000: Failed to obtain response to attempt 1 to send command 'CMD_READ_SENSOR_ID_NEW': timed out
Jun 15 10:02:50 sandbox weewx[24448] DEBUG user.gw1000: Next update in 60 seconds
```

gjr80, 2021-06-15:
The only use case I could think of to use both the driver and service would be if you had two GW1000, but then I would recommend running two instances of WeeWX rather than both the driver and service. I have never tested both the driver and service on a single WeeWX instance.

WeeWX must run with a driver but it can only run one driver. You might chose to run the GW1000 driver as a service if you had two (different) stations, for example an Acurite weather station and a GW1000. You would run the Acurite driver as the WeeWX driver and the GW1000 driver as a service. In this arrangement the Acurite driver creates the loop packets and the GW1000 driver (run as a service) augments those Acurite loop packets with data from the GW1000. Not every driver can be run as a driver or as a service, in fact most can only run as a driver.

In your case if the GW1000 is your only 'station' you should run the GW1000 driver as a driver (only). If you have another station and a GW1000 and want to amalgamate the data from both stations in the one WeeWX instance then you will need to run one driver as a driver and the other driver as a service. If your other station's driver cannot run as a service then you have no choice, it must be run as a driver and the GW1000 driver run as a service.

Hope that makes sense. Bottom line is running the GW1000 driver as a driver and service simultaneously on a single WeeWX instance is an untested configuration.

gjr80, 2021-06-15:
> without this this service line i still have the following errors in the log...
>
>
>
> ```
>
> Jun 15 10:01:39 sandbox weewx[24448] DEBUG user.gw1000: Failed to obtain response to attempt 1 to send command [...]

I am guessing your GW1000 firmware is up to date? The CMD_READ_SENSOR_ID_NEW command was only implemented in firmware version 1.5.4. 

Assuming the firmware is up to date can you show me a startup log with debug = 3. Same procedure as before, need to see the log from startup through until the error hits.

Zaschii, 2021-06-15:
Yes GW1000 is up to date, Firmware GW1000A_V1.6.8 is installed. i will recreate the log

gjr80, 2021-06-15:
Yes, but that log was when you were running both the driver and the service and it showed no `Failed to obtain response...` log entries. The startup portion of the log gives me detail on the WeeWX/driver config (which has changed) and the portion of the log in the lead up to the `Failed to obtain response...` will give me the data that was sent to the GW1000 along with any earlier responses that may have been received.

Zaschii, 2021-06-15:
At the moment it looks good... no errors

```
root@sandbox:/home/sasch#/etc/init.d/weewx status
● weewx.service - LSB: weewx weather system
   Loaded: loaded (/etc/init.d/weewx; generated)
   Active: active (running) since Tue 2021-06-15 10:07:31 CEST; 5h 43min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 24574 ExecStart=/etc/init.d/weewx start (code=exited, status=0/SUCCESS)
    Tasks: 3 (limit: 2254)
   Memory: 38.3M
   CGroup: /system.slice/weewx.service
           └─24588 python3 /usr/share/weewx/weewxd --daemon --pidfile=/var/run/weewx.pid /etc/weewx/weewx.conf

Jun 15 15:40:44 sandbox python3[24588]: weewx[24588] INFO weewx.imagegenerator: Generated 15 images for report SeasonsReport in 0.39 seconds
Jun 15 15:40:44 sandbox python3[24588]: weewx[24588] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Jun 15 15:45:59 sandbox python3[24588]: weewx[24588] INFO weewx.restx: MQTT: Published record 2021-06-15 15:45:00 CEST (1623764700)
Jun 15 15:45:59 sandbox python3[24588]: weewx[24588] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 0.64 seconds
Jun 15 15:46:00 sandbox python3[24588]: weewx[24588] INFO weewx.imagegenerator: Generated 15 images for report SeasonsReport in 0.41 seconds
Jun 15 15:46:00 sandbox python3[24588]: weewx[24588] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx
Jun 15 15:51:02 sandbox python3[24588]: weewx[24588] INFO weewx.restx: MQTT: Published record 2021-06-15 15:50:00 CEST (1623765000)
Jun 15 15:51:02 sandbox python3[24588]: weewx[24588] INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 0.64 seconds
Jun 15 15:51:03 sandbox python3[24588]: weewx[24588] INFO weewx.imagegenerator: Generated 15 images for report SeasonsReport in 0.39 seconds
Jun 15 15:51:03 sandbox python3[24588]: weewx[24588] INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx

```

gjr80, 2021-06-16:
OK. I will close this. If it recurs you can re-open this issue.

closed 2021-06-16 by gjr80