gjr80/weewx-gw1000 #43: Trouble timeout reading GW1000: 0.31 driver, firmware GW1000B_V1.6.8

tedroche, 2021-06-16:
Last week I updated firmware to 1.6.8. No other software changes. Over the weekend, weewx started failing to read through the GW1000 driver with timeout error messages. (Website: https://iayft.com/weewx/index.html) Updated driver from previous version to 0.3.1, powered down and restarted GW1000, network router. I can access GW1000 consistently through WSView, and no errors posted from GW1000 to Ecowitt.net and Wunderground (NHHOPKI9). weewx is typically running as a service on an Ubuntu 18.04 LTS server, Python 2.7.17, tried to run it as a daemon and it "seemed" to run longer, but still timeouts, anecdotally. 

Following along with the several other threads posted here, I updated the retry count to 10, set debug to 1 and ran to see if I could add to the symptoms you've seen. It ran without issue for about 12 hours (longest in the three days) and then died. Attached are sensors, debug set to 1, live data, and logs

Sensors:
Using configuration file /etc/weewx/weewx.conf
debug level is '1'

IP address obtained from station config
Port number obtained from station config

Interrogating GW1000 at 192.168.1.248:45000

Sensor     Status
WH68       sensor ID: 1bb  signal: 4  battery: 1.86 (OK)
WH65       sensor is registering...
WH40       sensor ID: fa2b  signal: 4  battery: 0 (OK)
WS80       sensor is registering...
WH26       sensor ID: a1  signal: 4  battery: 0 (OK)
WH31 ch2   sensor is registering...
WH31 ch1   sensor ID: 5e  signal: 4  battery: 0 (OK)
WH31 ch4   sensor is registering...
WH31 ch3   sensor is registering...
WH31 ch6   sensor is registering...
WH31 ch5   sensor is registering...
WH31 ch8   sensor is registering...
WH31 ch7   sensor is registering...
WH51 ch2   sensor is registering...
WH51 ch1   sensor is registering...
WH51 ch4   sensor is registering...
WH51 ch3   sensor is registering...
WH51 ch6   sensor is registering...
WH51 ch5   sensor is registering...
WH51 ch8   sensor is registering...
WH51 ch7   sensor is registering...
WH41 ch2   sensor is registering...
WH41 ch1   sensor ID: a  signal: 4  battery: 4 (OK)
WH41 ch4   sensor is registering...
WH41 ch3   sensor is registering...
WH55 ch1   sensor is registering...
WH57       sensor is registering...
WH55 ch3   sensor is registering...
WH55 ch2   sensor is registering...
WH34 ch1   sensor is registering...
WH55 ch4   sensor is registering...
WH34 ch3   sensor is registering...
WH34 ch2   sensor is registering...
WH34 ch5   sensor is registering...
WH34 ch4   sensor is registering...
WH34 ch7   sensor is registering...
WH34 ch6   sensor is registering...
WH45       sensor is registering...
WH34 ch8   sensor is registering...
WH35 ch2   sensor is registering...
WH35 ch1   sensor is registering...
WH35 ch4   sensor is registering...
WH35 ch3   sensor is registering...
WH35 ch6   sensor is registering...
WH35 ch5   sensor is registering...
WH35 ch8   sensor is registering...
WH35 ch7   sensor is registering...

Live Data:

# PYTHONPATH=/usr/share/weewx python -m user.gw1000 --live-data
Using configuration file /etc/weewx/weewx.conf
debug level is '1'

IP address obtained from station config
Port number obtained from station config

Interrogating GW1000 at 192.168.1.248:45000

GW1000 live sensor data (2021-06-15 09:15:06 EDT (1623762906)): absbarometer: 992.5hPa, daymaxwind: 4kph, gustspeed: 2kph, humid1: 68%, inhumid: 72%, intemp: 20.2°C, light: 19680lx, outhumid: 89%, outtemp: 19.4°C, pm251: 8.0µg/m³, pm251_24h_avg: 4.1µg/m³, rainday: 0.2mm, rainevent: 0.0mm, rainmonth: 25.6mm, rainrate: 0.0mm/h, rainweek: 8.0mm, rainyear: 182.9mm, relbarometer: 1007.4hPa, temp1: 21.1°C, uv: 10W/m², uvi: 1.0, wh26_batt: 0, wh26_sig: 4, wh31_ch1_batt: 0, wh31_ch1_sig: 4, wh40_batt: 0, wh40_sig: 4, wh41_ch1_batt: 4, wh41_ch1_sig: 4, wh68_batt: 1.9V, wh68_sig: 4, winddir: 314°, windspeed: 1kph

Logs (generated with journalctl --unit=weewx --lines 12551 >~/weewxlog.txt)

[weewxlog.txt](https://github.com/gjr80/weewx-gw1000/files/6662265/weewxlog.txt)

gjr80, 2021-06-17:
Each of those 'failed to obtain a response' log entries is a result of the driver interrogating the GW1000 API and getting no response, there are some 350 occurrences in the 12 hours covered by your log extract. On a timeout such as this the driver tries again after a short delay (you have the driver trying up to ten times with a ten second delay). The driver is almost always (except for the very end of the log extract) able to obtain a response on the 2nd or subsequent attempts. There are some errors where a response is received but it fails a validity check, likely because it is incomplete or somehow corrupt.

The fact that Ecowitt.net is OK is not surprising, the GW1000 uses an entirely different mechanism for updating Ecowitt.net; the GW1000 pushes data to Ecowitt.net whereas the GW1000 driver has to interrogate the GW1000. Not surprising that WU is OK as well, you have WeeWX updating WU and WeeWX is getting data from the GW1000 and synthesising archive records which are then used to update WU every 5 minutes (the driver is interrogating the GW1000 every 20 seconds or 15 times every five minutes, WeeWX only needs one of these interrogations to successfully deliver data for an archive record to be created). I believe (not sure) that the WS View app uses the same API as the GW1000 driver to obtain data from the GW1000. The difference with the WS View app is that you do not see any failures to obtain data from the GW1000, as per the log extract it is not a case of WeeWX not getting any data from the GW1000, its just a case of WeeWX not getting some data from the GW1000 (well at least that was the case until 9pm).

The ultimate failure at the end of the log extract was due to an interrogation of the GW1000 failing to obtain a response after 10 attempts, this caused WeeWX to effectively shutdown and attempt a restart in 60 seconds but during that restart the GW1000 could not be contacted at all and thus WeeWX restarts again and the process repeats. It is possible WeeWX would have eventually picked up the GW1000 but it really should not get to that.

I don't see a 20 second interval as being too short, I have two GW1000 that I run on 20 second interval with no problems.

It looks as it if the situation gets progressively worse, I am leaning towards thinking there are some issues on your network. What does your network topology look like? Is there anything else contacting the GW1000?

tedroche, 2021-06-17:
Thanks for the thorough response. As this is unlikely a driver issue, let's close this ticket, and I'll pursue issues with the network. I'm running a small home office and home off an ISP-supplied "modem" router-wap and have 5 computers, a gigabit backbone with 3 switches, a powerline bridge, 2 RaspPies, 3 smartphones, 2 printers, 2 Chromecasts and who knows what else snuck in over night. I've moved the GW1000 closer to the WAP and have seen zero timeouts in the first half-hour. Thanks for pointing me in the right direction.

closed 2021-06-17 by tedroche