gjr80/weewx-gw1000 #46: Something goes wrong with gw1100 and 2.0.4 firmware

AlexCastell, 2021-09-22:
I've a new  GW1100A with V2.0.4 firmware, tonight i've the following errors and weewx stop working. 
Can any one help me ?
Many Thanks
Alex 

PYTHONPATH=/usr/share/weewx python -m user.gw1000 --live-data
Interrogating GW1000 at 10.100.200.23:45000
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/share/weewx/user/gw1000.py", line 5906, in <module>
    main()
  File "/usr/share/weewx/user/gw1000.py", line 5900, in main
    direct_gw100.process_options()
  File "/usr/share/weewx/user/gw1000.py", line 4860, in process_options
    self.live_data()
  File "/usr/share/weewx/user/gw1000.py", line 5497, in live_data
    live_sensor_data_dict = collector.get_live_sensor_data()
  File "/usr/share/weewx/user/gw1000.py", line 2329, in get_live_sensor_data
    self.update_sensor_id_data()
  File "/usr/share/weewx/user/gw1000.py", line 2344, in update_sensor_id_data
    self.sensors_obj.set_sensor_id_data(sensor_id_data)
  File "/usr/share/weewx/user/gw1000.py", line 4108, in set_sensor_id_data
    batt_fn = Gw1000Collector.sensor_ids[data[index:index + 1]]['batt_fn']
KeyError: '0'

PYTHONPATH=/usr/share/weewx python -m user.gw1000 --sensors
Interrogating GW1000 at 10.100.200.23:45000
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/share/weewx/user/gw1000.py", line 5906, in <module>
    main()
  File "/usr/share/weewx/user/gw1000.py", line 5900, in main
    direct_gw100.process_options()
  File "/usr/share/weewx/user/gw1000.py", line 4858, in process_options
    self.sensors()
  File "/usr/share/weewx/user/gw1000.py", line 5432, in sensors
    collector.update_sensor_id_data()
  File "/usr/share/weewx/user/gw1000.py", line 2344, in update_sensor_id_data
    self.sensors_obj.set_sensor_id_data(sensor_id_data)
  File "/usr/share/weewx/user/gw1000.py", line 4108, in set_sensor_id_data
    batt_fn = Gw1000Collector.sensor_ids[data[index:index + 1]]['batt_fn']
KeyError: '0'

gjr80, 2021-09-25:
I am getting to this but unfortunately a bit short of time until Monday. I did a few quick checks with my GW1100 today and I had no problems, though I believe it is on firmware v2.0.2 so that could be a factor. In the meantime could I get you to try a few things for me and post some log extracts. Could I get you to try the following commands:
```
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --discover --debug=3
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --firmware-version --debug=3
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --sensors --debug=3
```

I suspect (hope) the first two commands will work without error. The addition of `--debug=3` will log the hexadecimal bytes sent to and received from the GW1100, so whilst the last command will likely fail hopefully the log will provide some useful data.

Once you have run the above commands can you post a copy of the console output showing the command run and the result as well as the WeeWX log covering the period where you ran the commands, the relevant portion of the log should be fairly obvious as it will (should) contain many pairs of hex characters.

Thanks.

gjr80, 2021-09-25:
Disregard my last request. It is a v2.0.4 firmware change that broke the driver, I think I have identified the issue and have fixed it in (the yet to be released) v0.3.2 of the GW1000 driver. Could you try downloading v0.3.2 using:
```
$wget -P /var/tmp https://raw.githubusercontent.com/gjr80/weewx-gw1000/master/bin/user/gw1000.py
```
Copy `/var/tmp/gw1000.py` in place of your existing driver file in the `/usr/share/weewx/user` directory and you should be good to go.

AlexCastell, 2021-09-25:
Many thanks.  The problem is solved succesfully.

closed 2021-09-25 by AlexCastell

gjr80, 2021-09-27:
Just to tie this off the [GW1000 driver v0.4.0](https://github.com/gjr80/weewx-gw1000/releases) was released today. This release includes, among other things, the fix for operating with the GW1100 v2.0.4 firmware.

AlexCastell, 2021-09-27:
I've installed it successfully ... now i'm running 0.4.0 without any problem

Many thanks
Alex