gjr80/weewx-gw1000 #50: CMD_READ_STATION_MAC

danitus13, 2021-12-11:
Nella guida fino al punto 6, quando faccio il test del drive, tutto funziona e la stazione scaricare correttamente i dati. 
Quando vado a riconfigurare weewx.conf, inizia a uscire questo errore CMD_READ_STATION_MAC
E in loop, ogni 60sec, continua a uscire lo stesso errore. 
Cosa posso fare? 
Grazie

gjr80, 2021-12-11:
Non è molto per continuare. Modifica `weewx.conf` e imposta `debug=1`. Salva `weewx.conf` e riavvia WeeWX. Lascia che WeeWX venga eseguito per almeno un intervallo di archiviazione e poi prendi un estratto dal registro di WeeWX che copra l'intero avvio di WeeWX fino al termine dell'intervallo di archiviazione. È importante acquisire l'intero avvio di WeeWX nell'estratto del registro. Posta qui l'estratto del log.

That is not very much to go on. Please edit `weewx.conf` and set `debug=1`. Save `weewx.conf` and restart WeeWX. Let WeeWX run for at least one archive interval and then take an extract from the WeeWX log covering the full WeeWX startup through until the archive interval passed. It is important to capture the full WeeWX startup in the log extract. Post the log extract here.

danitus13, 2021-12-11:
[image]

gjr80, 2021-12-12:
Non è proprio quello che ho chiesto, ma posso vedere abbastanza per dire che WeeWX non è in grado di contattare il GW1000 durante l'avvio e quindi non può caricare il driver.  Per impostazione predefinita, WeeWX tenta di ricaricare il driver ogni 60 secondi.

Puoi riavviare WeeWX e fornire un estratto del registro che mostri l'avvio completo di WeeWX (questo contiene importanti informazioni di configurazione) e il messaggio di errore.  Si prega di non pubblicare uno screenshot, è difficile da leggere sui dispositivi mobili e non mostra abbastanza del registro.  Se hai bisogno di aiuto per acquisire un buon estratto di log, fai riferimento alla seguente voce wiki WeeWX (https://github.com/weewx/weewx/wiki/Help!-Posting-to-weewx-user).

Infine, hai indicato di essere riuscito a contattare il GW1000 e ottenere i dati.  Potresti riprovare e postare il comando esatto che hai usato e la risposta ricevuta dal GW1000.

Not quite what I asked for but I can see enough to tell that WeeWX is unable to contact the GW1000 during startup and hence cannot load the driver. By defult WeeWX attempts to relod the driver every 60 seconds.

Can you please restart WeeWX and provide a log extract that shows the full WeeWX startup (this contains important config information) and the error message. Please do not post a screen shot, it is difficult to read on mobile devices and it does not show enough of the log. If you need help to capture a good log extract please refer to the following WeeWX wiki entry (https://github.com/weewx/weewx/wiki/Help!-Posting-to-weewx-user).

Finally, you indicated that you were able to successfully contact the GW1000 and obtain data. Could you please try that again and post the exact command you used and the response received from the GW1000.

danitus13, 2021-12-12:
Questo il log

`Dec 12 00:04:54 raspberrypi systemd[1]: Starting LSB: weewx weather system...
Dec 12 00:04:55 raspberrypi weewx[4518] INFO __main__: Initializing weewx version 4.5.1
Dec 12 00:04:55 raspberrypi weewx[4518] INFO __main__: Using Python 3.9.2 (default, Mar 12 2021, 04:06:34) #012[GCC 10.2.1 20210110]
Dec 12 00:04:55 raspberrypi weewx[4518] INFO __main__: Platform Linux-5.10.63-v7+-armv7l-with-glibc2.31
Dec 12 00:04:55 raspberrypi weewx[4518] INFO __main__: Locale is 'it_IT.UTF-8'
Dec 12 00:04:55 raspberrypi weewx[4518] INFO __main__: PID file is /var/run/weewx.pid
Dec 12 00:04:55 raspberrypi weewx[4521] INFO __main__: Using configuration file /etc/weewx/weewx.conf
Dec 12 00:04:55 raspberrypi weewx[4521] INFO __main__: Debug is 1
Dec 12 00:04:55 raspberrypi weewx[4521] DEBUG __main__: Initializing engine
Dec 12 00:04:55 raspberrypi weewx[4521] INFO weewx.engine: Loading station type GW1000 (user.gw1000)
Dec 12 00:04:55 raspberrypi weewx[4506]: Starting weewx weather system: weewx.
Dec 12 00:04:55 raspberrypi systemd[1]: Started LSB: weewx weather system.
Dec 12 00:04:55 raspberrypi weewx[4521] DEBUG user.gw1000: Failed attempt 1 to send command 'CMD_READ_STATION_MAC': [Errno 111] Connessione rifiutata
Dec 12 00:05:05 raspberrypi weewx[4521] DEBUG user.gw1000: Failed attempt 2 to send command 'CMD_READ_STATION_MAC': [Errno 111] Connessione rifiutata
Dec 12 00:05:15 raspberrypi weewx[4521] DEBUG user.gw1000: Failed attempt 3 to send command 'CMD_READ_STATION_MAC': [Errno 111] Connessione rifiutata
Dec 12 00:05:15 raspberrypi weewx[4521] ERROR user.gw1000: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts
Dec 12 00:05:15 raspberrypi weewx[4521] ERROR weewx.engine: Import of driver failed: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts (<class 'user.gw1000.GW1000IOError'>)
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 119, in setupStation
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 1708, in loader
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      return Gw1000Driver(**config_dict[DRIVER_NAME])
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 2084, in __init__
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      super(Gw1000Driver, self).__init__(**stn_dict)
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 1072, in __init__
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      self.collector = Gw1000Collector(ip_address=self.ip_address,
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 2427, in __init__
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      self.station = Gw1000Collector.Station(ip_address=ip_address,
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 3163, in __init__
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      self.mac = self.get_mac_address()
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 3557, in get_mac_address
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      return self.send_cmd_with_retries('CMD_READ_STATION_MAC')
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 3740, in send_cmd_with_retries
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****      raise GW1000IOError(_msg)
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL weewx.engine:     ****  user.gw1000.GW1000IOError: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL __main__: Unable to load driver: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts
Dec 12 00:05:15 raspberrypi weewx[4521] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...`

danitus13, 2021-12-12:
Quando ho seguito la guida per l'installazione del drive, arrivato al passaggio 5, dove si fa il test-drive con il comando
sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver

il test-driver mi restitutiva le stringhe di dati della stazione meteo, quindi funzionava. 

Quando poi ho ho eseguito il punto 6 e 8 della guida e lanciato weewx, anche il test-drive mi restituisce l'errore.

danitus13, 2021-12-12:
Non posso mandare il test-driver, perché ora mi esce l'errore. 
Ma la prima volta mi uscivano correttamente i dati della stazione, posso mandare solo l'immagine.
[image]

danitus13, 2021-12-12:
Ho disinstallato weewx e installato nuovamente. 
Arrivato al punto 5 e al test del drive, tutto funziona. 
Il problema è quando riconfiguro weewx e lancio weewx.

Questo il test del drive la prima volta

sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver
Using configuration file /etc/weewx/weewx.conf

Interrogating None at 192.168.X.XXX:45000

2021-12-12 12:43:57 CET (1639309437): UV: 0, dateTime: 1639309437, dayRain: 0.0, daymaxwind: 0.0, inHumidity: 53, inTemp: 20.0, luminosity: 0.0, monthRain: 0.0, outHumidity: 53, outTemp: 19.8, pressure: 960.1, rain: None, rainRate: 0.0, relbarometer: 1046.9, stormRain: 0.0, usUnits: 17, uvradiation: 0.0, weekRain: 0.0, wh25_batt: 0, wh25_sig: 4, wh65_batt: 0, wh65_sig: 4, windDir: 356, windGust: 0.0, windSpeed: 0.0, yearRain: 0.0
2021-12-12 12:44:17 CET (1639309457): UV: 0, dateTime: 1639309457, dayRain: 0.0, daymaxwind: 0.0, inHumidity: 53, inTemp: 20.0, luminosity: 0.0, monthRain: 0.0, outHumidity: 53, outTemp: 19.8, pressure: 960.1, rain: 0.0, rainRate: 0.0, relbarometer: 1046.9, stormRain: 0.0, usUnits: 17, uvradiation: 0.0, weekRain: 0.0, wh25_batt: 0, wh25_sig: 4, wh65_batt: 0, wh65_sig: 4, windDir: 356, windGust: 0.0, windSpeed: 0.0, yearRain: 0.0

gjr80, 2021-12-12:
Grazie. Quindi capisco la situazione come segue:

1. hai installato e configurato il driver GW1000

2. hai testato il driver usando il seguente comando:

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver
    
    e quel comando è stato eseguito senza errori e ha prodotto dati validi dal GW1000
    
3.  hai quindi riconfigurato WeeWX per utilizzare il driver GW1000 con il comando:

    sudo wee_config --reconfigure --driver=user.gw1000
    
4.  hai avviato WeeWX e successivamente hai ricevuto l'errore CMD_READ_STATION_MAC continuo

5.  dopo aver chiesto l'output --test-driver che hai tentato di eseguire

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver
    
    di nuovo ma questa volta ha dato un errore.
    
6.  hai reinstallato WeeWX e il driver GW1000 e ora il comando --test-driver funziona senza errori (e hai fornito l'output di questo comando con il GW1000 che ha fornito correttamente i dati)
    
7.  riconfigurando nuovamente WeeWX per utilizzare il driver GW1000 si ottiene l'errore CMD_READ_STATION_MAC originale
    
La mia comprensione è corretta?

Puoi fornire qualche informazione in più?:

1. Invia una copia igienizzata del tuo weewx.conf. Ci sono due modi per farlo. Puoi fare una copia del tuo weewx.conf esistente e offuscare tutte le informazioni sensibili; ad es. nomi utente, chiavi API password ecc. In alternativa è possibile utilizzare l'utilità WeeWX [wee_debug](http://weewx.com/docs/utilities.htm#wee_debug_utility) per generare una copia disinfettata di weewx.conf, è possibile controllare l'output di wee_debug (è buono ma non perfetto nell'offuscare informazioni sensibili) e pubblicare l'output di wee_debug.

2. Per favore posta l'output (sia out to screen che nel log WeeWX) dei seguenti comandi:

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --discover --debug=3
    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --mac-address --debug=3
    
    assicurati che l'acquisizione del testo dello schermo includa il comando che hai inserito
    
Non è necessario reinstallare WeeWX, questo è molto probabilmente un problema di configurazione che può essere risolto senza ulteriori reinstallazioni.

Thank you. So I understand the situation as follows:

1. you installed and configured the GW1000 driver

2. you tested the driver using the following command:

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver
    
    and that command executed without error and produced valid data from the GW1000
    
3.  you then re-configured WeeWX to use the GW1000 driver with the command:

    sudo wee_config --reconfigure --driver=user.gw1000
    
4.  you started WeeWX and subsequently received the continuous CMD_READ_STATION_MAC error

5.  after I asked for the --test-driver output you attempted to run

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --test-driver
    
    again but this time it gave an error.
    
6.  you re-installed WeeWX and the GW1000 driver and now the --test-driver command works without error (and you provided the output from this command with the GW1000 successfully providing data)
    
7.  re-configuring WeeWX again to use the GW1000 driver results in the original CMD_READ_STATION_MAC error
    
Is my understanding correct?

Can you provide some more information:

1. Please post a sanitised copy of your weewx.conf. There are two ways you can do this. You can make a copy of your existsing weewx.conf and obfuscate all of the sensitive information; eg user names, passwords API keys etc. Alternatively you can use the WeeWX [wee_debug](http://weewx.com/docs/utilities.htm#wee_debug_utility) utility to generate a sanitised copy of weewx.conf, you can check the wee_debug output (it's good but not perfect in obfuscating sensitive information) and post the wee_debug output.

2. Please post the output (both the out to screen and in the WeeWX log) of the following commands:

    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --discover --debug=3
    sudo PYTHONPATH=/usr/share/weewx python -m user.gw1000 --mac-address --debug=3
    
    please ensure your screen text capture includes the command you entered
    
There is no need to re-install WeeWX, this is most likely a config issue that can be solved without any more re-installing.

danitus13, 2021-12-18:
Ciao sono riuscito a far funzionare weewx.
Ho aggiunto manualmente su weewx.conf l'IP del GW1000, e ha iniziato a scaricare automaticamente i dati.
Grazie per l'aiuto.

closed 2021-12-18 by gjr80