gjr80/weewx-gw1000 #52: --show-all-batt commands returns help menu - Running v0.4.1

Rayvenhaus, 2022-01-10:
[image]

As shown in the above screen capture, issuing a --show-all-batt command displays the "unknown" command response.

gjr80, 2022-01-10:
Correct, that is intended behaviour. --show-all-batt is not a command in itself, rather it is used in conjunction with the --live-data command to display battery data for sensors for which signal level is zero (if signal level is zero that indicates the gw1000 is no longer in contact with that sensor so it is quite likely the battery state data is meaningless-this is the default approach used by the driver-when run from the command line this behaviour can be overridden by using --show-all-batt).

Rayvenhaus, 2022-01-10:
Thank you for the quick response and explanation.

closed 2022-01-10 by Rayvenhaus