gjr80/weewx-gw1000 #53: extraHumid3 causes ImageGenerator error

ajay10000, 2022-01-17:
Hi Gary, thanks for this solid gw1000 driver and your thorough explanations!

I'm not sure if this is a driver, weewx or me issue, but I suspect it's not the driver and I'm hoping you can confirm.  I have three WH31 temp/humidity sensors.  I'm running WeeWx v4.5.1 in docker, which explains the strange paths in the following excerpts:

Error shown in tail -f /var/log/messages:
```Jan 17 12:00:59 29390b92a6c7 user.info weewx[29] INFO weewx.engine: Starting main packet loop.
Jan 17 12:00:59 29390b92a6c7 user.info weewx[29] INFO user.gw1000: Using 'rainyear' for rain total
Jan 17 12:00:59 29390b92a6c7 user.info weewx[29] INFO user.gw1000: skipping rain measurement of 24.4: no last rain
Jan 17 12:06:02 29390b92a6c7 user.info weewx[29] INFO weewx.manager: Added record 2022-01-17 12:05:00 AEDT (1642381500) to database 'weewx.sdb'
Jan 17 12:06:02 29390b92a6c7 user.info weewx[29] INFO weewx.manager: Added record 2022-01-17 12:05:00 AEDT (1642381500) to daily summary in 'weewx.sdb'
Jan 17 12:06:03 29390b92a6c7 user.info weewx[29] INFO weewx.cheetahgenerator: Generated 9 files for report SeasonsReport in 1.44 seconds
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine: Caught unrecoverable exception in generator 'weewx.imagegenerator.ImageGenerator'
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****  extraHumid3
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****  Traceback (most recent call last):
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 196, in run
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****      obj.start()
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/reportengine.py", line 281, in start
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****      self.run()
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/imagegenerator.py", line 41, in run
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****      self.genImages(self.gen_ts)
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/imagegenerator.py", line 181, in genImages
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****      aggregate_interval=aggregate_interval)
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****    File "/home/weewx/bin/weewx/xtypes.py", line 94, in get_series
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****      raise weewx.UnknownType(obs_type)
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****  weewx.UnknownType: extraHumid3
Jan 17 12:06:04 29390b92a6c7 user.err weewx[29] ERROR weewx.reportengine:         ****  Generator terminated
Jan 17 12:06:04 29390b92a6c7 user.info weewx[29] INFO weewx.reportengine: Copied 5 files to /home/weewx/public_html`
```
driver reports (humid3):
```$ PYTHONPATH=/home/weewx/bin python -m user.gw1000 --live-data
Using configuration file /home/weewx/weewx.conf
Interrogating GW1000 at 192.168.25.92:45000
GW1000 live sensor data (2022-01-17 14:10:34 AEDT (1642389034)): absbarometer: 991.7 mbar, daymaxwind: 17 kph, gustspeed: 2 kph, humid1: 69%, humid2: 39%, humid3: 57%, inhumid: 63%, intemp: 23.6°C, light: 35169 lx, outhumid: 67%, outtemp: 20.8°C, rainday: 0.00 cm, rainevent: 0.00 cm, rainmonth: 2.44 cm, rainrate: 0.00 cm/h, rainweek: 0.00 cm, rainyear: 2.44 cm, relbarometer: 1011.7 mbar, soilmoist3: 52%, temp1: 22.6°C, temp2: 33.2°C, temp3: 24.7°C, uv: 87 W/m², uvi: 2.0, wh31_ch1_batt: 0, wh31_ch1_sig: 4, wh31_ch2_batt: 0, wh31_ch2_sig: 4, wh31_ch3_batt: 0, wh31_ch3_sig: 4, wh51_ch1_batt: None, wh51_ch1_sig: 0, wh51_ch3_batt: 1, wh51_ch3_sig: 4, wh65_batt: 0, wh65_sig: 4, winddir: 303°, windspeed: 0 kph
```
In weewx.conf:
```
             # Generic labels, keyed by an observation type.
            [[[[Generic]]]]                
                extraHumid2 = Greenhouse Humidity
                extraHumid3 = Orchard Humidity
```
In skin.conf:
```
         [[[dayhum2]]]
            [[[[extraHumid2]]]]
        [[[dayhum3]]]
            [[[[extraHumid3]]]]
```
In index.html.tmpl:
```
            <img src="dayhum2.png"      alt="$obs.label.extraHumid2" />
            <img src="dayhum3.png"      alt="$obs.label.extraHumid3" />
```
On the weewx webpage, extraHumid3 is reported correctly in Current Conditions:
```
Greenhouse Humidity | 38%
Orchard Humidity | 58%
```
However history on the same page is not updated due to the error.  extraHumid1, extraHumid2 and extraTemp3 are not causing an error.

gjr80, 2022-01-17:
Hi Andrew, what schema does your database use? It sounds like you may be using the legacy wview schema that only includes extraHumid1 and extraHumid2. Have a look in weewx.conf under [DataBindings] [[wx_binding]], what is the schema setting?

Gary

ajay10000, 2022-01-17:
Brilliant Gary, that was it.  I have been using weewx for a long time, so my weewx.conf hadn't been updated in a while.  Sorry to raise it as an issue, but many thanks for your help.
`schema = schemas.wview.schema`        
changed to:
`schema = schemas.wview_extended.schema`        
as per https://github.com/weewx/weewx/wiki/Switching-to-the-new-wview_extended-schema

Cheers - Andrew

closed 2022-01-17 by ajay10000