gjr80/weewx-gw1000 #54: Error when attempting to test the driver

Rayvenhaus, 2022-03-04:
When issuing the following command from the command line:
sudo PYTHONPATH=/home/weewx/bin/weewx python3 -m user.gw1000 --test-driver

the following error is received:
/usr/bin/python3: Error while finding module specification for 'user.gw1000' (ModuleNotFoundError: No module named 'user')

Although the driver is installed and appears to be working just fine with WeeWX

System: Raspberry PI 4B
OS: Raspberry OS Bullseye

gjr80, 2022-03-04:
If you have a standard WeeWX install then your `PYTHONPATH` is wrong, the command should be:
```
$ sudo PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --test-driver
```
I did a quick search of the GW1000 driver and wiki and could not find the `PYTHONPATH` value you have used, if you found it in the driver or wiki could you let me know where it is and I will fix it.

Rayvenhaus, 2022-03-04:
That was it. I must have made a typo when transferring the command to my notes. Sorry for the erroneous bug report.

closed 2022-03-04 by Rayvenhaus