gjr80/weewx-gw1000 #56: New firmware upgrade seems to have created a problem

steepleian, 2022-04-29:
Gary,

New firmware for GW1000 seems to have caused and issue. Output from syslog below. Tried a complete reset of the device back to factory settings and re-configuration but error persists. Any assistance would be appreciated.
Thanks,
Ian

Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****  Traceback (most recent call last):
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 2802, in run
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****      self.client.collect_sensor_data()
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 2277, in collect_sensor_data
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****      queue_data = self.get_live_sensor_data()
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 2324, in get_live_sensor_data
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****      self.update_sensor_id_data()
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 2339, in update_sensor_id_data
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****      self.sensors_obj.set_sensor_id_data(sensor_id_data)
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****    File "/home/weewx/bin/user/gw1000.py", line 4083, in set_sensor_id_data
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****      batt_fn = Gw1000Collector.sensor_ids[data[index:index + 1]]['batt_fn']
Apr 29 08:07:08 weatherpi4 weewx[3937] CRITICAL user.gw1000:     ****  KeyError: b'0'

steepleian, 2022-04-29:
Gary,
Well that has got to be the quickest fix ever!

I just spotted that you had updated the gw1000.py code in the api_1_6_4 branch. Took a 'flyer' on that and it solved the problem. When back to the previous version and the error returned. So I am now running with the api_1_6_4 as default.

However, would suggest that other users who have the same problem wait until the main branch is updated.

Thanks,
Ian

closed 2022-04-29 by steepleian

gjr80, 2022-04-29:
Ian, 
I think you will find the problem was an old version of the driver. The stack trace you posted shows code from the v0.3.0, current release is 0.4.2. Your trace looks near identical to [this weewx-user post](https://groups.google.com/g/weewx-user/c/lKw9GabD6F4/m/gpUZpT3sAwAJ) from yesterday, this guy was running v0.3.1 and he self solved by upgrading to 0.4.2 before I was able to comment.

Not sure what version you downloaded but if it was from the api_1_6_4 branch beware, that branch is still being developed. It should be fairly stable but I make no guarantees.

Gary

steepleian, 2022-04-30:
Gary,
I am deliberatly using the api_1_6_4 branch and understanding the risks. The good news from my point of view is that it appears pretty stable (famous last words)!
Ian