gjr80/weewx-gw1000 #57: wh_40 batt status: decimal position issue? tens vs. ones.

W0CHP, 2022-05-23:
The WH_40 battery volts is being returned with the volt decimal in tens. Example sensor data output:

```
Sensor     Status
WH65       sensor is registering...
WH68       sensor ID: 384  signal: 4  battery: 1.68 (OK)
WS80       sensor is registering...
WH40       sensor ID: e258  signal: 4  battery: 14.2  (OK)
WH25       sensor is registering...
WH26       sensor ID: ad  signal: 4  battery: 0 (OK)
WH31 ch1   sensor is registering...
```

Example live data snippet:
`'wh40_batt': '14.20', 'wh40_sig': '4.000000',`

W0CHP, 2022-05-26:
> 
> Could I ask you to run the driver directly with the following command and post the resulting WeeWX log entries (no need to stop WeeWX if it is running). For a setup.py install:
>

Absolutely! I now feel quite foolish not furnishing these details from the get-go. :-/

```
sudo PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --live-data --debug=3
Using configuration file /etc/weewx/weewx.conf
debug level is '3'

IP address obtained from station config
Port number obtained from station config
Battery state filtering ('False') using the default

Interrogating GW1100 at 10.1.10.250:45000

GW1100 live sensor data (2022-05-26 17:35:20 CDT (1653604520)): 'absbarometer': '968.600000', 'annual_reset': '3.000000', 'day_reset': '0.000000', 'daymaxwind': '11.160000', 'gustspeed': '5.400000', 'inhumid': '49.000000', 'intemp': '21.700000', 'light': '3560.000000', 'outhumid': '91.000000', 'outtemp': '10.900000', 'relbarometer': '1010.900000', 't_rainday': '0.100000', 't_rainevent': '0.000000', 't_rainhour': '1.000000', 't_rainmonth': '7.670000', 't_rainrate': '0.000000', 't_rainweek': '4.320000', 't_rainyear': '7.670000', 'uv': '0.000000', 'uvi': '0.000000', 'week_reset': '0.000000', 'wh26_batt': '0.000000', 'wh26_sig': '4.000000', 'wh40_batt': '14.20', 'wh40_sig': '4.000000', 'wh68_batt': '1.66', 'wh68_sig': '3.000000', 'winddir': '11.000000', 'windspeed': '3.240000'
```
> Could you also give me some details on the Ecowitt Gateway device you are using? Is it a GW1000/1100/2000? What firmware version is it using? Also how old (roughly) is your WH40?

Certainly! It's a brand-new GW1100 and brand-new WH-40 (both purchased within the last 3 weeks).

Everything works so well with this driver, and I'm sure grateful, so thank you. Also, I have a brand-new WH68; and your driver works *swimmingly* with it!

gjr80, 2022-05-27:
Thanks, but it's the WeeWX log I am after not the console output. When you run that command there should multiple entries appear in the log with many hexadecimal pairs. That is the raw data to and from your GW1100, it is that exchange that I am after.

W0CHP, 2022-05-27:
Ooops - sorry!

```
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 26 03 29' to 'ew-10.1.10.250:45000'                                                     
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 26 09 C4 5B BE 6E 55 34 03'                                                                        
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 50 03 53' to 'ew-gw1102.internal.cuccio.us:45000'                                                     
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 50 12 0E 47 57 31 31 30 30 42 5F 56 32 2E 31 2E 34 BA'                                             
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 30 03 33' to 'ew-gw1102.internal.cuccio.us:45000'                                                     
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 30 0B 02 01 62 90 42 58 10 03 DD'                                                                  
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 27 03 2A' to 'ew-gw1102.internal.cuccio.us:45000'                                                     
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 27 00 40 01 00 DF 06 2C 08 25 E3 09 27 8A 02 00 55 07 61 0A 00 18 0B 00 00 0C 00 00 15 00 00 00 00 16 00 00 17 00 19 00 0A 0E 00 00 10 00 00 11 01 B3 12 00 00 03 02 13 00 00 03 02 0D 00 00 BF'                                                                                          
May 27 03:15:36 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 57 03 5A' to 'ew-gw1102.internal.cuccio.us:45000'                                                     
May 27 03:15:37 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 57 00 25 0E 00 00 10 00 00 00 00 11 00 00 01 B3 12 00 00 03 02 13 00 00 03 02 0D 00 00 0F 00 64 88 00 00 03 99' 
May 27 03:15:37 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Parsed data: {'intemp': 22.3, 'inhumid': 44, 'absbarometer': 969.9, 'relbarometer': 1012.2, 'outtemp': 8.5, 'outhumid': 97, 'winddir': 24, 'windspeed': 0.0, 'gustspeed': 0.0, 'light': 0.0, 'uv': 0.0, 'uvi': 0, 'daymaxwind': 1.0, 't_rainrate': 0.0, 't_rainday': 0.0, 't_rainweek': 43.5, 't_rainmonth': 77.0, 't_rainyear': 77.0, 't_rainevent': 0.0, 't_rainhour': 10.0, 'day_reset': 0, 'week_reset': 0, 'annual_reset': 3, 'datetime': 1653639337}                                                
May 27 03:15:37 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Sending packet 'FF FF 3C 03 3F' to 'ew-gw1102.internal.cuccio.us:45000'                                                     
May 27 03:15:37 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Received response 'FF FF 3C 01 5B 00 FF FF FF FF FF 00 01 00 00 03 84 54 04 02 FF FF FF FF FF 00 03 00 00 E2 58 8E 04 04 FF FF FF FF 00 00 05 00 00 00 AD 00 04 06 FF FF FF FF 00 00 07 FF FF FF FF 00 00 08 FF FF FF FF 00 00 09 FF FF FF FF 00 00 0A FF FF FF FF 00 00 0B FF FF FF FF 00 00 0C FF FF FF FF 00 00 0D FF FF FF FF 00 00 0E FF FF FF FF 1F 00 0F FF FF FF FF 1F 00 10 FF FF FF FF 1F 00 11 FF FF FF FF 1F 00 12 FF FF FF FF 1F 00 13 FF FF FF FF 1F 00 14 FF FF FF FF 1F 00 15 FF FF FF FF 1F 00 16 FF FF FF FF 0F 00 17 FF FF FF FF 0F 00 18 FF FF FF FF 0F 00 19 FF FF FF FF 0F 00 1A FF FF FF FF 0F 00 1B FF FF FF FF 0F 00 1C FF FF FF FF 0F 00 1D FF FF FF FF 0F 00 1E FF FF FF FF 0F 00 1F FF FF FF FF FF 00 20 FF FF FF FF FF 00 21 FF FF FF FF FF 00 22 FF FF FF FF FF 00 23 FF FF FF FF FF 00 24 FF FF FF FF FF 00 25 FF FF FF FF FF 00 26 FF FF FF FF FF 00 27 FF FF FF FF 0F 00 28 FF FF FF FF FF 00 29 FF FF FF FF FF 00 2A FF FF FF FF FF 00 2B FF FF FF FF FF 00 2C FF FF FF FF FF 00 2D FF FF FF FF FF 00 2E FF FF FF FF FF 00 2F FF FF FF FF FF 00 30 FF FF FF FF FF 00 4F' 
May 27 03:15:37 WX-NTP-RPi2 weewx[14297]: DEBUG __main__: Processed parsed data: {'intemp': 22.3, 'inhumid': 44, 'absbarometer': 969.9, 'relbarometer': 1012.2, 'outtemp': 8.5, 'outhumid': 97, 'winddir': 24, 'windspeed': 0.0, 'gustspeed': 0.0, 'light': 0.0, 'uv': 0.0, 'uvi': 0, 'daymaxwind': 1.0, 't_rainrate': 0.0, 't_rainday': 0.0, 't_rainweek': 43.5, 't_rainmonth': 77.0, 't_rainyear': 77.0, 't_rainevent': 0.0, 't_rainhour': 10.0, 'day_reset': 0, 'week_reset': 0, 'annual_reset': 3, 'datetime': 1653639337, 'wh68_batt': 1.68, 'wh68_sig': 4, 'wh40_batt': 14.2, 'wh40_sig': 4, 'wh26_batt': 0, 'wh26_sig': 4} 
```

gjr80, 2022-05-27:
Thanks. The driver is decoding the raw data from your GW1100 as intended. In the interim, until I have a final solution, you can add the following entry in `weewx.conf` under `[StdCalibrate]` to have WeeWX use the correct (1.xV) value:
```
[StdCalibrate]
    [[Corrections]]
        wh40_batt = wh40_batt/10.0
```
You will need to restart WeeWX for this to take effect. Note that the GW1000 driver will still provide values that are 10 times to high (so if you run the driver directly you will still see values around 14V) but any data saved to database or included in reports will be correct.

W0CHP, 2022-05-27:
> wh40_batt = wh40_batt/10.0

Worked like a charm! Cheers! :-)

gjr80, 2022-06-03:
I believe I have tracked down the problem and put a fix in the driver. It appears there are a couple of things going on with the WH40 that have lead to this issue:

Early versions of the WH40 did not report battery state. Despite this the Ecowitt gateway device API provided WH40 battery state via a single bit indicating bad (0)/good (1). A later API revision changed the reported WH40 battery state to a voltage, with the API returning a byte containing 10x the battery voltage. The current API documentation still reflects this. It seems that despite the early WH40 versions not reporting battery state the current API returns a fixed value of `0x10` for battery state (which decodes to 1.6V). Ecowitt has confirmed that early WH40 do not provide battery state but the fixed `0x10` is based on anecdotal evidence from a number of users.

Later produced WH40 (date unknown but seemingly somewhere in the 2nd half of 2021) started providing battery state data. The problem is it appears (again based on a number of user's experiences) these new WH40 do indeed provide battery state in volts via a single byte, but instead of the byte containing 10x the battery voltage it is 100x the battery voltage. Hence when we decode the battery state byte for new WH40 (by dividing by 10 as per the API documentation) we get a value that is 10x the actual value (eg 14V instead of 1.4V).

What I have done is implement a crude check for legacy/new WH40 by first decoding the battery state by dividing by 10, if that results in a value <2.0 the data is discarded (legacy WH40 assumed) or if it is >=2.0 the data is again divided by 10 (new WH40 assumed) and the result is then used for WH40 battery state.

In any case, I have put together Ecowitt gateway driver v0.5.0b5 that, among other things, includes the WH40 battery state fix. If you want to give it a run on an existing installation:

1. rename your current `gw1000.py` to `gw1000_orig.py` in case you need to revert to the old driver (`gw1000.py` should be in the `/usr/share/weewx/user` or `/home/weewx/bin/user` directories depending on your WeeWX install)
2. Download the b5 driver file, the following command will download the file to the `/var/tmp` directory:
    `$ wget -P /var/tmp https://raw.githubusercontent.com/gjr80/weewx-gw1000/v0.5.0b5/bin/user/gw1000.py`
3. Copy the downloaded `gw1000.py` to `/usr/share/weewx/user` or `/home/weewx/bin/user` as applicable:
4. Remove/comment out the `[StdCalibrate] [[Corrections]]` entry for `wh40_batt` in `weewx.conf` 
5. You might wish to stop WeeWX and [run WeeWX directly](http://weewx.com/docs/usersguide.htm#Running_directly) to observe the loop packets and archive records on screen and confirm that `wh40_batt` is correctly reported
6. Once happy that `wh40_batt` is correct, you can stop running WeeWX directly and restart it as per normal, do keep an eye on your WeeWX log for any errors 

I believe b5 will fix the WH40 battery issue and I have done some limited testing of the code, but I cannot test this fully as I do not have a WH40. So please let me know if you encounter any problems. Also, there are a few other changes in b5 unrelated to the WH40, again if you notice odd behaviour or errors in the WeeWX log please let me know.

I would like to hear if b5 works, if it does I will release b5 to all users. 

If you do need to revert to the old driver, just delete `/usr/share/weewx/user/gw1000.py` or `/home/weewx/bin/user/gw1000.py`, rename `gw1000_orig.py` to `gw1000.py`, restore the `[StdCalibrate] [[Corrections]]` entry and restart WeeWX.

W0CHP, 2022-06-06:
@gjr80 I can confirm that `b5` works perfectly! The correct voltage is now being reported by the driver. Cheers!

gjr80, 2022-06-07:
That's great, thanks. I have had one user with an old version WH40 use b5 and it worked as expected so I think we have this one solved. I will leave it open for now and close when I release the v0.5.0 driver. Thanks for your help.

gjr80, 2023-07-22:
v0.5.0b5 was released in June 2022, think we can consider this fixed.

closed 2023-07-22 by gjr80