gjr80/weewx-gw1000 #58: Barometer inconsistencies with FineOffset and GW1100

bdwilson, 2022-07-27:
I recently ditched my Acurite all-in-one via RTL_433 for an Ambient setup with data capture via Ecowitt GW1100B.  I couldn't be happier having access to the data without jumping through a bunch of hoops. I'm also still bitter about the Acurite SmartHub drama a few years ago.

Anyway, I've moved weewx to capture data via the GW1000 API (thank you for this by the way!!) I am getting data, and most of that data appears to be accurate, minus the pressure. Before configuring any pressure offsets in the GW1100, I confirmed I was seeing data in Weewx (via debug output - note, I'm not sending data to Wunderground from Weewx, just using the data locally):

LOOP:   2022-07-26 17:42:17 EDT ([1658871737](1658871737)) altimeter: 30.039101382806564, appTemp: 89.46208415470079, barometer: 30.026459802683856, cloudbase: 899.6153853914325, dateTime: [1658871737](1658871737), daymaxwind: 10.2, dayRain: 0.1811023622047244, dewpoint: 75.94169230427771, heatindex: 80.65700000000001, humidex: 99.87324145788232, inDewpoint: 56.30802378104585, inHumidity: 51, inTemp: 75.74000000000001, luminosity: 16097.0, maxSolarRad: None, monthRain: 0.7716535433070867, outHumidity: 91, outTemp: 78.80000000000001, pressure: 29.766227400000002, radiation: 127.0481452249408, rain: None, rainRate: 0.0, relbarometer: 1008.0, stormRain: 0.1811023622047244, usUnits: 1, UV: 1, uvradiation: 20.0, weekRain: 17.6, wh65_batt: 0, wh65_sig: 4, windchill: 78.80000000000001, windDir: 316, windGust: 1.1184709259696521, windSpeed: 0.4473883703878609, yearRain: 0.7716535433070867

Weewx shows the correct barometer: 30.038 inHg (matching my local WX station almost perfect - 30.04), however WXView/GW1100 shows 29.76 (as does Wunderground which is connected to GW1100, not Weewx). 

In order to get the correct pressure in WXView/Wunderground, I used the [Pressure Calibration utilit](https://caboose700.github.io/ws-barometer-offset-calculator/)y linked on [WSForum](https://www.wxforum.net/index.php?topic=40645.0).  Based on my altitude (250ft/76.2m) and the pressure reported from WSView without any offsets configured, the reported pressure is 29.76 - note, this is what relbarometer from Weewx above directly translates to. My local [WX station](https://aviationweather.gov/metar/data?ids=KRDU&format=decoded&hours=0&taf=off&layout=on) currently reports 30.04. Using the calibration calculator,  this gives me a relative offset of +8.7 hPa and ABS offset of  -8.4 hPa. 

When I do this, WSView (and Wunderground) report Relative as 30.07 and Abs 21.37.  The relative pressure is close to 30.04.  When I go back to look at Weewx, altimeter and barometer are None! 

LOOP:   2022-07-26 17:47:40 EDT ([1658872060](1658872060)) altimeter: None, appTemp: 89.69719196303436, barometer: None, cloudbase: 750.1857129645515, dateTime: [1658872060](1658872060), daymaxwind: 10.2, dayRain: 0.1811023622047244, dewpoint: 76.41918286295598, heatindex: 80.553, humidex: 100.2012467210435, inDewpoint: 57.01182055322032, inHumidity: 52, inTemp: 75.92, luminosity: 15741.0, maxSolarRad: None, monthRain: 0.7716535433070867, outHumidity: 93, outTemp: 78.62, pressure: None, radiation: 124.23835832675611, rain: 0.0, rainRate: 0.0, relbarometer: 1018.3, stormRain: 0.1811023622047244, usUnits: 1, UV: 1, uvradiation: 19.2, weekRain: 17.6, wh65_batt: 0, wh65_sig: 4, windchill: 78.62, windDir: 316, windGust: 1.1184709259696521, windSpeed: 0.22369418519393044, yearRain: 0.7716535433070867

My issue is that the pressure data in Weewx doesn't match the calibrated data coming from GW1100. Given that I believe barometer value is calculated in Weewx, I'm at a loss for why these are showing as None? I even lowered the MinMax values, but still None.  Here is my configuration if that would help: [weewx.conf](https://bubba.d.pr/C6vcpL). I appreciate any guidance or direction on this; I've been pulling my hair out for a few days. Also asking this of the Weewx user forums in case it’s an issue on my end. 

Thanks!

gjr80, 2022-07-27:
Brian, am currently away from home with very limited internet. Am back home Friday so will have a closer look then.

Gary

bdwilson, 2022-07-28:
More info:

$ sudo PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --live-data
Using configuration file /etc/weewx/weewx.conf
debug level is '2'

IP address to be obtained by discovery
Port number obtained from station config
Battery state filtering ('False') using the default

Interrogating GW1100 at 192.168.1.71:45000

GW1100 live sensor data (2022-07-28 09:34:33 EDT (1659015273)): absbarometer: 722.1 mbar, daymaxwind: 20 kph, gustspeed: 9 kph, inhumid: 51%, intemp: 24.0°C, light: 55930 lx, outhumid: 77%, outtemp: 28.5°C, rainday: 0.00 cm, rainevent: 1.65 cm, rainmonth: 3.15 cm, rainrate: 0.00 cm/h, rainweek: 2.95 cm, rainyear: 3.15 cm, relbarometer: 1016.7 mbar, uv: 153 W/m², uvi: 4.0, wh65_batt: 0, wh65_sig: 4, winddir: 264°, windspeed: 7 kph

Seeing this, I wonder if I need to do some StdCalibrate work on absbarometer or relbarometer so that proper pressure can be computed.

bdwilson, 2022-07-29:
I appear to have solved this by the following under the [GW1000] stanza.  Not sure if this is the correct way to do it, but the barometer values match relative pressure in WXView now. 

    [[field_map_extensions]]
        barometer = relbarometer

$ sudo PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --get-calibration
Using configuration file /etc/weewx/weewx.conf
debug level is '2'

IP address to be obtained by discovery
Port number obtained from station config
Battery state filtering ('False') using the default

Interrogating GW1100 at 192.168.1.71:45000
Calibration
      Solar radiation gain:  1.0
                   UV gain:  1.0
                 Wind gain:  1.0
                 Rain gain:  1.0
 Inside temperature offset:  0.0 °C
    Inside humidity offset:  0.0 %
  Absolute pressure offset: -284.5 hPa
  Relative pressure offset: 294.6 hPa
Outside temperature offset:  0.0 °C
   Outside humidity offset:  0.0 %
     Wind direction offset:  0.0 °

closed 2022-07-29 by bdwilson

gjr80, 2022-07-29:
I think the core of your problem is the different ways in which the Ecowitt gateway devices and WeeWX calculate barometric pressure. The Ecowitt devices provide two pressure values; what Ecowitt refers to as absolute pressure and relative pressure. Ecowitt's relative pressure is simply absolute pressure offset by a fixed value, this a poor approximation for what we know as `barometer` as it does not include any correction for temperature (barometer is air pressure corrected for altitude, temperature and frequently humidity the Ecowitt approximation corrects for altitude but for a single temperature only). WeeWX on the other hand calculates `barometer` by correcting `pressure` for altitude and temperature (as an aside provided one of the three 'pressure' fields (`pressure`, `barometer` or `altimeter`) are provided WeeWX will calculate the other two). For this reason the default behaviour of the GW1000 driver is pass the Ecowitt absolute pressure value through to WeeWX as the WeeWX field `pressure` and the Ecowitt relative pressure field is passed through as WeeWX field `relbarometer`. This way WeeWX uses the absolute pressure to calculate `barometer` and `altimeter`. In this case the WeeWX calculated `barometer` field will almost certainly be different to the Ecowitt relative pressure value. If the Ecowitt device is publishing direct to Weather Underground you will see the same pressure discrepancy as WU uses `barometer`/relative pressure.

I cannot explain the three `None` values you saw; for some reason WeeWX received no pressure fields so consequently could not calculate the missing pressure fields. The large discrepancy of hundreds of hPa is due to the large pressure offsets being used.

What to do really depends on personal preference. If you want all of your pressure values to be consistent across all devices/sites then you need to do what you did and map the Ecowitt `relbarometer` field to WeeWX field `barometer`. This will cause WeeWX to use the Ecowitt relative pressure field as the WeeWX `barometer` field. If you are not happy with using Ecowitt's simplistic approach to calculating relative pressure as `barometer`, you should only only pass the Ecowitt absolute pressure through to WeeWX and let WeeWX calculate `barometer`. In this case if you are concerned about WU you should have WeeWX publish to WU rather than the Ecowitt device.

The GW1000 driver default action is to not pass the Ecowitt relative pressure through to WeeWX as field `barometer` thus leaving WeeWX to calculate the filed `barometer`.