gjr80/weewx-gw1000 #59: wn34 labels changed in new version on the gw2000

jrtaylor71, 2022-11-03:
The wn34 now shows up as tf_ch1 to 8 and tf_batt1 to 8. This also has a new ws90 connected to it.

'PASSKEY=5147A167F8127580864FBA85B8B824DB&stationtype=GW2000B_V2.1.9&runtime=139543&dateutc=2022-11-03+15:59:27&tempinf=77.72&humidityin=43&baromrelin=31.012&baromabsin=29.864&tempf=58.82&humidity=75&winddir=191&windspeedmph=7.83&windgustmph=10.29&maxdailygust=15.21&solarradiation=76.01&uv=0&rrain_piezo=0.000&erain_piezo=0.000&hrain_piezo=0.000&drain_piezo=0.000&wrain_piezo=0.000&mrain_piezo=0.000&yrain_piezo=0.000&ws90cap_volt=4.7&ws90_ver=126&soilmoisture1=31&tf_ch1=49.10&soilbatt1=1.6&tf_batt1=1.58&wh90batt=3.16&freq=915M&model=GW2000B'

gjr80, 2022-11-04:
I'm not sure what your point or issue is. The data you have posted looks very much like a post from a GW2000 to a custom server in Ecowitt format. The WeeWX Ecowitt Gateway Driver does not use data in this format, rather it obtains device and sensor data from the gateway device via the Ecowitt Gateway Device API.

jrtaylor71, 2022-11-04:
Sorry grabbed the wrong data. This is missing the soil temp sensors as they change in the new firmware.

2022-11-03 20:58:53 CDT (1667527133): 'dateTime': '1667527133', 'daymaxwind': '8.0', 'extraTemp9': '10.2', 'inHumidity': '46', 'inTemp': '26.2', 'luminosity': '0.0', 'outHumidity': '85', 'outTemp': '15.0', 'p_dayRain': '0.0', 'p_monthRain': '0.0', 'p_rain': 'None', 'p_rainRate': '0.0', 'p_stormRain': '0.0', 'p_weekRain': '0.0', 'p_yearRain': '0.0', 'pressure': '1007.7', 'relbarometer': '1046.6', 'soilMoist1': '31', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.6', 'wh51_ch1_sig': '3', 'windDir': '169', 'windGust': '1.2', 'windSpeed': '1.1', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.16', 'ws90_sig': '4'

gjr80, 2022-11-04:
The Ecowitt Gateway Device driver puts WN34 temperature data (ch 1-8) in fields `extraTemp9` to `extraTemp16` and I see you have `'extraTemp9': '10.2'`. Is that not correct?

jrtaylor71, 2022-11-04:
Yes I have 1 WN34 on CH1. I was expecting tf_ch1 not extraTemp9. So the only thing I need to do is add them to /etc/weewx/skins/Seasons/skin.conf as I am using weewx 4.9.1?

gjr80, 2022-11-04:
Early versions of the Ecowitt Gateway API used observation type based names (eg 'temp', 'humid' etc) followed by an index number (not necessarily channel number) rather than sensor model/channel number. When the Gateway device driver was first written this was the approach adopted, though there is a further layer of abstraction as we then have WeeWX field names that the driver needs to map to. Ecowitt has subsequently added to the Gateway device API piecemeal and consequently there are (well appear to be) some discrepancies at times.

The [Field Map page](https://github.com/gjr80/weewx-gw1000/wiki/Field-map) in the Gateway driver wiki was put together for this very reason.

If you wish to add your WN34 data to the 4.9.1 Seasons skin, then yes you will need to modify some of the entries under the `[DisplayOptions]` stanza in the Seasons `skin.conf`. if you intend to plot any of the data then you will need to add plot definitions to the `[ImageGenerator]` stanza as well.

jrtaylor71, 2022-11-04:
Still cannot get them to show up graph. Here is the data I am getting.

Nov  4 11:32:19 weewx-gw2517 weewx[25943] INFO user.gw1000: GatewayDriver: Packet 2022-11-04 11:32:19 CDT (1667579539): {'dateTime': '1667579539', 'daymaxwind': '8.5', 'extraTemp9': '9.7', 'inHumidity': '48', 'inTemp': '24.5', 'luminosity': '8840.0', 'outHumidity': '90', 'outTemp': '3.7', 'p_dayRain': '1.9', 'p_monthRain': '6.7', 'p_rain': '0.0', 'p_rainRate': '0.0', 'p_stormRain': '6.7', 'p_weekRain': '6.7', 'p_yearRain': '6.7', 'pressure': '1012.1', 'relbarometer': '1012.1', 'soilMoist1': '47', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '4', 'windDir': '342', 'windGust': '4.4', 'windSpeed': '3.7', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.16', 'ws90_sig': '4'}

skins.conf

    # This list determines which types will appear in the "current conditions"
    # section, as well as in which order.
    observations_current = outTemp, heatindex, windchill, dewpoint, outHumidity, barometer, windSpeed, p_rain, p_rainRate, rain, rainRate, UV, radiation, lightning_strike_count, inTemp, inHumidity, extraTemp1, extraHumid1, extraTemp2, extraHumid2, pm1_0, pm2_5, pm10_0, extraTemp9, extraTemp10, extraTemp11, extraTemp12, extraTemp13, extraTemp14, extraTemp15, extraTemp16

    # This list determines which types will appear in the "statistics" and
    # "statistical summary" sections, as well as in which order.
    observations_stats = outTemp, heatindex, windchill, dewpoint, outHumidity, barometer, windSpeed, p_rain, p_rainRate, p_dayRain, p_monthRain, rain, rainRate, hourRain, ET, hail, hailRate, snow, UV, radiation, lightning_strike_count, lightning_distance, inTemp, inHumidity, extraTemp1, extraHumid1, extraTemp2, extraHumid2, extraTemp3, extraHumid3, extraTemp4, extraHumid4, extraTemp5, extraHumid5, extraTemp6, extraHumid6, extraTemp7, extraHumid7, extraTemp8, extraHumid8, extraTemp9, leafTemp1, leafTemp2, leafWet1, leafWet2, soilTemp1, soilTemp2, soilTemp3, soilTemp4, soilMoist1, soilMoist2, soilMoist3, soilMoist4, pm1_0, pm2_5, pm10_0, co, co2, nh3, no2, o3, so2

    # This list determines which types will appear in the RSS feed.
    observations_rss = outTemp, inTemp, barometer, windSpeed, p_rain, p_rainRate, p_dayRain, p_monthRain, rain, rainRate, windchill, heatindex, dewpoint, outHumidity, inHumidity

    # Some observations display a sum rather than min/max values
    obs_type_sum = rain, ET, hail, snow, lightning_strike_count

    # Some observations display only the max value
    obs_type_max = rainRate, hailRate, snowRate, UV

    # The sensor status information is used in the sensor pages.  These lists
    # determine which database fields will be shown, as well as the order in
    # which they will be displayed.
    sensor_connections = rxCheckPercent, signal1, signal2, signal3, signal4, signal5, signal6, signal7, signal8, ws80_sig, wn34_ch1_sig, wh51_ch1_sig
    sensor_batteries = outTempBatteryStatus, inTempBatteryStatus, rainBatteryStatus, hailBatteryStatus, snowBatteryStatus, windBatteryStatus, uvBatteryStatus, txBatteryStatus, batteryStatus1, batteryStatus2, batteryStatus3, batteryStatus4, batteryStatus5, batteryStatus6, batteryStatus7, batteryStatus8, wn34_ch1_batt, ws90_batt, wh51_ch1_batt
    sensor_voltages = consBatteryVoltage, heatingVoltage, supplyVoltage, referenceVoltage, wn34_ch1_batt, wh51_ch1_batt

.............

[ImageGenerator]

.............

       [[[daytempext3]]]
            [[[[extraTemp9]]]]

gjr80, 2022-11-05:
Those settings look fine.

Some quick background on how the Seasons skin works. The Seasons skin will only display data that appears in what we call [archive records](http://weewx.com/docs/customizing.htm#LOOP_packets_vs._archive_records). Basically, if your data is not in an archive record it will never display in the Seasons skin. Observations that appear in the 'Current Conditions' pane only need appear in the current archive record, they do not necessarily need to be saved in your database. Observations displayed in the 'Statistics' pane and any observations that are plotted, must appear in archive records **and** must be saved to your database.

When WeeWX is first run it creates your database using a database schema. The default schema is the [`wview_extended`](https://github.com/weewx/weewx/blob/master/bin/schemas/wview_extended.py) schema, this schema includes support for some 100 plus fields. Provided a given field name exists in your database schema **and** the current archive record, WeeWX will automatically save the data for that field to database. If a field exists in the database schema but not the archive record, or vice versa, that field will not be saved to database.

In you case, considering your WN34 data, the sensor temperature data appears in field `extraTemp9`. Based on what you have shown above `extraTemp9` should be in your archive record. To check that it indeed appearing in your archive records you should stop WeeWX and then [run WeeWX directly](http://weewx.com/docs/usersguide.htm#Running_directly). You will now see loop packets and archive records appear on the console. Loop packets start with 'LOOP:' and archive records start with 'REC:'. Does `extraTemp9` appear in the archive records (REC:)? If so good. If not we need to fix that.

The next thing to check is that the field `extraTemp9` is in your database schema. If you are using the default `wview_extended` schema it will be not there unless you have manually added it. If not already done so have a look at the `wview_extended` schema link I provided above, you will see that the schema includes fields `extraTemp1` to `extraTemp8` but no `extraTemp9`. To add a new field to your database you use the utility `wee_database`, the process is covered in the [Modifying the database _in situ_](http://weewx.com/docs/customizing.htm#add_archive_type) section of the [Customization Guide](http://weewx.com/docs/customizing.htm).

I would restart WeeWX after making the above changes.

Once you have `extraTemp9` appearing in archive records and `extraTemp9` is in your database schema, you should see `extraTemp9` appearing in your 'Current Conditions' and 'Stats' panes (after the next report cycle). There is one final step to get the `extraTemp9` plot to show up, you may have done it I cannot tell, but the `plot_groups` setting under `[DisplayOptions]` in `skin.conf` will need to have `tempext3` added to the list.

jrtaylor71, 2022-11-06:
Yes extraTemp9 does appear.
REC:    2022-11-06 09:55:00 CST (1667750100) 'altimeter': '30.818477775545563', 'appTemp': '40.94851237397133', 'barometer': '30.8481654497766', 'cloudbase': '2781.7928906689076', 'dateTime': '1667750100', 'daymaxwind': '5.1', 'dayRain': '0.07480314960629922', 'dewpoint': '38.43851128105682', 'ET': 'None', 'extraTemp9': '45.98', 'heatindex': '43.88666666666666', 'humidex': '46.07600000000001', 'inDewpoint': '54.56010229850933', 'inHumidity': '48.0', 'inTemp': '75.65599999999999', 'interval': '5.0', 'luminosity': '16201.333333333334', 'maxSolarRad': '314.92374604545023', 'monthRain': '0.26377952755905515', 'outHumidity': '74.53333333333333', 'outTemp': '46.07600000000001', 'p_dayRain': '0.0905511811023622', 'p_monthRain': '0.4566929133858268', 'p_rain': '0.0', 'p_rainRate': '0.0', 'p_stormRain': '0.19291338582677167', 'p_weekRain': '0.0905511811023622', 'p_yearRain': '0.4566929133858268', 'pressure': '29.674881305333326', 'rain': '0.0', 'rainRate': '0.0', 'relbarometer': '1004.9066666666664', 'signal8': '1', 'soilMoist1': '52.0', 'stormRain': '0.26377952755905515', 'usUnits': '1', 'UV': '0.9333333333333333', 'uvradiation': '9.240000000000002', 'weekRain': '6.7', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '4.0', 'windchill': '44.06393856303561', 'windDir': '227.79428659219633', 'windGust': '11.408403444890453', 'windGustDir': '254.0', 'windrun': '0.3864938421961799', 'windSpeed': '4.637926106354159', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4.0', 'ws90_batt': '3.14', 'ws90_sig': '4.0', 'yearRain': '0.26377952755905515'

##############################################################################

#   This section binds a data store to a database.

[DataBindings]

    [[wx_binding]]
        # The database must match one of the sections in [Databases].
        # This is likely to be the only option you would want to change.
        database = archive_sqlite
        # The name of the table within the database
        table_name = archive
        # The manager handles aggregation of data for historical summaries
        manager = weewx.manager.DaySummaryManager
        # The schema defines the structure of the database.
        # It is *only* used when the database is created.
        schema = schemas.wview_extended.schema

##############################################################################

#   This section defines various databases.

[Databases]

    # A SQLite database is simply a single file
    [[archive_sqlite]]
        database_name = weewx.sdb
        database_type = SQLite

    # MySQL
    [[archive_mysql]]
        database_name = weewx
        database_type = MySQL

##############################################################################

jrtaylor71, 2022-11-06:
I added extraTemp9 to 16

Archive info
  Database name:        weewx.sdb
  Table name:           archive
  Version               4.0
  Unit system:          1 (US)
  First good timestamp: 2022-11-04 10:05:00 CDT (1667574300)
  Last good timestamp:  2022-11-06 10:15:00 CST (1667751300)
  Number of records:    589
  weewx (weewx.conf) is set to use an archive interval of 300 seconds.
  The station hardware was not interrogated in determining archive interval.

Databases configured in weewx.conf
  Database name:        weewx.sdb
  Database driver:      weedb.sqlite

  Database name:        weewx
  Database driver:      weedb.mysql
  Database host:        localhost

Supported SQL keys
dateTime                  usUnits                   interval                 
altimeter                 appTemp                   appTemp1                 
barometer                 batteryStatus1            batteryStatus2           
batteryStatus3            batteryStatus4            batteryStatus5           
batteryStatus6            batteryStatus7            batteryStatus8           
cloudbase                 co                        co2                      
consBatteryVoltage        dewpoint                  dewpoint1                
ET                        extraHumid1               extraHumid2              
extraHumid3               extraHumid4               extraHumid5              
extraHumid6               extraHumid7               extraHumid8              
extraTemp1                extraTemp2                extraTemp3               
extraTemp4                extraTemp5                extraTemp6               
extraTemp7                extraTemp8                forecast                 
hail                      hailBatteryStatus         hailRate                 
heatindex                 heatindex1                heatingTemp              
heatingVoltage            humidex                   humidex1                 
inDewpoint                inHumidity                inTemp                   
inTempBatteryStatus       leafTemp1                 leafTemp2                
leafWet1                  leafWet2                  lightning_distance       
lightning_disturber_count lightning_energy          lightning_noise_count    
lightning_strike_count    luminosity                maxSolarRad              
nh3                       no2                       noise                    
o3                        outHumidity               outTemp                  
outTempBatteryStatus      pb                        pm10_0                   
pm1_0                     pm2_5                     pressure                 
radiation                 rain                      rainBatteryStatus        
rainRate                  referenceVoltage          rxCheckPercent           
signal1                   signal2                   signal3                  
signal4                   signal5                   signal6                  
signal7                   signal8                   snow                     
snowBatteryStatus         snowDepth                 snowMoisture             
snowRate                  so2                       soilMoist1               
soilMoist2                soilMoist3                soilMoist4               
soilTemp1                 soilTemp2                 soilTemp3                
soilTemp4                 supplyVoltage             txBatteryStatus          
UV                        uvBatteryStatus           windBatteryStatus        
windchill                 windDir                   windGust                 
windGustDir               windrun                   windSpeed                
extraTemp9                extraTemp10               extraTemp11              
extraTemp12               extraTemp13               extraTemp14              
extraTemp15               extraTemp16              

Supported observation keys
altimeter                 appTemp                   appTemp1                 
barometer                 batteryStatus1            batteryStatus2           
batteryStatus3            batteryStatus4            batteryStatus5           
batteryStatus6            batteryStatus7            batteryStatus8           
cloudbase                 co                        co2                      
[... 10 lines cut ...]
inDewpoint                inHumidity                inTemp                   
inTempBatteryStatus       leafTemp1                 leafTemp2                
leafWet1                  leafWet2                  lightning_distance       
lightning_disturber_count lightning_energy          lightning_noise_count    
lightning_strike_count    luminosity                maxSolarRad              
nh3                       no2                       noise                    
o3                        outHumidity               outTemp                  
outTempBatteryStatus      pb                        pm10_0                   
pm1_0                     pm2_5                     pressure                 
radiation                 rain                      rainBatteryStatus        
rainRate                  referenceVoltage          rxCheckPercent           
signal1                   signal2                   signal3                  
signal4                   signal5                   signal6                  
signal7                   signal8                   snow                     
snowBatteryStatus         snowDepth                 snowMoisture             
[... cut ...]
soilTemp4                 supplyVoltage             txBatteryStatus          
UV                        uvBatteryStatus           windBatteryStatus        
windchill                 windDir                   windGust                 
windGustDir               windrun                   windSpeed                
extraTemp9                extraTemp10               extraTemp11              
extraTemp12               extraTemp13               extraTemp14              
extraTemp15               extraTemp16

jrtaylor71, 2022-11-06:
I added tempext3 to plot_groups and it is now plotting the soil temps. 

Question: how is tempext3 defined and where?

jrtaylor71, 2022-11-07:
I did find one more thing. Exactly 3 times an hour it does not show the GatewayDriver.

Nov  6 22:08:07 weewx-gw2517 weewx[43351] INFO user.gw1000: GatewayDriver: Packet 2022-11-06 22:08:07 CST (1667794087): {'dateTime': '1667794087', 'daymaxwind': '12.6', 'extraTemp9': '7.9', 'inHumidity': '41', 'inTemp': '25.0', 'luminosity': '0.0', 'outHumidity': '46', 'outTemp': '3.0', 'p_dayRain': '2.3', 'p_monthRain': '11.6', 'p_rainRate': '0.0', 'p_stormRain': '4.9', 'p_weekRain': '2.3', 'p_yearRain': '11.6', 'pressure': '1018.7', 'rain': '0.0', 'relbarometer': '1018.7', 'soilMoist1': '49', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '0', 'windDir': '345', 'windGust': '5.2', 'windSpeed': '3.7', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.14', 'ws90_sig': '4'}
Nov  6 22:08:28 weewx-gw2517 weewx[43351] INFO user.gw1000: GatewayDriver: Packet 2022-11-06 22:08:28 CST (1667794108): {'dateTime': '1667794108', 'daymaxwind': '12.6', 'extraTemp9': '7.9', 'inHumidity': '41', 'inTemp': '25.0', 'luminosity': '0.0', 'outHumidity': '46', 'outTemp': '3.0', 'p_dayRain': '2.3', 'p_monthRain': '11.6', 'p_rainRate': '0.0', 'p_stormRain': '4.9', 'p_weekRain': '2.3', 'p_yearRain': '11.6', 'pressure': '1018.6', 'rain': '0.0', 'relbarometer': '1018.6', 'soilMoist1': '49', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '0', 'windDir': '354', 'windGust': '4.6', 'windSpeed': '4.0', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.14', 'ws90_sig': '4'}
Nov  6 22:08:49 weewx-gw2517 weewx[43351] INFO user.gw1000: GatewayDriver: Packet 2022-11-06 22:08:49 CST (1667794129): {'dateTime': '1667794129', 'daymaxwind': '12.6', 'extraTemp9': '7.9', 'inHumidity': '41', 'inTemp': '25.0', 'luminosity': '0.0', 'outHumidity': '46', 'outTemp': '3.0', 'p_dayRain': '2.3', 'p_monthRain': '11.6', 'p_rainRate': '0.0', 'p_stormRain': '4.9', 'p_weekRain': '2.3', 'p_yearRain': '11.6', 'pressure': '1018.6', 'rain': '0.0', 'relbarometer': '1018.6', 'soilMoist1': '49', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '0', 'windDir': '348', 'windGust': '5.2', 'windSpeed': '4.4', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.14', 'ws90_sig': '4'}

Otherwise it does.

Nov  6 22:11:53 weewx-gw2517 weewx[43351] INFO user.gw1000: GatewayDriver: Received GW2000 data: 2022-11-06 22:11:53 CST (1667794313) {'absbarometer': '1018.8', 'annual_reset': '0', 'datetime': '1667794313', 'day_reset': '0', 'daymaxwind': '12.6', 'gain0': '1.0', 'gain1': '1.0', 'gain2': '1.0', 'gain3': '1.0', 'gain4': '1.0', 'gain5': '1.0', 'gain6': '1.0', 'gain7': '1.0', 'gain8': '1.0', 'gain9': '1.0', 'gustspeed': '3.9', 'inhumid': '41', 'intemp': '25.0', 'light': '0.0', 'outhumid': '46', 'outtemp': '2.9', 'p_rainday': '2.3', 'p_rainevent': '4.9', 'p_rainmonth': '11.6', 'p_rainrate': '0.0', 'p_rainweek': '2.3', 'p_rainyear': '11.6', 'rain_source': '2', 'relbarometer': '1018.8', 'soilmoist1': '49', 'temp9': '7.8', 'uv': '0.0', 'uvi': '0', 'week_reset': '0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '1', 'winddir': '344', 'windspeed': '2.5', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.14', 'ws90_sig': '3'}
Nov  6 22:11:53 weewx-gw2517 weewx[43351] INFO user.gw1000: GatewayDriver: Mapped GW2000 data: {'dateTime': '1667794313', 'daymaxwind': '12.6', 'extraTemp9': '7.8', 'inHumidity': '41', 'inTemp': '25.0', 'luminosity': '0.0', 'outHumidity': '46', 'outTemp': '2.9', 'p_dayRain': '2.3', 'p_monthRain': '11.6', 'p_rainRate': '0.0', 'p_stormRain': '4.9', 'p_weekRain': '2.3', 'p_yearRain': '11.6', 'pressure': '1018.8', 'rain': '0.0', 'relbarometer': '1018.8', 'soilMoist1': '49', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'wh51_ch1_batt': '1.5', 'wh51_ch1_sig': '1', 'windDir': '344', 'windGust': '3.9', 'windSpeed': '2.5', 'wn34_ch1_batt': '1.58', 'wn34_ch1_sig': '4', 'ws90_batt': '3.14', 'ws90_sig': '3'}

gjr80, 2022-11-08:
Apologies, heading away on a short break and WeeWX took a back seat.

> Question: how is tempext3 defined and where? Why would it not work to just add extraTemp9?

You don't actually define `tempext3` anywhere. It's all to do with the parameterisation/auto provisioning of the Seasons skin introduced in WeeWX v4.6.0. Once you add a plot name (eg `tempext3`) to the `plot_groups` setting the Seasons skin will look for a plot file `daytempext3.png` and automatically display it on the day plots tab, it will look for a plot file `weektempext3.png` and display it on the week plots tabs and so on for month and year. So to generate the plot file `daytempext3.png` you create a `[[[daytempext3]]]` plot definition under `[[day_images]]` in `skin.conf`, for `weektempext3.png` you create a `[[[weekempext3]]]` plot definition under `[[week_images]]` and so on for month and year. Before  parameterisation/auto provisioning of the Seasons skin you had to add boilerplate HTML code to a Seasons skin HTML template as well as creating the plot definitions if you wanted to display a new plot, now it is just a case of adding an entry to `plot_groups` and creating the plot definitions.

gjr80, 2022-11-08:
> I did find one more thing. Exactly 3 times an hour it does not show the GatewayDriver.

I'm not quite sure what you mean. The first stanza of output shows the three loop packets the driver passed to WeeWX (note the use of WeeWX field names, eg `extraTemp9`.

The second stanza of output shows the data received from the GW2000 (note `temp9`) and then the GW2000 data mapped to WeeWX field names (note use of `extraTemp9`).

I don't see anything unusual in either, please correct me if there is something odd there.

jrtaylor71, 2022-11-09:
[image]

jrtaylor71, 2022-11-09:
Ok, I did just notice that if the GW2000 get disconnected from the network the driver crashes. Mine is on wifi and I had to reboot my AP.

Nov  8 17:25:39 weewx-gw2517 weewx[46259] CRITICAL __main__: Caught WeeWxIOError: Failed to obtain response to command 'CMD_GW1000_LIVEDATA' after 3 attempts
Nov  8 17:25:39 weewx-gw2517 weewx[46259] CRITICAL __main__:     ****  Waiting 60 seconds then retrying...
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****  Traceback (most recent call last):
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/weewx/engine.py", line 119, in setupStation
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      self.console = loader_function(config_dict, self)
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 1598, in loader
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      return GatewayDriver(**config_dict[DRIVER_NAME])
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 2038, in __init__
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      super(GatewayDriver, self).__init__(**stn_dict)
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 872, in __init__
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      self.collector = GatewayCollector(ip_address=self.ip_address,
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 2428, in __init__
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      self.station = GatewayCollector.Station(ip_address=ip_address,
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 2966, in __init__
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      self.mac = self.get_mac_address()
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 3358, in get_mac_address
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      return self.send_cmd_with_retries('CMD_READ_STATION_MAC')
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****    File "/usr/share/weewx/user/gw1000.py", line 3556, in send_cmd_with_retries
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****      raise GWIOError(_msg)
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL weewx.engine:     ****  user.gw1000.GWIOError: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL __main__: Unable to load driver: Failed to obtain response to command 'CMD_READ_STATION_MAC' after 3 attempts
Nov  8 17:27:12 weewx-gw2517 weewx[46259] CRITICAL __main__:     ****  Exiting...

After this you have to restart weewx. Might be better to just wait until the next interval to pull data than to exit.

gjr80, 2022-11-09:
> Ok, I did just notice that if the GW2000 get disconnected from the network the driver crashes. Mine is on wifi and I had to reboot my AP.
> 
> ......
> 
> After this you have to restart weewx. Might be better to just wait until the next interval to pull data than to exit.

I'm not saying the driver is perfect, but this is likely part WeeWX architecture and part GW2000 behaviour. Experience has been that Ecowitt gateway devices (GW1000/1100/2000) occasionally ignore an API command. The driver behaviour is to try again up to a default three tries total before raising the problem for the WeeWX engine to deal with. If the problem occurred during WeeWX startup or loading of the driver WeeWX considers this a critical error (there is a likely a config issue that means continuing is pointless) and WeeWX will exit. If the error occurs at other times (ie once the driver is loaded) WeeWX will wait 60 seconds and try again.

WeeWX can be set to not exit on such errors but continually 'try again' by setting [`loop_on_init = True`](http://weewx.com/docs/usersguide.htm#General) in `weewx.conf`. Some folks have had success with this (ie the Ecowitt device comes back) other have not. If the device does not come back then either the device needs to be power cycled, the wifi network brought down/up or the router restarted. Once the Ecowitt device is back WeeWX will pickup normal operation.

Anecdotally it appears that avoiding the need to broadcast and find the Ecowitt device lessens these issues. Essentially this means having your DHCP server allocate a fixed IP to the Ecowitt device and then specify the IP address off the device in the `[GW1000]` stanza in `weewx.conf`.

W0CHP, 2022-11-09:
> Anecdotally it appears that avoiding the need to broadcast and find the Ecowitt device lessens these issues. Essentially this means having your DHCP server allocate a fixed IP to the Ecowitt device and then specify the IP address off the device in the `[GW1000]` stanza in `weewx.conf`.

This is exactly what I do; my GW has a DHCP reservation and it's address is specified in `weewx.conf`. Additionally I have `loop_on_init = True` specified.

However, it's not the driver or weewx that's flawed;  the flaw seems to be the DHCP client stack/implementation in the GW itself causing disconnects:

1) It's not honoring the lease time in my `isc-dhcp` server config (renews many times in 24 hours when leases are set to 1 week)
2) As it renews its lease, it does indeed it seems to drop its IP *first* then renews, causing a brief disconnect (and this is when the driver logic will detect and retry - `loop_on_init = True` saves my ass here.).

I figured I'd put this out there, since the driver is working as designed; and the GW's DHCP client implementation is *not.* It's a shame we can't plug static IPs directly on the device.

jrtaylor71, 2022-11-10:
Thank you gjr80, this driver is awesome and works very well. Hopefully this issue I have opened will help others. Going to set `loop_on_init = True` and my DHCP server gives out the same ip to devices very long term. Going to create a DHCP reservation to help.

W0CHP, 2022-11-20:
> However, it's not the driver or weewx that's flawed; the flaw seems to be the DHCP client stack/implementation in the GW itself causing disconnects:
>
> 1. It's not honoring the lease time in my [...]

FYI: Since the release of firmware V2.2.0 this week, I've noticed not a single random GW disconnect. 👍

gjr80, 2023-07-22:
Closing as complete, can be re-opened if required.

closed 2023-07-22 by gjr80