gjr80/weewx-gw1000 #63: wee_device --info problem

belidi, 2023-07-17:
I have this problem  "
sudo ./wee_device --info
Using configuration file /home/weewx/weewx.conf
Using GW1000 driver version 0.6.0b1 (user.gw1000)
Traceback (most recent call last):
  File "/home/weewx/bin/wee_device", line 76, in <module>
    main()
  File "/home/weewx/bin/wee_device", line 72, in main
    device.configure(config_dict)
AttributeError: 'NoneType' object has no attribute 'configure'

and same for other wwe_device options. Normal?

mafen, 2023-07-17:
do you get the same issue if you use the 0.5 version?

belidi, 2023-07-17:
the installed version is the latest version that I downloaded from the "zip" file of the gituhub site. version which seems to be 0.6 and therefore newer than 0.5. however I do not see where I could find version 0.5

mafen, 2023-07-17:
https://github.com/gjr80/weewx-gw1000/releases/tag/v0.5.0b5 here is the beta of version 0.5

belidi, 2023-07-17:
sudo /home/weewx/util/scripts/./wee_extension --install= gw1000-0.5.0b5.tar.gz

Traceback (most recent call last):

  File "/home/weewx/bin/wee_extension", line 92, in <module>
    main()
  File "/home/weewx/bin/wee_extension", line 65, in main
    config_path, config_dict = weecfg.read_config(options.config, _args)
  File "/home/weewx/bin/weecfg/__init__.py", line 150, in read_config
    config_dict = configobj.ConfigObj(config_path,
  File "/usr/lib/python3/dist-packages/configobj.py", line 1229, in __init__
    self._load(infile, configspec)
  File "/usr/lib/python3/dist-packages/configobj.py", line 1287, in _load
    content = self._handle_bom(content)
  File "/usr/lib/python3/dist-packages/configobj.py", line 1437, in _handle_bom
    return self._decode(infile, self.encoding)
  File "/usr/lib/python3/dist-packages/configobj.py", line 1517, in _decode
    infile[i] = line.decode(encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte

gjr80, 2023-07-17:
A few topics have been covered in this issue, comments below.

>I have this problem "
>sudo ./wee_device --info
>Using configuration file /home/weewx/weewx.conf
>Using GW1000 driver version 0.6.0b1 (user.gw1000)
>Traceback (most recent call last):
>File "/home/weewx/bin/wee_device", line 76, in 
>main()
>File "/home/weewx/bin/wee_device", line 72, in main
>device.configure(config_dict)
>AttributeError: 'NoneType' object has no attribute 'configure'
>
>and same for other wwe_device options. Normal?

Yes. Not all drivers support `wee_device` and the Ecowitt gateway device driver is one of those drivers. `wee_device` is essentially used for configuring the station hardware (as distinct from configuring WeeWX and the driver), for the Ecowitt gateway devices this is done through the WSView Plus app or the device web interface (the Ecowitt gateway device API supports setting many of the same device parameters as can be set via WSView Plus and the web interface but I have chosen not to implement this at this time - no need and too many other things to do). The other thing that `wee_device` can do is display station hardware settings (again distinct from WeeWX/driver config/settings), the Ecwowitt gateway device driver does this by running the driver directly. Running the driver directly with the `--help` command line option displays what device info can be obtained when running the driver directly. Try using the following command:
```
PYTHONPATH=/home/weewx/bin python -m user.gw1000 --help
```

Note also that you seem to have an unusual WeeWX install location so you may need to change the path in the above command for it to work.

All that being said using `wee_device` with the Ecowitt gateway driver should give a more informative user friendly response, I will look at that for the future.

>the installed version is the latest version that I downloaded from the "zip" file of the gituhub site. version which seems to be 0.6 and therefore newer than 0.5. however I do not see where I could find version 0.5

The current released version is v0.5.0b5 available on the GitHub ['releases' tab](https://github.com/gjr80/weewx-gw1000/releases). v0.6.0 is almost ready for release but there are a few things I still need to tidy up, unfortunately I am away from home until late August so v0.6.0 has had to wait. That being said v0.6.0 should be fairly stable, no guarantees though and I cannot guarantee support at present.

>sudo /home/weewx/util/scripts/./wee_extension --install= gw1000-0.5.0b5.tar.gz
>
>Traceback (most recent call last):
>
>File "/home/weewx/bin/wee_extension", line 92, in 
>main()
>File "/home/weewx/bin/wee_extension", line 65, in main
>config_path, config_dict = weecfg.read_config(options.config, _args)
>File "/home/weewx/bin/weecfg/init.py", line 150, in read_config
>config_dict = configobj.ConfigObj(config_path,
>File "/usr/lib/python3/dist-packages/configobj.py", line 1229, in init
>self._load(infile, configspec)
>File "/usr/lib/python3/dist-packages/configobj.py", line 1287, in _load
>content = self._handle_bom(content)
>File "/usr/lib/python3/dist-packages/configobj.py", line 1437, in _handle_bom
>return self._decode(infile, self.encoding)
>File "/usr/lib/python3/dist-packages/configobj.py", line 1517, in _decode
>infile[i] = line.decode(encoding)
>UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8b in position 1: invalid start byte

Not an issue with the driver or WeeWX, but rather a problem with your config file, `weewx.conf`. Your `weewx.conf` has a byte 0x8b (in ASCII it's a left pointing arrow head ‹) as the first character and the python library used by WeeWX for handling config files is choking on that character. You will need to remove that character.

gjr80, 2023-07-22:
Implementing `wee_device` support was easier than expected and will be included in the v0.6.0 release.

The driver will still be able to be run directly, albeit with a reduced vocabulary.

closed 2023-07-22 by gjr80