gjr80/weewx-gw1000 #64: discovery does not appear return any devices

gjr80, 2023-09-10:
Using `--discover` results in the following error:
```
gary@vm-weewx-egd-29:~$ PYTHONPATH=/home/weewx/bin python3 -m user.gw1000 --discover --debug=3
Using configuration file /home/weewx/weewx.conf
debug level is '3'

IP address to be obtained by discovery
Port number set to default port number
Battery state filtering is 'False' (using the default)

Discovering devices on the local network. Please wait...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/weewx/bin/user/gw1000.py", line 8229, in <module>
    main()
  File "/home/weewx/bin/user/gw1000.py", line 8225, in main
    direct_gw.process_options()
  File "/home/weewx/bin/user/gw1000.py", line 6776, in process_options
    self.discover()
  File "/home/weewx/bin/user/gw1000.py", line 7778, in discover
    collector = GatewayCollector()
  File "/home/weewx/bin/user/gw1000.py", line 2608, in __init__
    self.device = GatewayDevice(ip_address=ip_address, port=port,
  File "/home/weewx/bin/user/gw1000.py", line 6024, in __init__
    self.api = GatewayApi(ip_address=ip_address,
  File "/home/weewx/bin/user/gw1000.py", line 4749, in __init__
    raise GWIOError(_msg)
__main__.GWIOError: Failed to detect device IP address and/or port after 3 attempts
```

It appears that discovery is not returning any device addresses.

gjr80, 2023-09-10:
Local testing of `GatewayApi.discover()` appears to indicate on-line gateway devices are responding, the response is received and the response is valid. 

Need to check decoding/parsing of the responses.

W0CHP, 2023-09-10:
Strange - discovery (and the driver) works great here. Debian 11/bullseye, Py3, WeeWX 4.10.2 installed via apt, and gw1000 driver v0.5.0b6:

```bash
$ PYTHONPATH=/usr/share/weewx python -m user.gw1000 --discover
Using configuration file /etc/weewx/weewx.conf

Discovering devices on the local network. Please wait...

GW2000 discovered at IP address 192.168.28.8 on port 45000
```

unicornis-pl, 2023-10-08:
Same problem here. Weewx 4.10.2 installed via apt on  Operating System: Raspbian GNU/Linux 10 (buster), Kernel: Linux 5.10.103-v7+, Architecture: arm, gw1000 driver v. 0.6.0b2.
```
pi@clockpi:/etc/weewx $ PYTHONPATH=/usr/share/weewx python3 -m user.gw1000 --discover
Using configuration file /etc/weewx/weewx.conf
```

```
Discovering devices on the local network. Please wait...
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/share/weewx/user/gw1000.py", line 8229, in <module>
    main()
  File "/usr/share/weewx/user/gw1000.py", line 8225, in main
    direct_gw.process_options()
  File "/usr/share/weewx/user/gw1000.py", line 6776, in process_options
    self.discover()
  File "/usr/share/weewx/user/gw1000.py", line 7778, in discover
    collector = GatewayCollector()
  File "/usr/share/weewx/user/gw1000.py", line 2616, in __init__
    log_unknown_fields=log_unknown_fields, debug=debug)
  File "/usr/share/weewx/user/gw1000.py", line 6036, in __init__
    debug=debug)
  File "/usr/share/weewx/user/gw1000.py", line 4749, in __init__
    raise GWIOError(_msg)
__main__.GWIOError: Failed to detect device IP address and/or port after 3 attempts`
```