gjr80/weewx-gw1000 #65: Install fails with v5.0rc3

JonFear, 2024-01-09:
Trialling the new rc for v5 Weewx and install fails. See below:

`jon@weewx-v5:~$ weectl extension install https://github.com/gjr80/weewx-gw1000/releases/download/v0.4.2/gw1000-0.4.2.tar.gz
Using configuration file /etc/weewx/weewx.conf
Install extension 'https://github.com/gjr80/weewx-gw1000/releases/download/v0.4.2/gw1000-0.4.2.tar.gz' (y/n)? y
Extracting from tar archive /tmp/tmp24ik1yds
Traceback (most recent call last):
  File "/usr/share/weewx/weectl.py", line 74, in <module>
    main()
  File "/usr/share/weewx/weectl.py", line 66, in main
    namespace.func(namespace)
  File "/usr/share/weewx/weectllib/__init__.py", line 121, in dispatch
    namespace.action_func(config_dict, namespace)
  File "/usr/share/weewx/weectllib/extension_cmd.py", line 116, in install_extension
    ext.install_extension(namespace.source, no_confirm=namespace.yes)
  File "/usr/share/weewx/weecfg/extension.py", line 122, in install_extension
    extension_name = self._install_from_file(test_fd.name, info.get_content_subtype())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/extension.py", line 160, in _install_from_file
    extension_name = self.install_from_dir(extension_dir)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/extension.py", line 171, in install_from_dir
    installer_path, installer = weecfg.get_extension_installer(extension_dir)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/__init__.py", line 715, in get_extension_installer
    installer = loader()
                ^^^^^^^^
  File "/tmp/tmpyvn3ukgg/gw1000/install.py", line 309, in loader
  File "/tmp/tmpyvn3ukgg/gw1000/install.py", line 314, in __init__
  File "/usr/lib/python3.11/distutils/version.py", line 40, in __init__
    self.parse(vstring)
  File "/usr/lib/python3.11/distutils/version.py", line 137, in parse
    raise ValueError("invalid version number '%s'" % vstring)
ValueError: invalid version number '5.0.0rc3'`

Thought it maybe sudo related so:

`jon@weewx-v5:~$ sudo weectl extension install https://github.com/gjr80/weewx-gw1000/releases/download/v0.4.2/gw1000-0.4.2.tar.gz
sudo: unable to resolve host weewx-v5.local: Name or service not known
Using configuration file /etc/weewx/weewx.conf
Install extension 'https://github.com/gjr80/weewx-gw1000/releases/download/v0.4.2/gw1000-0.4.2.tar.gz' (y/n)? y
Extracting from tar archive /tmp/tmpqg4rpysm
Traceback (most recent call last):
  File "/usr/share/weewx/weectl.py", line 74, in <module>
    main()
  File "/usr/share/weewx/weectl.py", line 66, in main
    namespace.func(namespace)
  File "/usr/share/weewx/weectllib/__init__.py", line 121, in dispatch
    namespace.action_func(config_dict, namespace)
  File "/usr/share/weewx/weectllib/extension_cmd.py", line 116, in install_extension
    ext.install_extension(namespace.source, no_confirm=namespace.yes)
  File "/usr/share/weewx/weecfg/extension.py", line 122, in install_extension
    extension_name = self._install_from_file(test_fd.name, info.get_content_subtype())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/extension.py", line 160, in _install_from_file
    extension_name = self.install_from_dir(extension_dir)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/extension.py", line 171, in install_from_dir
    installer_path, installer = weecfg.get_extension_installer(extension_dir)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/__init__.py", line 715, in get_extension_installer
    installer = loader()
                ^^^^^^^^
  File "/tmp/tmp0vhzhu2g/gw1000/install.py", line 309, in loader
  File "/tmp/tmp0vhzhu2g/gw1000/install.py", line 314, in __init__
  File "/usr/lib/python3.11/distutils/version.py", line 40, in __init__
    self.parse(vstring)
  File "/usr/lib/python3.11/distutils/version.py", line 137, in parse
    raise ValueError("invalid version number '%s'" % vstring)
ValueError: invalid version number '5.0.0rc3'`

Any thoughts?

Best wishes

Jon

JonFear, 2024-01-09:
Also doing a clone of the latest code still fails install but with slightly different messages. Rather than confuse the whole thing, I best leave it to someone who knows what they are talking about (clue, it ain't me!)!

gjr80, 2024-01-09:
This is a known issue with extension packages that use `distils.StrictVersion` to check version numbers. In a nutshell, `StrictVersion` does not handle version strings that annotate release candidates with `rc`. There are several threads in [weewx-user](https://groups.google.com/g/weewx-user) and [weewx-development](https://groups.google.com/g/weewx-development) on the subject, most include crude workarounds, just search for 'StrictVersion'. The problem will disappear once  a non-release candidate WeeWX version is installed.

Nonetheless, I have created a v0.6.0b3 extension package that uses an installer that does not use `StrictVersion`, it should work fine with any WeeWX release including release candidates. It can be found on the [releases page](https://github.com/gjr80/weewx-gw1000/releases).

JonFear, 2024-01-09:
Fantastic, thank you. That looked better. Surprised Tom has not built this in as the Ecowitt is becoming quite a large community...

closed 2024-01-09 by gjr80