gjr80/weewx-gw1000 #68: gw1000-0.6.0b6.tar.gz nicht verfügbar

Alter49, 2024-02-03:
Auflösen des Hostnamens github.com (github.com)… 140.82.121.4
Verbindungsaufbau zu github.com (github.com)|140.82.121.4|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2024-02-03 17:39:43 FEHLER 404: Not Found.

gjr80, 2024-02-08:
I don't know where a link to `gw1000-0.6.0b6.tar.gz` was posted as I never released this file publicly.

Please use [v0.6.0](https://github.com/gjr80/weewx-gw1000/releases/tag/v0.6.0) instead.

closed 2024-02-08 by gjr80

Alter49, 2024-02-08:
Hallo, einige WIKI beschreiben die Nutzung des gw1000 Treibers und verlinken dann auf Treiber downloads im Arbeitsstatus.
Dadurch kommt es zu diesen Fehlermeldungen. Ist nur ein Hinweis. 
Da kann das Team wahrscheinlich gar nichts dafür.