gjr80/weewx-gw1000 #73: Error message with using as service

GameWatchProjects, 2024-02-22:
Hello,

at first, thank you very much for spending your time to give us this driver for the Ecowitt gateway GW1000 ...

I have use a clean install of WeeWX 5.0.2 and install the plugin with your installations instruction.

First step i install the plugin as a WeeWX driver and with the command

`$ sudo  PYTHONPATH=/usr/share/weewx python3 /etc/weewx/bin/user/gw1000.py --live-data --ip-address=192.168.12.8`
i become the follow output

> Using configuration file /etc/weewx/weewx.conf
> 
> Interrogating GW1100 at `192.168.12.8:45000`
> 
> Displaying data using the WeeWX METRIC unit group.
> 
> GW1100 live sensor data (2024-02-23 00:01:13 CET (1708642873)): 'absbarometer': '969.400000', 'annual_reset': '0.000000', 'day_reset': '0.000000', 'daymaxwind': '21.960000', 'gustspeed': '11.160000', 'heap_free': '29232 B', 'inhumid': '59.000000', 'intemp': '18.000000', 'light': '0.000000', 'lightningcount': '0.000000', 'lightningdettime': 'None', 'lightningdist': 'None', 'outhumid': '63.000000', 'outtemp': '8.600000', 'rain_priority': '1.000000', 'relbarometer': '969.400000', 'soilmoist1': '66.000000', 't_rainday': '0.000000', 't_rainevent': '1.470000', 't_raingain': '0.100000', 't_rainmonth': '5.840000', 't_rainrate': '0.000000', 't_rainweek': '3.480000', 't_rainyear': '5.840000', 'temp9': '8.700000', 'temperature_comp': '1.000000', 'uv': '0.000000', 'uvi': '0.000000', 'week_reset': '0.000000', 'wh51_ch1_batt': '1.60', 'wh51_ch1_sig': '4.000000', 'wh57_batt': '5.000000', 'wh57_sig': '4.000000', 'wh65_batt': '0.000000', 'wh65_sig': '4.000000', 'winddir': '276.000000', 'windspeed': '7.920000', 'wn34_ch1_batt': '1.68', 'wn34_ch1_sig': '4.000000'

With the command:
`$ sudo  PYTHONPATH=/usr/share/weewx python3 /etc/weewx/bin/user/gw1000.py --test-driver --ip-address=192.168.12.8`
i become the follow output

> Using configuration file /etc/weewx/weewx.conf
> 
> Interrogating GW1100 at 192.168.12.8:45000
> 
> 2024-02-23 00:08:08 CET (1708643288): 'dateTime': '1708643288', 'daymaxwind': '8.7', 'dayRain': '0.0', 'extraTemp9': '8.6', 'heap_free': '29236', 'inHumidity': '59', 'inTemp': '17.9', 'lightning_distance': 'None', 'lightning_last_det_time': 'None', 'lightning_strike_count': 'None', 'lightningcount': '0', 'luminosity': '0.0', 'monthRain': '58.4', 'outHumidity': '63', 'outTemp': '8.5', 'pressure': '969.8', 'rain': 'None', 'rainRate': '0.0', 'relbarometer': '969.8', 'soilMoist1': '66', 'stormRain': '14.7', 'usUnits': '17', 'UV': '0', 'uvradiation': '0.0', 'weekRain': '34.8', 'wh51_ch1_batt': '1.6', 'wh51_ch1_sig': '4', 'wh57_batt': '5', 'wh57_sig': '4', 'wh65_batt': '0', 'wh65_sig': '4', 'windDir': '252', 'windGust': '5.1', 'windSpeed': '3.2', 'wn34_ch1_batt': '1.68', 'wn34_ch1_sig': '4', 'yearRain': '58.4'

But when i want to the installation as WeeWX-Service and use the command
`$ sudo  PYTHONPATH=/usr/share/weewx python3 /etc/weewx/bin/user/gw1000.py --test-service --ip-address=192.168.12.8`

i become this error message

 sing configuration file /etc/weewx/weewx.conf
Traceback (most recent call last):
  File "/etc/weewx/bin/user/gw1000.py", line 8570, in <module>
    main()
  File "/etc/weewx/bin/user/gw1000.py", line 8566, in main
    direct_gw.process_options()
  File "/etc/weewx/bin/user/gw1000.py", line 7059, in process_options
    self.test_service()
  File "/etc/weewx/bin/user/gw1000.py", line 8385, in test_service
    engine = weewx.engine.StdEngine(config)
  File "/usr/share/weewx/weewx/engine.py", line 89, in __init__
    self.loadServices(config_dict)
  File "/usr/share/weewx/weewx/engine.py", line 157, in loadServices
    obj = weeutil.weeutil.get_object(svc)(self, config_dict)
  File "/usr/share/weewx/weeutil/weeutil.py", line 1404, in get_object
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'user'

gjr80, 2024-02-23:
Thanks for picking this up. The number of different path permutations under WeeWX v5 is significantly more than there were under earlier versions. Unfortunately, I seem to have omitted some `PYTHONPATH` settings when using `--test-service`. The command you need is:
```
PYTHONPATH=/usr/share/weewx:/etc/weewx/bin python3 /etc/weewx/bin/user/gw1000.py --test-service --ip-address=192.168.12.8
```
There should be no need for `sudo`.

I have updated the readmes, you should find corrected version on the [Ecowitt Gateway Driver repo home page](https://github.com/gjr80/weewx-gw1000?tab=readme-ov-file#ecowitt-gateway-formerly-gw1000-driver).

GameWatchProjects, 2024-02-23:
@gjr80

[image] ... Thank you very much, so i've no problem and the service runs without error's ...

closed 2024-02-28 by gjr80