gjr80/weewx-gw1000 #74: wee_device --firmware crashes unless an explicit IP address is given

dougbert-b, 2024-02-23:
With the latest GW1000 driver, most wee_device options can auto-detect the GW1000's IP address:

```
blackbox:/etc/weewx>wee_device --live-data
Using configuration file /etc/weewx/weewx.conf
Using GW1000 driver version 0.6.1 (user.gw1000)

Interrogating GW1000 at 192.168.1.165:45000

Displaying data using the WeeWX METRIC unit group.

GW1000 live sensor data (2024-02-22 21:18:23 EST (1708654703)): 'absbarometer': '995.200000', 'annual_reset': '1.000000', 'day_reset': '0.000000', 'daymaxwind': '23.040000', 'gain0': '1.000000', 'gain1': '1.000000', 'gain2': '1.000000', 'gain3': '1.000000', 'gain4': '1.000000', 'gain5': '1.000000', 'gain6': '1.000000', 'gain7': '1.000000', 'gain8': '1.000000', 'gain9': '1.000000', 'gustspeed': '6.840000', 'humid1': '70.000000', 'humid2': '56.000000', 'inhumid': '32.000000', 'intemp': '21.100000', 'light': '0.000000', 'outhumid': '95.000000', 'outtemp': '3.200000', 'p_rainday': '0.190000', 'p_rainevent': '0.170000', 'p_rainmonth': '3.180000', 'p_rainrate': '0.240000', 'p_rainweek': '0.190000', 'p_rainyear': '3.180000', 'rain_priority': '2.000000', 'relbarometer': '1006.000000', 'temp1': '-11.900000', 'temp2': '3.800000', 'temperature_comp': '1.000000', 'uv': '0.000000', 'uvi': '0.000000', 'week_reset': '0.000000', 'wh31_ch1_batt': '0.000000', 'wh31_ch1_sig': '4.000000', 'wh31_ch2_batt': '0.000000', 'wh31_ch2_sig': '4.000000', 'winddir': '123.000000', 'windspeed': '6.480000', 'ws90_batt': '3.28', 'ws90_sig': '4.000000'
```

But the --firmware option will crash unless you explicitly provide the IP address.

This works:

```
blackbox:/etc/weewx>wee_device --firmware --ip-address=192.168.1.165
Using configuration file /etc/weewx/weewx.conf
Using GW1000 driver version 0.6.1 (user.gw1000)

Interrogating GW1000 at 192.168.1.165:45000

    installed GW1000 firmware version is GW1000B_V1.7.7

    could not determine if a firmware update is available for this GW1000
```

This crashes:

```
blackbox:/etc/weewx>wee_device --firmware 
Using configuration file /etc/weewx/weewx.conf
Using GW1000 driver version 0.6.1 (user.gw1000)

Interrogating GW1000 at 192.168.1.165:45000

    installed GW1000 firmware version is GW1000B_V1.7.7
Traceback (most recent call last):
  File "/usr/share/weewx/wee_device", line 76, in <module>
    main()
  File "/usr/share/weewx/wee_device", line 72, in main
    device.configure(config_dict)
  File "/usr/share/weewx/weewx/drivers/__init__.py", line 67, in configure
    self.do_options(options, parser, config_dict, not options.noprompt)
  File "/usr/share/weewx/user/gw1000.py", line 2486, in do_options
    direct_gw.process_options()
  File "/usr/share/weewx/user/gw1000.py", line 7087, in process_options
    self.firmware()
  File "/usr/share/weewx/user/gw1000.py", line 7890, in firmware
    ws90_fw = device.ws90_firmware_version
  File "/usr/share/weewx/user/gw1000.py", line 6474, in ws90_firmware_version
    sensors = self.http.get_sensors_info()
  File "/usr/share/weewx/user/gw1000.py", line 6092, in get_sensors_info
    page_1 = self.request('get_sensors_info', data={'page': 1})
  File "/usr/share/weewx/user/gw1000.py", line 5971, in request
    stem = ''.join(['http://', self.ip_address])
TypeError: sequence item 1: expected str instance, NoneType found
blackbox:/etc/weewx>
```

There is another strange result if you give both the --live-data and --firmware options, in that order:

```
blackbox:/etc/weewx>wee_device --live-data --firmware 
Using configuration file /etc/weewx/weewx.conf
Using GW1000 driver version 0.6.1 (user.gw1000)

Unable to connect to device at None: Failed to detect device IP address and/or port after 3 attempts

    Things to check include that the correct device IP address is being used,
    the device is powered on and the device is not otherwise disconnected from
    the local network.
blackbox:/etc/weewx>
```

gjr80, 2024-02-23:
For some reason a part of the driver code is not receiving the gateway device IP address unless it is specified on the command line. I have been unable to reproduce this behaviour on any of my devices. Could I ask for some further information:
1. Could you provide the console output of the following command:
```
wee_device --firmware --debug=3
```
2. Could you check the WeeWX log for the output when the above command was run and post a copy of the log output back here. The log output should be in the same location as your normal WeeWX log output
3. Could you provide a copy of your `[GW1000]` stanza from `weewx.conf`

gjr80, 2024-02-23:
Actually, stand by on my last. I've now been able to reproduce the problem. Let me work through to find the cause.

gjr80, 2024-02-23:
OK, I think it is fixed. Can you confirm that you did not specify an IP address in your `[GW1000]` stanza in `weewx.conf`? 

Can you please try v0.6.2:
1. download the v0.6.2 driver file (`gw1000.py`) to your WeeWX machine using the following command:
```
wget -P /var/tmp https://raw.githubusercontent.com/gjr80/weewx-gw1000/master/bin/user/gw1000.py
```
2. copy the downloaded `gw1000.py` from `/var/tmp` to `/usr/share/weewx/user`:
```
sudo cp /var/tmp/gw1000.py /usr/share/weewx/user/
```
then try your commands again.

dougbert-b, 2024-02-23:
>    1. copy the downloaded gw1000.py from /var/tmp to /usr/share/weewx/user

Yes, the --firmware command works for me when IP address discovery is
used.  Thanks!

BTW, if I do "wee_device --live-data --firmware", the firmware data is
printed, but no live data is printed.
(It's not clear to me if this is actually a bug...)

Doug

gjr80, 2024-02-28:
> BTW, if I do "wee_device --live-data --firmware", the firmware data is
> printed, but no live data is printed.
> (It's not clear to me if this is actually a bug...)

It's not a bug, `wee_device` only accepts one 'verb' at a time; so you can use `--live-data` or `--firmware` but not both together (well you can but only one will be actioned and that will be the one that encountered first when the driver processes the command line arguments).

I could add code to ensure that only one `verb` can be specified, but given this is the first time the issue has been raised and as I'm working on v0.7.0 (which will require WeeWX v5 and uses `weectl device` in lieu of `wee_device`) I will look at such changes for 0.7.0 rather than add them to another 0.6.x release.

swfrx, 2024-12-27:
Tried this with WeeWx 5.1 and a GW3000. The IP Address is setup in weewx.conf

```
$ weectl device --firmware
Using configuration file /etc/weewx/weewx.conf
Using driver user.gw1000.
Using GW1000 driver version 0.6.3 (user.gw1000)

Interrogating 'unknown model' at 192.168.1.163:45000

    installed 'unknown model' firmware version is GW3000A_V1.0.0

    the firmware is up to date for this 'unknown model'
```

I have a GW1000 on the network as well so can't test if setting the IPA to 'auto' causes problems.