gjr80/weewx-gw1000 #76: Rain gauge information does not translate for WeeWX and beyond

svenjr, 2024-04-12:
Information shows up via EcoWitt (see below)

[image]

but this data does not track through to WeeWx
[image]

gjr80, 2024-04-12:
If you have a piezo rain gauge you need to configure the driver and/or WeeWX to use the rain data from the piezo rain gauge. The easiest way to do this is to alter the driver field map to map the piezo rain data to the standard WeeWX rain fields. Did you do this?

Refer to the _[I have an Ecowitt piezo rain gauge, how do I get my rainfall data into WeeWX?](https://github.com/gjr80/weewx-gw1000/wiki/Frequently-asked-questions#sensors)_ FAQ in the driver wiki, it should help you.

gjr80, 2024-08-24:
Closing due to no response. Can be re-opened if required.

closed 2024-08-24 by gjr80