gjr80/weewx-gw1000 #79: GW3000 backfill

swfrx, 2024-12-25:
For me, the GW3000 works with this driver. What would be great is backfilling from the SD card. That would mean the GW3000 could be plugged into a UPS, and when WeeWx is restarted it can use that file to backfill.

A stand-along utility to do that would be fine in my book - or an extension of `weectl database`.

mafen, 2024-12-25:
Looking at the manual, it does the exports in CSV and weewx can read data from CSV https://weewx.com/docs/5.1/utilities/weectl-import-csv/ so it is just matter for format of the CSV, can you provide a sample?

swfrx, 2024-12-27:
I hadn’t noticed that utility. I should be able to work that out myself, having used databases at work for many years!

Very small sample attached.